wmbusmeters/wmbusmeters #987: rtl_wmbus not found although correctly installed

Solid48, 2023-06-30:
### Type of request

Bug

### OS version

Ubuntu 23.04

### wmbusmeters version

it says "_", but i downloaded it like 2 to 3 months ago

### Your message goes here

I installed wmbusmeters and tried to use it with an rtl-sdr dongle. After doing so I ran into the error:
`Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path`
(I am running this via bash with the command `wmbusmeters auto:t1`)

I already installed both the Osmocom rtl-sdr driver and the rtl_wmbus software by xaelsouth. I also checked that rtl_wmbus is working. When using the command
`rtl_sdr -f 868.95M -s 1600000 -`
I always first get the message that there is a rtl_sdr plugged in (`Using device 0: Generic RTL2832U OEM`) and shortly after I get random letters (or symbols and the questionmark in the square which indicates unoccupied Ascii codes), which I would interpret as the white noise the antenna is whitnessing. If I pull out the antenna it says `No supported device found.`.

Apparently there is an issue communicating between rtl-wmbus and wmbusmeters which I do not know how to resolve.

BIBOLV, 2023-07-01:
Have you tried running `wmbusmeters rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)` and also check location of rtl_sdr and rtl_wmbus with `which rtl_sdr && which rtl_wmbus` - are they located located in /usr/bin/

Solid48, 2023-07-02:
Thanks a lot for the help, I would not have noticed the mistake without your help @BIBOLV. 
Apparently rtl_sdr was installed but rtl_wmbus was not. This is due to the fact that xaelsouths library has a weird way of comunicating between parts. They actually install (at least on linux) only the driver which gets data from the antenna but not the Software decoding it. The executable for that software stays in the subfolder build. If you navigate to the folder of the Library and typed these two lines
`rtl_sdr samples.bin -f 868.95M -s 1600000`
`cat samples.bin | build/rtl_wmbus`
you would get the output wmbusmeters wants. 
If someone runs into a similar issue I recommend the following steps (which worked for me): 

1.: 
navigate to the directory of the rtl-wmbus library

2.:
check if everything is working (otherwise you need to use a different approach for your problem)
use these commands: 
`rtl_sdr samples.bin -f 868.95M -s 1600000`
`cat samples.bin | build/rtl_wmbus`
(run the first one for a few seconds then cancel with ctrl + c and run the second one)
The output schould look something like this: 
`C1;1;1;2023-07-02 09:26:30.000;120;119;89621521;` followed by a long Hexadecimal number and that pattern should appear multiple times.

3.:
Copy the file into /usr/bin with the following command
`cp build/rtl_wmbus /usr/bin/`

4.:
Confirm that it is working now by typing the following commands similar to step 2
`rtl_sdr samples.bin -f 868.95M -s 1600000`
`cat samples.bin | rtl_wmbus`
And obviously also try if wmbusmeters is working now with the following command
`wmbusmeters auto:t1`

closed 2023-07-02 by Solid48