wmbusmeters/wmbusmeters #989: Stopped working after upgrading to 1.14.0

grest, 2023-07-03:
### Type of request

Bug

### HAOS version

 2023.6.3

### wmbusmeters version

1.14.0

### Your message goes here

After upgrading to 1.14.0 the addin stopped working. In the logs, everything looks fine, but the sensors are not updating at all, showing the last available value before the upgrade.

grest, 2023-07-03:
Downgrade to 1.13.1 did not help. Only the CPU utilisation is lower now, from 50% to 29%.

grest, 2023-07-04:
Ok, I'll set loglevel=debug and upgrade again. By the way, I reproduced the problem already two times by upgrading and downgrading, so this will be the third attempt.

Yes, exactly. The CPU utilisation is much higher - I have a very poor hardware, but it was working till now.

weetmuts, 2023-07-04:
Thanks, I need the debug log to see if anything weird is going on.

Can you please check if it is rtl-wmbus or wmbusmeters that eat the additional cpu?

grest, 2023-07-04:
> Thanks, I need the debug log to see if anything weird is going on.
> 
> Can you please check if it is rtl-wmbus or wmbusmeters that eat the additional cpu?

Sorry for my ignorance, but how to distinct CPU utilisation?

grest, 2023-07-04:
More verbose logs

[13:43:58] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00263629_timestamp/config
[13:43:58] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00263629_total_energy_consumption_kwh/config
[13:43:58] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00263629_total_energy_production_kwh/config
[13:43:58] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00263629_voltage_at_phase_1_v/config
[13:43:59] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00263629_voltage_at_phase_2_v/config
[13:43:59] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00263629_voltage_at_phase_3_v/config
[13:43:59] INFO: MQTT Discovery cleanup...
[13:44:00] INFO: Starting web configuration service.
[13:44:00] INFO: Running wmbusmeters ...
(config) "logtelegrams" "false"
(wmbusmeters) version: 1.14.0
(config) using device: rtlwmbus:c1 
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based SDR-receivers
rtl_sdr version 0.8 github.com/librtlsdr (May  3 2023)
rtl-sdr library 0.8 github.com/librtlsdr (May  3 2023)

Usage:	rtl_sdr -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-w tuner_bandwidth (default: automatic)]
	[-d device_index or serial (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-O	set RTL driver options seperated with ':', e.g. -O 'bc=30000:agc=0' ]
		f=<freqHz>            set tuner frequency
		bw=<bw_in_kHz>        set tuner bandwidth
		bc=<if_in_Hz>         set band center relative to the complex-base-band '0' frequency
		                        puts the tuner frequency onto this if frequency (default: 0)
		sb=<sideband>         set tuner sideband/mirror: 'L' or '0' for lower side band,
		                        'U' or '1' for upper side band. default for R820T/2: 'L'
		agc=<tuner_gain_mode> activates tuner agc with '1'. deactivates with '0'
		gain=<tenth_dB>       set tuner gain. 400 for 40.0 dB
		ifm=<tuner_if_mode>   set R820T/2 tuner's variable-gain-amplifier (VGA). default: 10011
		                        0: activate agc controlled from RTL2832's feedback
		                        around 0: set gain in 10th dB. 408 for +40.8 dB
		                        5000+val: set gain to val in 10th dB. 5408 for +40.8 dB
		                        10000+idx: set gain idx 0 .. 15: 10015 for maximum gain
		dagc=<rtl_agc>        set RTL2832's digital agc (after ADC). 1 to activate. 0 to deactivate
		ds=<direct_sampling>  deactivate/bypass tuner with 1
		dm=<ds_mode_thresh>   set dynamic direct threshold mode or threshold frequency:
		                        0: use I & Q; 1: use I; 2: use Q; 3: use I below threshold frequency;
		                        4: use Q below threshold frequency (=RTL-SDR v3)
		                        other values set the threshold frequency
		harm=<Nth_harmonic>   R820T/2: use Nth harmonic for frequencies above 1.76 GHz. default: 5
		vcocmin=<current>     set R820T/2 VCO current min: 0..7: higher value is more current
		vcocmax=<current>     set R820T/2 VCO current max: 0..7
		vcoalgo=<algo>        set R820T/2 VCO algorithm. 0: default. 1: with vcomax=3.9G. 2: Youssef/Carl
		Tp=<gpio_pin>         set GPIO pin for Bias T, default =0 for rtl-sdr.com compatible V3
		T=<bias_tee>          1 activates power at antenna one some dongles, e.g. rtl-sdr.com's V3
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	[-N no dithering (default: use dithering)]
	[-H write wave Header to file (default: off)]
	filename (a '-' dumps samples to stdout)

(serial) EVENT thread interrupted
<<<
(shell) return code 1
(shell) 285 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
[... 7 lines cut ...]
Started config rtlwmbus[00000001] listening on c1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 291 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
[13:44:05] INFO: Starting Nginx...
2023/07/04 13:44:05 [notice] 83#83: using the "epoll" event method
2023/07/04 13:44:05 [notice] 83#83: nginx/1.24.0
2023/07/04 13:44:05 [notice] 83#83: OS: Linux 5.10.0-23-amd64
2023/07/04 13:44:05 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2023/07/04 13:44:05 [notice] 83#83: start worker processes
2023/07/04 13:44:05 [notice] 83#83: start worker process 315
172.30.32.2 - - [04/Jul/2023:13:44:38 +0200] "GET / HTTP/1.1" 200 1299 "http://dhomer:8123/c8a990ad_wmbusmeters-ha-addon" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
172.30.32.2 - - [04/Jul/2023:13:44:38 +0200] "GET /bootstrap.bundle.min.js HTTP/1.1" 200 80762 "http://dhomer:8123/api/hassio_ingress/DmK9DBjq8d4w6nbj32uuS8qptCCRYoiSIraDsSrI2QA/" "Mozilla/5.0 (Windows NT 10.0; W [... cut ...]
172.30.32.2 - - [04/Jul/2023:13:44:38 +0200] "GET /wmbusmeters.png HTTP/1.1" 200 55338 "http://dhomer:8123/api/hassio_ingress/DmK9DBjq8d4w6nbj32uuS8qptCCRYoiSIraDsSrI2QA/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
172.30.32.2 - - [04/Jul/2023:13:44:38 +0200] "GET /get_json HTTP/1.1" 200 532 "http://dhomer:8123/api/hassio_ingress/DmK9DBjq8d4w6nbj32uuS8qptCCRYoiSIraDsSrI2QA/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"

weetmuts, 2023-07-04:
@BIBOLV is there a way to run top inside the docker container?

grest, 2023-07-04:
actually I know how to do that :) Assuming container contains this package.

weetmuts, 2023-07-04:
Great! Given the increased cpu usage and lack of telegrams, it might be rtl_wmbus behaving bad.

grest, 2023-07-04:
Yes, indeed. Compared top from 1.13.1 and 1.14.0 it's purely relate to rtl_wmbus

root     R      824   0%   0  50% /usr/bin/rtl_wmbus -s

weetmuts, 2023-07-04:
Can you restart the docker to make sure the rtl_sdr and rtl_wmbus are started fresh? ie there is not a second rtl_wmbus running hogging the telegrams and using cpu?

weetmuts, 2023-07-04:
What does this command print:
ps axcf | grep rtl

grest, 2023-07-04:
> Can you restart the docker to make sure the rtl_sdr and rtl_wmbus are started fresh? ie there is not a second rtl_wmbus running hogging the telegrams and using cpu?

Do you mean the docker host on the linux (HA supervised)? Or the docker container? 

By the way, I'm not sure if this is important, but on the downgrade I need to unplug and plug again the DVBT dongle to make it working again. Backup restore + downgrade, and restarting addon is not enough, I need to additionally remove the DVBT dongle from the USB port, plug it in, and start addon again.

grest, 2023-07-04:
> What does this command print: ps axcf | grep rtl

Run within container as I understand?

BIBOLV, 2023-07-04:
Yes, please run them inside container. Also, if you could try to restart whole HA, just to ensure that there are no residual processes left.

grest, 2023-07-05:
> Yes, please run them inside container.

  291 root      0:00 /bin/sh -c /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2
  292 root      0:02 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
  293 root      8:03 /usr/bin/rtl_wmbus -s

grest, 2023-07-05:
> Can you restart the docker to make sure the rtl_sdr and rtl_wmbus are started fresh? ie there is not a second rtl_wmbus running hogging the telegrams and using cpu?

I did it. I even restarted the host machine

oandry72, 2023-07-06:
Hi there,
I have the same issue.
No any telegrams from my meters after the last upgrade from 1.13.1 to 1.14.0
Timestamp of meters is still the same (July 4, 2023 at 22:52:57) before I updated addon.

**Home Assistant: 2023.7.0
Supervisor: 2023.07.1
HA addon Wmbusmeters: 1.14.0**

log
```
[19:37:13] INFO: Adding meter-0003 ...
[19:37:14] INFO: Adding meter-0004 ...
[19:37:14] INFO: Adding meter-0005 ...
[19:37:14] INFO: Adding meter-0006 ...
[19:37:14] INFO: Adding meter-0007 ...
[19:37:15] INFO: Adding meter-0008 ...
[19:37:15] INFO: Adding meter-0009 ...
[19:37:15] INFO: Generating MQTT configuration ... 
[19:37:16] INFO: Broker core-mosquitto will be used.
[19:37:17] INFO: 
MQTT Discovery ...
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
[19:37:18] INFO:  Adding meter: Cold Water Meter ...
[19:37:19] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00897110_rssi_dbm/config
[19:37:19] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00897110_timestamp/config
[19:37:20] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00897110_total_m3/config
[19:37:20] INFO:  Adding meter: Hot Water Meter ...
[19:37:20] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00817495_rssi_dbm/config
[19:37:21] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00817495_timestamp/config
[19:37:21] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00817495_total_m3/config
[19:37:21] INFO:  Adding meter: Kitchen Heat Meter ...
[19:37:22] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_current_hca/config
[19:37:22] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_esb_date/config
[19:37:22] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_previous_hca/config
[19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_rssi_dbm/config
[19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_season_start_date/config
[19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_temp_room_avg_c/config
[19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_temp_room_prev_avg_c/config
[19:37:24] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_timestamp/config
[19:37:24] INFO:  Adding meter: Bedroom Heat Meter ...
[19:37:24] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_current_hca/config
[19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_esb_date/config
[19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_previous_hca/config
[19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_rssi_dbm/config
[19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_season_start_date/config
[19:37:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_temp_room_avg_c/config
[19:37:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_temp_room_prev_avg_c/config
[19:37:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_timestamp/config
[19:37:27] INFO:  Adding meter: Kidsroom Heat Meter ...
[19:37:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_current_hca/config
[19:37:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_esb_date/config
[19:37:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_previous_hca/config
[19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_rssi_dbm/config
[19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_season_start_date/config
[19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_temp_room_avg_c/config
[19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_temp_room_prev_avg_c/config
[19:37:29] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_timestamp/config
[19:37:29] INFO:  Adding meter: LivingroomL Heat Meter ...
[19:37:29] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_current_hca/config
[19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_esb_date/config
[19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_previous_hca/config
[19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_rssi_dbm/config
[19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_season_start_date/config
[19:37:31] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_temp_room_avg_c/config
[19:37:31] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_temp_room_prev_avg_c/config
[19:37:31] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_timestamp/config
[19:37:32] INFO:  Adding meter: LivingroomR Heat Meter ...
[19:37:32] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262835_current_hca/config
[19:37:32] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262835_esb_date/config
[... 24 lines cut ...]
[19:37:39] INFO: MQTT Discovery cleanup...
[19:37:40] INFO: Starting web configuration service.
[19:37:40] INFO: Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on t1
[19:37:41] INFO: Starting Nginx...
2023/07/06 19:37:41 [notice] 83#83: using the "epoll" event method
2023/07/06 19:37:41 [notice] 83#83: nginx/1.24.0
[... cut ...]
172.30.32.2 - - [06/Jul/2023:19:41:28 +0200] "GET /get_json HTTP/1.1" 200 1604 "https://andry72.duckdns.org:8123/api/hassio_ingress/Xr5BmVOxp7ougwP2yAW2q81wMeNw167S-IWxcecPxy8/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
```

```
c8a990ad-wmbusmeters-ha-addon:/wmbusmeters# ps axcf | grep rtl
  708 root      0:00 /bin/sh -c /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
  709 root      0:37 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
  710 root     10:56 /usr/bin/rtl_wmbus -s
```

grest, 2023-07-08:
@weetmuts, @BIBOLV - any advice how to debug this further?

weetmuts, 2023-07-08:
It is indeed strange...there are no changes between 1.13 and 1.14 that should affect this code...

Can you run just the rtl wmbus command inside the docker. I.e.

/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - | /usr/bin/rtl_wmbus -s

perhaps the piping of stderr to dev null hides some important message.

BIBOLV, 2023-07-09:
Tested RTL-SDR and Wmbusmeters 1.14.0 with Home Assistant 2023.4.6 and Home Assistant 2023.7.1 also Wmbusmeters 1.13.1-45 with Home Assistant 2023.7.1 - was not able to replicate any isses, telegrams were received as expected. I was not able to replicate upgrade between versions though.

koles142, 2023-07-12:
> Hi there, I have the same issue. No any telegrams from my meters after the last upgrade from 1.13.1 to 1.14.0 Timestamp of meters is still the same (July 4, 2023 at 22:52:57) before I updated addon.
> 
> But I don't have issues with the Add-on CPU Usage. ~ 11-13% before update ~ 9-10% after update
> 
> **Home Assistant: 2023.7.0 Supervisor: 2023.07.1 HA addon Wmbusmeters: 1.14.0**
> 
> log
> 
> ```
> [19:37:13] INFO: Adding meter-0003 ...
> [19:37:14] INFO: Adding meter-0004 ...
> [19:37:14] INFO: Adding meter-0005 ...
> [19:37:14] INFO: Adding meter-0006 ...
> [19:37:14] INFO: Adding meter-0007 ...
> [19:37:15] INFO: Adding meter-0008 ...
> [19:37:15] INFO: Adding meter-0009 ...
> [19:37:15] INFO: Generating MQTT configuration ... 
> [19:37:16] INFO: Broker core-mosquitto will be used.
> [19:37:17] INFO: 
> MQTT Discovery ...
> tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
> [19:37:18] INFO:  Adding meter: Cold Water Meter ...
> [19:37:19] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00897110_rssi_dbm/config
> [19:37:19] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00897110_timestamp/config
> [19:37:20] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00897110_total_m3/config
> [19:37:20] INFO:  Adding meter: Hot Water Meter ...
> [19:37:20] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00817495_rssi_dbm/config
> [19:37:21] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00817495_timestamp/config
> [19:37:21] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00817495_total_m3/config
> [19:37:21] INFO:  Adding meter: Kitchen Heat Meter ...
> [19:37:22] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_current_hca/config
> [19:37:22] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_esb_date/config
> [19:37:22] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_previous_hca/config
> [19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_rssi_dbm/config
> [19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_season_start_date/config
> [19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_temp_room_avg_c/config
> [19:37:23] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_temp_room_prev_avg_c/config
> [19:37:24] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262770_timestamp/config
> [19:37:24] INFO:  Adding meter: Bedroom Heat Meter ...
> [19:37:24] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_current_hca/config
> [19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_esb_date/config
> [19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_previous_hca/config
> [19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_rssi_dbm/config
> [19:37:25] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_season_start_date/config
> [19:37:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_temp_room_avg_c/config
> [19:37:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_temp_room_prev_avg_c/config
> [19:37:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262845_timestamp/config
> [19:37:27] INFO:  Adding meter: Kidsroom Heat Meter ...
> [19:37:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_current_hca/config
> [19:37:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_esb_date/config
> [19:37:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_previous_hca/config
> [19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_rssi_dbm/config
> [19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_season_start_date/config
> [19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_temp_room_avg_c/config
> [19:37:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_temp_room_prev_avg_c/config
> [19:37:29] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35265048_timestamp/config
> [19:37:29] INFO:  Adding meter: LivingroomL Heat Meter ...
> [19:37:29] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_current_hca/config
> [19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_esb_date/config
> [19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_previous_hca/config
> [19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_rssi_dbm/config
> [19:37:30] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_season_start_date/config
> [19:37:31] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_temp_room_avg_c/config
> [19:37:31] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_temp_room_prev_avg_c/config
> [19:37:31] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262778_timestamp/config
> [19:37:32] INFO:  Adding meter: LivingroomR Heat Meter ...
> [19:37:32] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262835_current_hca/config
> [19:37:32] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262835_esb_date/config
> [19:37:33] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/35262835_previous_hca/config
[... cut ...] HTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
> ```
> 
>  ```
> c8a990ad-wmbusmeters-ha-addon:/wmbusmeters# ps axcf | grep rtl
>   708 root      0:00 /bin/sh -c /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
>   709 root      0:37 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
>   710 root     10:56 /usr/bin/rtl_wmbus -s
> ```

The same here but still on HA: 2023.5.4. After update the add-on to 1.14.0 I have not any new telegrams and data from my water meter.

BIBOLV, 2023-07-12:
@grest @oandry72 @koles142 - could you run in addon container `/usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - | /usr/bin/rtl_wmbus -s` but firstly stop existing processes that are running. As was mentioned above - to understand if actually there is something received on rtl_sdr

koles142, 2023-07-12:
How can I do that?

weetmuts, 2023-07-12:
Is the old docker container still running and blocking the rtlsdr?

BIBOLV, 2023-07-12:
Killing wmbusmeters process should do a trick. So find out wmbusmeters PID, do `kill -9 <pid>` on it and check if no rtl* processes are left running. Then run that command.

nug321, 2023-07-12:
I have the same issue.
I can ssh into wmbustermeter container, but when I kill any of the processes related to rtl, they are restored by the app.
If I kill the app, container is being destroyed...

`
/wmbusmeters # ps -a
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   14 root      0:00 {rc.init} /bin/sh -e /run/s6/basedir/scripts/rc.init top /run.sh
   15 root      0:00 s6-supervise s6-linux-init-shutdownd
   16 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdownd -c /run/s6/basedir -g 3000 -C -B
   24 root      0:00 s6-supervise s6rc-fdholder
   25 root      0:00 s6-supervise s6rc-oneshot-runner
   33 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/admin/s6/command/s6-ipcserver-access -v0 -E -l0 -i data/rules -- /package/admin/s6/command
   81 root      0:00 s6-supervise nginx
   84 root      0:00 nginx: master process nginx
  111 root      0:00 bash /usr/bin/bashio /run.sh
  204 root      0:00 python3 /flask/app.py
  206 root      0:01 /wmbusmeters/wmbusmeters --useconfig=/config/wmbusmeters
  237 nginx     0:00 nginx: worker process
  284 root      0:00 sh
 1298 root      0:00 sh
 1406 root      0:00 /bin/sh -c /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
 1407 root      0:00 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
 1408 root      0:01 /usr/bin/rtl_wmbus -s
 1421 root      0:00 ps -a
/wmbusmeters # kill -9 1406; kill -9 1407; kill -9 1408; /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - | /usr/bin/rtl_wmbus -s
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000000

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Sampling at 1600000 S/s.
Bandwidth set to automatic resulted in 0 Hz.
Tuned to 868.625000 MHz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
`

BIBOLV, 2023-07-12:
I was afraid it would happen. Ok, then try this - get container name by running `docker ps`, mark down container id of wmbusmeters addon, stop addon and then run `docker run -it --entrypoint /bin/sh <container-id>`, then you should have access to container without any processes started.

nug321, 2023-07-12:
I think once it is stopped, the image is destroyed..

```
➜  ~ docker run -it --entrypoint /bin/sh bc1c74b0059f
Unable to find image 'bc1c74b0059f:latest' locally
docker: Error response from daemon: pull access denied for bc1c74b0059f, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
➜  ~
```

oandry72, 2023-07-12:
Ok, I performed couple of tests.

Hardware:
Raspberry Pi 4 Model B Rev 1.2
Realtek Semiconductor Corp. RTL2838 DVB-T

Software:
Debian GNU/Linux 11 (bullseye)
Linux hassio 5.15.74-v8+ #1595 SMP PREEMPT Wed Oct 26 11:07:24 BST 2022 aarch64 GNU/Linux
Docker version 20.10.21, build baeda1f

Finally we have following:
1. If I use the docker image "wmbusmeters/wmbusmeters:latest" I receive telegrams.
2. If I use the docker image "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.14.0" I don't receive telegrams.

TEST 1
```
$ docker run -it --privileged --name=wmbusmeters --entrypoint /bin/sh -v /opt/wmbusmeters:/wmbusmeters_data -v /dev/:/dev/ -v /etc/localtime:/etc/localtime:ro wmbusmeters/wmbusmeters
/wmbusmeters # 
/wmbusmeters # /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - | /usr/bin/rtl_wmbus -s
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Sampling at 1600000 S/s.
Tuned to 868625000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
T1;1;1;2023-07-12 16:54:12.000;22;30;35280427;0x25441486270428350408b60afffff4450186c11d9b58a0a10000000000000000ec2e98130815
T1;1;1;2023-07-12 16:54:12.000;34;25;35280427;0x25441486270428350408b60afffff4450186c11d9b58a0a10000000000000000ec2e98130815
T1;1;1;2023-07-12 16:54:13.000;22;22;35280427;0x25441486270428350408b60afffff4450186c11d9b58a0a10000000000000000ec2e98130815
T1;1;1;2023-07-12 16:54:13.000;32;23;35280427;0x25441486270428350408b60afffff4450186c11d9b58a0a10000000000000000ec2e98130815
T1;1;1;2023-07-12 16:54:13.000;19;22;35280427;0x25441486270428350408b60afffff4450186c11d9b58a0a10000000000000000ec2e98130815
T1;1;1;2023-07-12 16:54:13.000;21;28;35280427;0x25441486270428350408b60afffff4450186c11d9b58a0a10000000000000000ec2e98130815
T1;0;0;2023-07-12 16:54:13.000;27;21;01178972;0x5a441486728917010507b60afffff4450106c11db6587ab9004085f48d77a9d8f0151b835c5ab3e40041fd02aab09247f30e020b5acac0672efe17c0724417ae3774ffd45cea2f929174ae343261e7176f34c36f2e33419acabbcc
T1;0;0;2023-07-12 16:54:13.000;19;35;01178972;0x5a441486728917010507b60afffff4450106c11db6587ab9004085f48d77a9d8f0151b835c5ab3e40041fd02aab09247f30e020b5acac0672efe17c0724417ae3774ffd45cea2f929174ae343261e7176f34c36f2e33419acabbcc
T1;0;0;2023-07-12 16:54:13.000;30;34;01178972;0x5a441486728917010507b60afffff4450106c11db6587ab9004085f48d77a9d8f0151b835c5ab3e40041fd02aab09247f30e020b5acac0672efe17c0724417ae3774ffd45cea2f929174ae343261ff176f34c36f2e33419acabbcc
```

TEST 2
```
$ docker run -it --privileged --name=wmbusmeters --entrypoint /bin/sh -v /opt/wmbusmeters:/wmbusmeters_data -v /dev/:/dev/ -v /etc/localtime:/etc/localtime:ro wmbusmeters/wmbusmeters-ha-addon-aarch64:1.14.0
/wmbusmeters # 
/wmbusmeters # /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - | /usr/bin/rtl_wmbus -s
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Sampling at 1600000 S/s.
Bandwidth set to automatic resulted in 0 Hz.
Tuned to 868.625000 MHz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers

^CSignal caught, exiting!

User cancel, exiting...
```

oandry72, 2023-07-12:
one more conclusion

1. If I use the docker image "wmbusmeters/wmbusmeters:latest" I receive telegrams.
2. If I use the docker image "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.13.1" I receive telegrams.
3. If I use the docker image "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.14.0" I stop receiving telegrams.
4. If I use any of docker images above I don't receive telegrams.

When I unplug DVB-T dongle and plug it again.
I can receive telegrams with images "wmbusmeters/wmbusmeters:latest" or "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.13.1"

grest, 2023-07-13:
> one more conclusion
> 
> 1. If I use the docker image "wmbusmeters/wmbusmeters:latest" I receive telegrams.
> 2. If I use the docker image "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.13.1" I receive telegrams.
> 3. If I use the docker image "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.14.0" I stop receiving telegrams.
> 4. If I use any of docker images above after using "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.14.0" I don't receive telegrams.
> 
> When I unplug DVB-T dongle and plug it again. I can receive telegrams with images "wmbusmeters/wmbusmeters:latest" or "wmbusmeters/wmbusmeters-ha-addon-aarch64:1.13.1"

@weetmuts, @BIBOLV  - I did the same test as @oandry72, including using  "wmbusmeters/wmbusmeters:latest" docker image and I got the same conclusions! It means that problem needs to be somewhere in the addon code. Perhaps something wrong with MQTT?

BIBOLV, 2023-07-13:
Thanks for thorough testing and detailed description, but based on [testing](https://github.com/wmbusmeters/wmbusmeters/issues/989#issuecomment-1632920555) where in TEST1 telegrams were received and in TEST2 were not, then it is related to RTL_SDR - either code or hardware, more over with [finding](https://github.com/wmbusmeters/wmbusmeters/issues/989#issuecomment-1632961729) that other version of container image will work only after dongle is plugged it points more to hardware issue, like wrong configuration is being applied when making connection to dongle.
I  did my tests on x86 arch so this might be related also to arm version of container only. 
Until I`m looking for possible causes of issue, please use edge channel that works currently 😄

grest, 2023-07-15:
Thanks. Hopefully problem will be resolved. Strange enough rtl-sdr repo did not change since December, so I'm wondering why previous version of wmbusmeters addon worked and the current doesn't.

oandry72, 2023-07-27:
Hi guys,

Yesterday I run into the issue that HA-Addon v 1.13.1 stoped working as well.
I use 1,5 meter USB cable to connect my DVB-T to RPi.
But when I plugged my the dongle to RPi directly without any cables the v1.13.1  started working again.
I decided to try v1.14.0, but no success.

Than I started playing with different USB ports on my RasperryPi and once I received few telegrams with v1.14.0 when I plugged my DVB-T dongle to one of USB3 ports.
I was waiting for more telegrams more than 10 mins without luck.

```
$ docker run -it --privileged --name=wmbusmeters14 --entrypoint /bin/sh -v /opt/wmbusmeters:/wmbusmeters_data -v /dev/:/dev/ -v /etc/localtime:/etc/localtime:ro wmbusmeters/wmbusmeters-ha-addon-aarch64:1.14.0
/wmbusmeters # /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - | /usr/bin/rtl_wmbus -s
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Fitipower FC0012 tuner
Sampling at 1600000 S/s.
Bandwidth set to automatic resulted in 0 Hz.
Tuned to 868.625000 MHz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
T1;0;0;2023-07-27 09:08:09.000;28;17;00819576;0x5a441486769581000507b60afffff8450106321da45d7ac500408509f11c6e3986087ddb2a04ab23bf58ca8f3c1c28272e3ca2ffffffffffffff98ffffff65ffffffffffff72ffffffffffffffff30ffffffffffffff9effffffff
T1;1;1;2023-07-27 09:08:09.000;26;16;00819576;0x5a441486769581000507b60afffff8450106321da45d7ac500408509f11c6e3986087ddb2a04ab23bf58ca8f3c1c28272e3ca2726ffd2a2ff8193536eb88ceba316cd847f8a4da0d8fcc94d10e120837327267dbae9739ad2521a9
T1;0;0;2023-07-27 09:08:09.000;44;39;00819576;0x5a441486769581000507b60afffff3450106af3da2587ac500408509f11c6e3986087ddbfffffffffffff1ffffffff4affffffa4ffffffffffff0effff55f9ff06ffffff75ffffff57ffffffff64ff0a04a4ffff19ffffff4e46ff
T1;1;1;2023-07-27 09:08:09.000;52;56;00819576;0x5a441486769581000507b60afffff3450106af3da2587ac500408509f11c6e3986087ddb2a04ab23bf58ca8f3c1c28272e3ca2726ffd2a2ff8193536eb88ceba316cd847f8a4da0d8fcc94d10e120837327267dbae9739ad2521a9
T1;0;0;2023-07-27 09:08:09.000;31;17;00819576;0x5a441486769581000507b60afffff3450106af3da2587ac500408509f11c6e3986087ddb2a04ab23bf58ca8f3c1c28272e3ca2726ffd2a2ff8193536eb88ceba316cd847f8a4da0d8fcc94d10e120837327267dbae9739ad2521a9
T1;1;1;2023-07-27 09:08:09.000;27;39;00819576;0x5a441486769581000507b60afffff3450106af3da2587ac500408509f11c6e3986087ddb2a04ab23bf58ca8f3c1c28272e3ca2726ffd2a2ff8193536eb88ceba316cd847f8a4da0d8fcc94d10e120837327267dbae9739ad2521a9
T1;1;1;2023-07-27 09:08:09.000;13;12;00819576;0x5e441486769581000507b60efffff7450106321da45daf3dac7b7ac500408509f11c6e3986087ddb2a04ab23bf58ca8f3c1c28272e3ca2726ffd2a2ff8193536eb88ceba316cd847f8a4da0d8fcc94d10e120837327267dbae9739ad2521a9
T1;1;1;2023-07-27 09:08:09.000;16;10;00819576;0x5e441486769581000507b60efffff7450106321da45daf3dac7b7ac500408509f11c6e3986087ddb2a04ab23bf58ca8f3c1c28272e3ca2726ffd2a2ff8193536eb88ceba316cd847f8a4da0d8fcc94d10e120837327267dbae9739ad2521a9

^CSignal caught, exiting!

User cancel, exiting...
Reattached kernel driver
```

All other attempts get telegrams with v1.14.0 and changing USB ports were unsuccessful.
I used the docker image wmbusmeters/wmbusmeters-ha-addon-aarch64:1.14.0.

code4me2all, 2023-07-29:
HI, 
install the addon on a fresh installed Rpi 4. Same here the rtlsdr won't work, no telegrams receiving. 

On my debian linux i having following working setting in wmbusmeter.conf for the device
`device=rtlwmbus:CMD(rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus -p s -a)`
So i tried to use the different CMD config from the Rpi debug output on my linux system
>(rtlwmbus) using command: /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
changed on my linux to
`device=rtlwmbus:CMD(rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | rtl_wmbus  -s)`
and also not working.
After some trying i found out the frequence must be  set to 868.95M for receiving telegrams and also
`|  rtl_wmbus  -s` 
won't work for me , only with
`rtl_wmbus -p s -a`

How can i change the **wmbusmeters HA addon configuration**  settings to reflect to my changes?

Adranus, 2023-08-03:
Hello there,

I just want to report the same issue. Just before the newest version dropped (v1.14.0) everything was working. My last telegram received was '2023-07-02T22:34:22Z'. Since then the add-on didn't receive or log any telegrams. I'm using HomeAssistant OS in a VM on proxmox. So no docker for me. After reinstalling and testing the hardware im unable to receive anything. First i was concerned because i did hardware upgrades of my system (from i3 to i7 cpu and RAM upgrade). But there is no issue there. Is there a possibility to add a repo with older version for me to test ?

greinet, 2023-08-04:
For me version [1.13.1-45](https://github.com/wmbusmeters/wmbusmeters-ha-addon/tree/42ed19ee01943bf7fea8d947728c90a9a50b035e) was also the last working version of the addon.
The[ next commit aka the update to 1.14.0](https://github.com/wmbusmeters/wmbusmeters-ha-addon/tree/6468a849fec18b22b9b92e9e21082b577d559136) breaks the plugin and i can't receive telegrams anymore.

@Adranus To install an older version of a Home Assistant addon, simply download the code as zip ([e.g. version 1.13.1-45](https://github.com/wmbusmeters/wmbusmeters-ha-addon/archive/42ed19ee01943bf7fea8d947728c90a9a50b035e.zip)) and upload that zip file to your Home Assistant e.g. via the VS Code addon. Then move it to the /addon directory and unzip. After that a new section called "Local add-ons" will appear in the addon store from where you can install your local version. I would suggest removing the official repository first from the addon store.

Krotu, 2023-08-05:
broken for me too since last update, i am running home assistant os on an rpi4, i am not seeing any errors but none of my meter readings are being recorded. If I restore a backup it works and then it stops again if I update.

Adranus, 2023-08-06:
@greinet thank you. I did just as you suggested using VS code uploading a local .zip file. Used the version 1.13.1 and unfortunately this version is also not working for me.... I also noticed quite high CPU usage (25%). I have allocated 4 cores from an i7-4765t.  I don't know what files does this addon creates but maybe there are some undeleted system files preventing any other versions to work ? I really don't want to delete and reinstall the whole HA virtual machine. Can I delete just the residual files ? Where can I find them ? Also maybe some requirement rtl_sdr module needs to be deleted or reinstalled. If there is something like that where can I find it ?

thecem, 2023-08-06:
Did you was able to start the addon? If not you have to wait until the docker build is finished.

try the latest edge version.

what is your config?

grest, 2023-08-06:
> @greinet thank you. I did just as you suggested using VS code uploading a local .zip file. Used the version 1.13.1 and unfortunately this version is also not working for me.... I also noticed quite [...]

Have you unplugged and plugged again the reader from the usb port, after the downgrade?

nug321, 2023-08-06:
> For me version [1.13.1-45](https://github.com/wmbusmeters/wmbusmeters-ha-addon/tree/42ed19ee01943bf7fea8d947728c90a9a50b035e) was also the last working version of the addon. The[ next commit aka the [...]

Thx, This is working for me. HA on VM.

VoyteckPL, 2023-08-07:
Where to create this folder "addon"? I did this in main HA folder and I don't see lacal addons

VoyteckPL, 2023-08-10:
For me, latest edge is not receiving telegram.
On 1.13.1 -45 I receive telegrams but with errors in log:
[image]
Although it is read by MQTT and I can see all the sensors live.

BIBOLV, 2023-08-10:
Do you have a spare sd where you could install just clean ha and wmbusmeters addon and see if any telegrams are being received?

code4me2all, 2023-08-10:
Full updating home assistant and reinstall the addon (edge), it works with driver hydros!
(Running MQTT server is required)
.
After reinstall i did:
1.) removed the Meter config accessable with the sidebar  and  after save(restart) i see the telegrams.
2.) add Meter and set log to true to see if all works ok (watch log file).
3.)  Modify yaml file, reload yaml and add to Energie dashboard and set log back to false. 
Only one think is missing, is the mqtt-discovery but i think i should open another request ticket...

Adranus, 2023-08-10:
So I was playing with the addon today and maybe hound something. I was playing with the version 1.13.1 which was reported in this thread to work. But for me it didn't. So because I have a proxmox machine I created a fresh VM with HA OS and installed the addon. And even on a fresh install it was not working. So I played with various settings with no luck. After a good hour of tinkering with it I accepted loss and deleted the test VM. After that I returned to my original HA OS instance to setup other automations and I said to my self that I will try for the last time and start the addon. And surprise surprise IT WAS WORKING !! Okay interesting now it works. I didn't do anything all the time this VM was offline. Then with hesitation I wanted to test the new version 1.14.0 .  Maybe it will work now right ? Nope didn't work.... So I downgraded to the 1.31.1 only to find out that is AGAIN NOT WORKING !!! So now its personal and I tested various combinations of Instance restarts, full proxmox restarts, disconnecting and reconecting the dongle while ON and while OFF etc.... And I got it to work again on the 1.13.1. I did debug logging so I immediately looked in to the differences in logs when working and when not. This is the only difference : 

The working instance has this : 

[22:53:18] INFO: Broker core-mosquitto will be used.
[22:53:19] INFO: 
MQTT Discovery ...
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
[22:53:20] INFO: MQTT Discovery cleanup...

The not working instance has this : 

[22:20:37] INFO: Broker core-mosquitto will be used.
[22:20:37] INFO: 
MQTT Discovery ...
tar: invalid magic
tar: short read
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
[22:20:43] INFO: MQTT Discovery cleanup...

So I guess magic didn't happen at that time ? How do I make the magic happen all the time ? I don't want to do the restart reconnect procedure every time I restart the machine... Is my PC low on mana or something ? Hope it helps.

BIBOLV, 2023-08-11:
Ah, valiant troubleshooter of the arcane arts, you have uncovered the elusive source of the enchantment's hiccup! It appears that with each awakening of the mystical machine, the latest meter templates embark on a grand adventure of [downloading](https://github.com/wmbusmeters/wmbusmeters-ha-addon/blob/main/wmbusmeters-ha-addon-edge/mqtt_discovery.sh#L29). Ah, but behold the twist in fate, for the link itself was but a mirage, corrected only in the sacred scrolls of version [1.14.0-5-2](https://github.com/wmbusmeters/wmbusmeters-ha-addon/commit/dd3a2011ffda3d8e89751507aadfe1b539576697).

Though I, too, am puzzled by the connection between this ethereal dance of files and the functionality of the RTL-SDR dongle, perhaps it is a cosmic reminder that even the tiniest pebble can ripple a vast pond. The remedy, it seems, rests in altering the ritual – the templates must no longer journey from distant lands, but rather flourish from local grounds. Thus, the dance of constant downloads shall cease, and your incantations shall evoke a performance unmarred by such trivial enchantments.

BIBOLV, 2023-08-14:
I made changes in edge version, so if someone was on edge channel and had issues - it would be nice if newer version could be validated.

VoyteckPL, 2023-08-16:
I tried on a fresh install... same issue:
[image]

grest, 2023-08-19:
Not sure how it can be overwritten.  

I just installed edge, reused my configuration (you can find it on the top of this discussion) and  run addon - no improvement unfortunately.

drschlaumeier, 2023-08-19:
Same for me new version is not working. I went back to 1.13.1-45 as proposed here https://github.com/wmbusmeters/wmbusmeters/issues/989#issuecomment-1665846564
That rocks :-)

grest, 2023-08-23:
Perhaps I'm naive but is it possible to generate diff, for at least the core libraries and dependencies between 1.13 and 1.14 to analyse and find the issue? How the versions of those dependencies changed over these releases?

weetmuts, 2023-08-23:
@grest >Not sure how it can be overwritten.

you write the cli command to capture telegrams using rtl_wmbus and rtl_sdr like this in the config:

device=rtlwmbus:CMD(rtl_sdr .... | rtl_wmbus...)

chpego, 2023-08-26:
same problem for me, switching to version 1.14 of the addon on HA no longer returns a value; in addition, CPU usage has been increased to 100%.

Krotu, 2023-08-26:
I have had success disconnecting/reconnect my usb DVB-T device after an update, restarting HA and rebooting my rpi do not work, maybe because it stays powered? It only stops working after an update.

weetmuts, 2023-08-26:
@Krotu just to be clear, you are saying you can run version 1.14 if you disconnect/reconnect the usb DVB-T?

Krotu, 2023-08-26:
yes, I get a sharp spike in readings (catching up from when the readings stopped) a few minutes after the replug and then it carries on working after that. I am running 1.14 at the moment.

chpego, 2023-08-27:
I may have found something interesting, but it certainly has nothing to do with the version change; I'll explain what I observed anyway.
if I want to change the device option in the following configuration
[image]
and I add the frequency (for example :868.25M, or using CMD(cmd), then I no longer receive data via MQTT. In MQTT, there's a record based on the device name, in my case rtlwmbus[777xxxxxxxx00].

Could this be related?

twitch86, 2023-09-09:
Any updates on this? Wanted to add my Sensus Iperl to HA with the USB Stick iM871A.
But it does not catch anything. USB 2.0 Port in my Dell Server - 30cm next to the Watermeter. 

[image]

[41 more replies]

closed 2024-02-14 by grest