wmbusmeters/wmbusmeters #990: Qalcosonic W1 - no Driver Support?

HerrICMPPing, 2023-07-05:
### Type of request

Existing device improvement

### Meter description

Hello everyone, as I can see my water meter supported "axioma qalcosonic w1" is the yellow one. I've also seen that there is also a q400:t1 driver for this, which I've tried without success. If I specify the driver manually it doesn't work and with auto it doesn't find a driver.

I have already requested an AES key from my municipal utility, let's see if I can get it

### Logged telegrams from meter in decrypted format

```shell
-----------------------------------------------------------------------
wmbusmeters auto:t1
-----------------------------------------------------------------------
Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 06178691
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07) encrypted
                   ver: 0x01
                device: rtlwmbus[00000001]
                  rssi: 146 dBm
                driver: unknown!
-----------------------------------------------------------------------              
wmbusmeters --verbose --debug auto:t1
-----------------------------------------------------------------------
(wmbusmeters) version: 1.14.0
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 691 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 693 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(serial) EVENT thread interrupted
(main) opening rtlwmbus
Started auto rtlwmbus[00000001] listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 699 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) received ascii "T1;1;1;2023-07-05 07:11:53.000;137;139;06178691;0x434409079186170601078c2029900f002c2529bc0000c54008ad52136cac7a29002007100f96dae5b41683c9c3f942170aff29e60182a9a95b58ca9ac3381509717f560d<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-07-05 07:11:53.000;137;139;06178691;0x434409079186170601078c2029900f002c2529bc0000c54008ad52136cac7a29002007100f96dae5b41683c9c3f942170aff29e60182a9a95b58ca9ac3381509717f560d<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 68
(wmbus) parseELL @10 58
(wmbus) parseNWL @13 55
(wmbus) parseAFL @13 55
(wmbus) parseTPL @30 38
(wmbus) input to kdf for enc "0029BC00009186170607070707070707"
(wmbus) no key, thus cannot execute kdf.
(wmbus) parseDLL @0 68
(telegram) DLL L=43 C=44 (from meter SND_NR) M=0709 (AXI) A=06178691 VER=01 TYPE=07 (Water meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=137
(wmbus) parseELL @10 58
(telegram) ELL CI=8c CC=20 (slow_resp sync) ACC=29
(wmbus) parseNWL @13 55
(wmbus) parseAFL @13 55
(telegram) AFL CI=90
(wmbus) parseTPL @30 38
(wmbus) input to kdf for enc "0029BC00009186170607070707070707"
(wmbus) no key, thus cannot execute kdf.
Received telegram from: 06178691
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07) encrypted
                   ver: 0x01
                device: rtlwmbus[00000001]
                  rssi: 137 dBm
                driver: unknown!
(wmbus) 000   : 43 length (67 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 0907 dll-mfct (AXI)
(wmbus) 004   : 91861706 dll-id (06178691)
(wmbus) 008   : 01 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : 29 ell-acc
(wmbus) 013   : 90 afl-ci-field (AFL: Authentication and Fragmentation Sublayer)
(wmbus) 014   : 0f afl-len (15)
(wmbus) 015   : 002c afl-fc (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)
(wmbus) 017   : 25 afl-mcl (AES_CMAC_128_8 MessCounter)
(wmbus) 018   : 29bc0000 afl-counter (48169)
(wmbus) 022   : C54008AD52136CAC afl-mac 8 bytes
(wmbus) 030   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 031   : 29 tpl-acc-field
(wmbus) 032   : 00 tpl-sts-field (OK)
(wmbus) 033   : 2007 tpl-cfg 0720 (AES_CBC_NO_IV )
(wmbus) 035   : 10 tpl-cfg-ext (KDFS=1)
(wmbus) telegram from 06178691 ignored by all configured meters!
(serial) received ascii "T1;1;1;2023-07-05 07:11:53.000;127;141;06178691;0x434409079186170601078c2029900f002c2529bc0000c54008ad52136cac7a29002007100f96dae5b41683c9c3f942170aff29e60182a9a95b58ca9ac3381509717f560d<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-07-05 07:11:53.000;127;141;06178691;0x434409079186170601078c2029900f002c2529bc0000c54008ad52136cac7a29002007100f96dae5b41683c9c3f942170aff29e60182a9a95b58ca9ac3381509717f560d<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=68.
```

HerrICMPPing, 2023-07-05:
-----------------------------------------------------------------------
wmbusmeters --logtelegrams --format=json rtlwmbus:868.95M:t1 MyWater auto '06178691' NOKEY
-----------------------------------------------------------------------
```
Started config rtlwmbus[00000001] listening on t1 using fq 868.95M
(meter) MyWater: meter detection could not find driver for id: 06178691 mfct: (AXI) UAB Axis Industries, Lithuania (0x709) type: Water meter (0x07) ver: 0x01
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (MyWater 06178691 auto) did not handle telegram!
```

HerrICMPPing, 2023-07-05:
I now think "telegram=|" have what you always need when it comes to support.

--------------------------------------------------------------
wmbusmeters --verbose --logtelegrams --debug auto:t1
--------------------------------------------------------------
(wmbusmeters) version: 1.14.0
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 2042 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 2044 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(serial) EVENT thread interrupted
(main) opening rtlwmbus
Started auto rtlwmbus[00000001] listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 2050 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) received ascii "T1;1;1;2023-07-05 08:01:45.000;137;135;06178691;0x434409079186170601078c20d4900f002c25d4bc0000d9044b4d98ef90567ad4002007108fe73487b2bc1fa4c67caae13a32ebf0f2a2f5a75dee0bda25c37e5c85df448a<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-07-05 08:01:45.000;137;135;06178691;0x434409079186170601078c20d4900f002c25d4bc0000d9044b4d98ef90567ad4002007108fe73487b2bc1fa4c67caae13a32ebf0f2a2f5a75dee0bda25c37e5c85df448a<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 68
(wmbus) parseELL @10 58
(wmbus) parseNWL @13 55
(wmbus) parseAFL @13 55
(wmbus) parseTPL @30 38
(wmbus) input to kdf for enc "00D4BC00009186170607070707070707"
(wmbus) no key, thus cannot execute kdf.
(wmbus) parseDLL @0 68
(telegram) DLL L=43 C=44 (from meter SND_NR) M=0709 (AXI) A=06178691 VER=01 TYPE=07 (Water meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=137
(wmbus) parseELL @10 58
(telegram) ELL CI=8c CC=20 (slow_resp sync) ACC=d4
(wmbus) parseNWL @13 55
(wmbus) parseAFL @13 55
(telegram) AFL CI=90
(wmbus) parseTPL @30 38
(wmbus) input to kdf for enc "00D4BC00009186170607070707070707"
[... 171 lines cut ...]
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 0907 dll-mfct (AXI)
(wmbus) 004   : 91861706 dll-id (06178691)
(wmbus) 008   : 01 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : d7 ell-acc
(wmbus) 013   : 90 afl-ci-field (AFL: Authentication and Fragmentation Sublayer)
(wmbus) 014   : 0f afl-len (15)
(wmbus) 015   : 002c afl-fc (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)
(wmbus) 017   : 25 afl-mcl (AES_CMAC_128_8 MessCounter)
(wmbus) 018   : d7bc0000 afl-counter (48343)
(wmbus) 022   : A978CE5382115BAE afl-mac 8 bytes
(wmbus) 030   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 031   : d7 tpl-acc-field
(wmbus) 032   : 00 tpl-sts-field (OK)
(wmbus) 033   : 2007 tpl-cfg 0720 (AES_CBC_NO_IV )
(wmbus) 035   : 10 tpl-cfg-ext (KDFS=1)
telegram=|_434409079186170601078C20D7900F002C25D7BC0000A978CE5382115BAE7AD70020071053FF3837160E62027293F8EF94D9190219AE2BBF0D8D7E27ED43807846D17F29|+61
(wmbus) telegram from 06178691 ignored by all configured meters!
(serial) received ascii "T1;1;1;2023-07-05 08:02:38.000;117;138;06178691;0x434409079186170601078c20d7900f002c25d7bc0000a978ce5382115bae7ad70020071053ff3837160e62027293f8ef94d9190219ae2bbf0d8d7e27ed43807846d17f29<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-07-05 08:02:38.000;117;138;06178691;0x434409079186170601078c20d7900f002c25d7bc0000a978ce5382115bae7ad70020071053ff3837160e62027293f8ef94d9190219ae2bbf0d8d7e27ed43807846d17f29<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=68.

BIBOLV, 2023-07-06:
Hi,
Firstly you would need to get a key in order to decrypt telegrams. If then they are still not recognized then post few telegrams - only lines that start with "telegram=|" are needed.

HerrICMPPing, 2023-07-06:
> Hi, Firstly you would need to get a key in order to decrypt telegrams. If then they are still not recognized then post few telegrams - only lines that start with "telegram=|" are needed.

Okay thanks. So far I haven't got an answer about my AES key, I'll keep you posted. However, I actually thought that the driver should still be recognized? Or is this only recognized if you specify a decrypt key?

BIBOLV, 2023-07-09:
When telegram is decrypted then it is being parsed and decoded based on created driver.

closed 2023-10-08 by github-actions[bot]