wmbusmeters/wmbusmeters #994: “Analyze” Button on https://wmbusmeters.org/ does nothing

jonas240, 2023-07-15:
### Type of request

Bug

### OS version

-

### wmbusmeters version

-

### Your message goes here

Hi,

I am trying to analyze some telegrams using [https://wmbusmeters.org/]. However, upon pressing the button “Analyze” nothing happens. Using a specific website like https://wmbusmeters.org/analyze/244465323819801318077A7D000000_8C04136242110082046CFE2601FD0C11046D070EEF27 does work. But this uses the default “auto” driver. Changing the driver in the dropdown menu and then pressing “Analyze” does again nothing. I tried using Firefox and Edge.

I greatly appreciate your help.

weetmuts, 2023-07-15:
Just tried, it works for me. :-) Both in Firefox and Edge. Do you have javascript enabled?
You can also suffix the url wth :driver eg. see below where it will use the c5isf driver.

https://wmbusmeters.org/analyze/244465323819801318077a7d0000008c04136242110082046cfe2601fd0c11046d070eef27:c5isf

jonas240, 2023-07-15:
The problem was caused by the browser extension uBlock Origin. Disabling uBlock Origin temporally solved the issue.

closed 2023-07-15 by jonas240

weetmuts, 2023-07-15:
Ok. Something is wrong with uBlock Origin since the only thing wmbusmeters.org does is use javascript to compose a new url to its own domain and jump there. Oh well, good that you figured it out.

jonas240, 2023-07-15:
yea, I opend an issue with uBlock Origin on that