wmbusmeters/wmbusmeters #995: ha-addon docker bash: wmbusmeters: command not found

jonas240, 2023-07-15:
### Type of request

Question

### OS version

HomeAssistant OS 10.3 (RasperryPi4)

### wmbusmeters version

1.14.0

### Your message goes here

Hi,
I tried to use the command 'wmbusmeters 32446850399721226980A0019F2D310FF02E1E036B0971090100010002' in the command line of the docker container (using the portainer add on) of the ha-add-on to decode a telegram I received. But that returned 'wmbusmeters: command not found'.

How can I decode a telegram using the command line when using the ha-addon

I greatly appreciate your help.

BIBOLV, 2023-07-15:
Hi,
Try `/wmbusmeters/wmbusmeters --analyze 32446850399721226980A0019F2D310FF02E1E036B0971090100010002`

jonas240, 2023-07-15:
Thanks that solved it

closed 2023-07-15 by jonas240