wmbusmeters/wmbusmeters #998: (IMT) INTEGRA METERING TOPAS SONIC encryption key

gh0stface33, 2023-07-20:
Hi, anyone has encryption key for my water meter ?

Integra metering topas sonic

I ask my water provider and the answer was they have not any key 

Received telegram from: 10107821
          manufacturer: (IMT) INTEGRA METERING, Switzerland (0x25b4)
                  type: Water meter (0x07) encrypted
                   ver: 0x10
                device: rtlwmbus[00000001]
                  rssi: 81 dBm
                driver: unknown!
(wmbus) 000   : 4e length (78 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : b425 dll-mfct (IMT)
(wmbus) 004   : 21781010 dll-id (10107821)
(wmbus) 008   : 10 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : a2 tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 4025 tpl-cfg 2540 (synchronous AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
(wmbus) 015 CE: B60E046A81EFE16EE6E3DADE86367E2F0C1099CC0FBECF329EB7D7E9C9149CF81AF7DADB22FACAF1E563AEAA257A1671078AC91E49F54A7BE1FB86338A60D222 encrypted
telegram=|_4E44B4252178101010077AA2004025B60E046A81EFE16EE6E3DADE86367E2F0C1099CC0FBECF329EB7D7E9C9149CF81AF7DADB22FACAF1E563AEAA257A1671078AC91E49F54A7BE1FB86338A60D222|+20
(wmbus) telegram from 10107821 ignored by all configured meters!
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-07-20 17:02:02.000;74;84;10107821;0x4e44b4252178101010077aa2004025b60e046a81efe16ee6e3dade86367e2f0c1099cc0fbecf329eb7d7e9c9149cf81af7dadb22facaf1e563aeaa257a1671078ac91e49f54a7be1fb86338a60d222<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=79.
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) EVENT thread interrupted
(shell) sending SIGTERM to process group 456214
(serial) TIMER thread interrupted
(serial) event loop stopped!
(bgshell) sent sigterm, now waiting for child 456214 to exit.
(bgshell) 456214 terminated due to signal 15
(wmbus) disconnected 00000001 rtlwmbus
(serialcmd) closed /bin/sh pid=456214 fd=5 (rtlwmbus)
(wmbus) deleted rtlwmbus
(serial) closing 1 devices

weetmuts, 2023-07-21:
It definitely looks encrypted. 

Perhaps:

1) the person on the water provider side did not understand the question

2) the water provider has signed a bad contract with the water meter manufacturer
where the water provider does not actually have they keys! They have to use computer/software
hardware from the meter manufacturer to decode the telegrams. Yes this happens!
https://gisbarbados.gov.bb/blog/effort-to-obtain-codes-for-smart-water-meters/

3) it is a default key for all meters, that they do not want to reveal.

4) it is a default key for all meters, that they could reveal but they do not know it.
I tried an all zeroes key 00000000000000000000000000000000 but that did not work.

closed 2023-10-20 by github-actions[bot]

timrosu, 2024-07-07:
From where are you and which water provider do you have?
I'm from Slovenia and Komunala Kranj has the same meters.

gh0stface33, 2024-07-07:
@timrosu 

i’m from France and my water provider is local.

timrosu, 2024-07-07:
Have you managed to aquire decription key?

gh0stface33, 2024-07-07:
@timrosu 
I have already ask the key but my water provider don’t want give me the key.

timrosu, 2024-07-07:
@gh0stface33 
That's a shame. I'll call them tomorrow.

Offenbacher10, 2025-12-29:
did anybody get a decryption key for the topas sonic water meter from their provider? if so how, with which arguments? My provider just says "data privacy" :/ which is nonsens because this data is created by me (my water consumption). its ridiculous.