wmbusmeters/wmbusmeters #999: Meter detection could not find driver

philipdzierzon, 2023-07-20:
### Type of request

New device support

### Meter description

mfct: (ITW) ITRON Water, France (0x2697) 
type: Water meter (0x07) 
ver: 0x00

### Logged telegrams from meter in decrypted format

```shell
384497265909312000077A930000A0_041360B50100066D101295F427004413AC570100426CDF2C047F0000060C027F6C2A0E79000000000000
384497269704312000077AAC0000A0_0413D7AF0000066D280B95F42700441329910000426CDF2C047F0000060C027F6C2A0E79000000000000
384497265909312000077A940000A0_041360B50100066D0F1795F427004413AC570100426CDF2C047F0000060C027F6C2A0E79000000000000
384497269704312000077AAD0000A0_0413D7AF0000066D281095F42700441329910000426CDF2C047F0000060C027F6C2A0E79000000000000
384497265909312000077A950000A0_041360B50100066D0E1C95F427004413AC570100426CDF2C047F0000060C027F6C2A0E79000000000000
384497269704312000077AAE0000A0_0413D7AF0000066D281595F42700441329910000426CDF2C047F0000060C027F6C2A0E79000000000000
384497265909312000077A960000A0_041360B50100066D102195F427004413AC570100426CDF2C047F0000060C027F6C2A0E79000000000000
384497269704312000077AAF0000A0_0413D7AF0000066D281A95F42700441329910000426CDF2C047F0000060C027F6C2A0E79000000000000
384497265909312000077A970000A0_041360B50100066D0E2695F427004413AC570100426CDF2C047F0000060C027F6C2A0E79000000000000
384497269704312000077AB00000A0_0413D7AF0000066D281F95F42700441329910000426CDF2C047F0000060C027F6C2A0E79000000000000
384497265909312000077A980000A0_041360B50100066D0F2B95F427004413AC570100426CDF2C047F0000060C027F6C2A0E79000000000000
```

weetmuts, 2023-07-21:
Thank you! It seems like the itron driver supported all fields already, I only had to add the new version to the auto-detector. https://wmbusmeters.org/analyze/384497265909312000077A930000A0041360B50100066D101295F427004413AC570100426CDF2C047F0000060C027F6C2A0E79000000000000

Do you have any input on what the two manufacturer specific fields (that currently translate to WOOT) can mean?

weetmuts, 2023-07-21:
Fix pushed to github main. https://github.com/wmbusmeters/wmbusmeters/commit/217f0a25fece05474a1b20ad3bed88eece3930ae

philipdzierzon, 2023-07-21:
I have no idea what those fields mean. I had a look at the manufacturers publicly available documentation and could not find anything about these.
This model is installed: https://www.itron.com/de/solutions/product-catalog/equascan-wmiu-rf

I am currently using wmbusmeters on my raspberry installed with snap. Is the new version available there already? It is my first time using snap.

philipdzierzon, 2023-07-24:
I switched to the edge channel and the changes work as expected. Thank you!

closed 2023-07-24 by philipdzierzon