wmbusmeters/wmbusmeters #1000: Meter detection could not find driver

philipdzierzon, 2023-07-22:
### Type of request

New device support

### Meter description

mfct: (ITW) ITRON Water, France (0x2697) 
type: Heat meter (0x04) 
ver: 0x00

### Logged telegrams from meter in decrypted format

```shell
404497262900132000047A0D0000A0_04063D1200000C1439640400066D2A2FCEF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A0E0000A0_04063D1200000C1439640400066D2934CEF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A0F0000A0_04063D1200000C1439640400066D2A39CEF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A100000A0_04063D1200000C1439640400066D2A02CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A110000A0_04063D1200000C1439640400066D2907CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A120000A0_04063D1200000C1439640400066D290CCFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A130000A0_04063D1200000C1439640400066D2911CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A140000A0_04063D1200000C1439640400066D2916CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A150000A0_04063D1200000C1439640400066D291BCFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A160000A0_04063D1200000C1439640400066D2A20CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A170000A0_04063D1200000C1439640400066D2925CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A180000A0_04063D1200000C1439640400066D292ACFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A190000A0_04063D1200000C1439640400066D282FCFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A1A0000A0_04063D1200000C1439640400066D2934CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
404497262900132000047A1B0000A0_04063D1200000C1439640400066D2939CFF627004406450F0000426CDF2C0D7FEB0200000006040C626F04370C7956302520
```

BIBOLV, 2023-07-23:
Some values are decoded with microclima driver, could you check if they are correct and?
```
{
    "media":"heat",
    "meter":"microclima",
    "name":"",
    "id":"20130029",
    "meter_datetime":"2023-07-22 14:47:42",
    "status":"OK",
    "total_energy_consumption_kwh":4669,
    "total_volume_m3":464.39,
    "consumption_at_set_date_1_kwh":3909,
    "set_date_1_date":"2022-12-31",
    "timestamp":"2023-07-23T18:51:24Z"
}
```

philipdzierzon, 2023-07-24:
Thanks for the reply!

It's an itron EquaScan h MIU RF. 
I have no way to check `total_volume_m3`. The other values are correct. `timestamp` is not part of the dataset but the time of analysing it, right? 
I sadly have no information about the remaining uninterpreted data.

closed 2023-10-23 by github-actions[bot]