wmbusmeters/wmbusmeters #1002: Driver unknown for sagem device

tonyGTT, 2023-07-23:
### Type of request

New device support

### Meter description

Hello,
I would like to read my water meter. The device comes from SagemCom, SICONIA SM15-M. The driver is unknown.
Thank you for your support.
BR

### Logged telegrams from meter in decrypted format

```shell
toto@toto-VivoBook-ASUSLaptop-X515JAB-F515JA:~$ wmbusmeters auto:c1 MyTapeWater auto 00087346 527020020000D9A7
Started auto rtlwmbus[00000001] listening on c1
(meter) MyTapeWater: meter detection could not find driver for id: 00087346 mfct: (SET) Sagemcom Energy Telecom, France (0x4cb4) type: Water meter (0x07) ver: 0x00
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(wmbus) Unknown tpl-ci-field 7b
(meter) newly created meter (MyTapeWater 00087346 auto) did not handle telegram!
```

BIBOLV, 2023-07-23:
Hi,
Please follow documentation and provide decrypted telegrams:

> Logged telegrams from meter in decrypted format
In order to log telegrams, set logtelegrams=true in configuration and gather at least 10 telegrams from meter. Check that telegrams are not encrypted, it can be verified in logs or by pasting them [here](https://wmbusmeters.org/). If telegram is encrypted you need to get encryption key, decrypt messages and them post them as we can not do anything with encrypted ones.

tonyGTT, 2023-07-25:
Hello,

Thank you for your response. The telegram seems not encrypted. I don't know which driver is used.

BR

4044B44c6675060000077B32001c2a0003cef804007B0d11e7911160828d45052cB0300d481a346f57906743d5f12d34fBf3a7B1e2a3e397cc9c26d001d51BB9e6

Auto driver  : not found!
Best driver  : unknown 00/00
Using driver : unknown 00/00
000   : 40 length (64 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b44c dll-mfct (SET)
004   : 66750600 dll-id (00067566)
008   : 00 dll-version
009   : 07 dll-type (Water meter)
010  ?: 7b unknown ci-field

{
    "media":"water",
    "meter":"unknown",
    "name":"",
    "id":"00067566",
    "timestamp":"1970-01-01T00:00:00Z"
}

tonyGTT, 2023-09-18:
Hi,

the Sagem products seem to be encrypted AES, so I have not found a solution, for now.

BR

closed 2023-12-18 by github-actions[bot]

Samqpa, 2025-02-27:
Hi,

Is it possible to read SagemCom Siconia WM20-L water meter?