wmbusmeters/wmbusmeters #1008: MQTT Sensor naming changed in HA 2023.8

killer0071234, 2023-08-03:
### Type of request

Question

### HAOS version

10.3

### wmbusmeters version

1.14.0

### Your message goes here

Starting with HA 2023.8.0 there are new naming conventions for (MQTT) entities.
Currently every entity is named as following:
`DeviceName SensorName`

With the new HA version the DeviceName part is taken from the Device section.
see [entity-naming](https://developers.home-assistant.io/docs/core/entity/#entity-naming) and [-change-naming-mqtt-entities](https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/).

Therefore in the folder `mqtt_discovery` in every file the following section needs to changed:
`"name": "{name} total energy consumption",` to `"name": "total energy consumption",`.

If you agree i can do that change.

BR Daniel

thecem, 2023-08-03:
The entity’s ID will be remaining the same?

„sensor.devicename_entityname“

killer0071234, 2023-08-03:
> The entity’s ID will be remaining the same?
> 
> „sensor.devicename_entityname“

Yes, it's just affecting the name.
And even the name stays the same, because HA is taking the device name as prefix. 

BR Daniel

BIBOLV, 2023-08-09:
Tnx for PR - it was accepted!

closed 2023-08-09 by BIBOLV