wmbusmeters/wmbusmeters #1009: MQTT discovery missing

greinet, 2023-08-03:
### Type of request

Bug

### HAOS version

2023.8.0

### wmbusmeters version

1.14.0

### Your message goes here

When I run the Home Assistant addon, i get the error message 
`[23:00:23] INFO: 
MQTT Discovery ...
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive`

It seems that the mqtt_discovery was moved to the addon repository at https://github.com/wmbusmeters/wmbusmeters-ha-addon/tree/main/wmbusmeters-ha-addon/mqtt_discovery at some point if I am not mistaken, so of course when running
`wget -O - https://github.com/wmbusmeters/wmbusmeters/archive/refs/heads/master.tar.gz 2> /dev/null | tar xz --strip=3 "wmbusmeters-master/ha-addon/mqtt_discovery" -C $templatedir || true` 
in the mqtt_discovery.sh script inside the[ Home Assistant addon](https://github.com/wmbusmeters/wmbusmeters-ha-addon/blob/main/wmbusmeters-ha-addon/mqtt_discovery.sh#L29C5-L29C191) the directory can not be found as it is not in the main wmbusmeters repository anymore.
Is this feature not used or ist this a bug?

Greetings,
Andy

BIBOLV, 2023-08-09:
Tnx for reporting - URL was changed.

closed 2023-08-09 by BIBOLV

mikkel75, 2023-12-07:
Just installed latest version: 1.14.0  - I still get this error:

`
[11:12:27] INFO: CONFIG_CONF ...
[11:12:27] INFO: {"device":"/dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_018D112F-if00-port0:im871a:t1,c1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"verbose","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[11:12:27] INFO: CONFIG_METERS ...
[11:12:27] INFO: []
[11:12:27] INFO: Syncing wmbusmeters configuration ...
[11:12:27] INFO: Registering meters ...
[11:12:27] INFO: Generating MQTT configuration ... 
[11:12:27] INFO: Broker core-mosquitto will be used.
[11:12:27] INFO: 
MQTT Discovery ...
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
[11:12:28] INFO: MQTT Discovery cleanup...
[11:12:29] INFO: Starting web configuration service.
[11:12:29] INFO: Running wmbusmeters ...
(wmbusmeters) using log file /dev/stdout
(wmbusmeters) version: 1.14.0
(config) using device: /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_018D112F-if00-port0:im871a:t1,c1 
(config) number of meters: 0
`

Any suggestions?