wmbusmeters/wmbusmeters #1011: SensoStar2, Qundis module 5.5 and Qheat 5: zero values

HubUser0, 2023-08-06:
### Type of request

Question

### OS version

Raspberry Pi OS 11 (bullyeye)

### wmbusmeters version

1.14.0

### Your message goes here

Hi,

First of all, thank you for this project and the effort you have put in to make it possible to read all these meters!

I have been trying to receive information from several water (with radio add-on modules) and heat meters with a **nanoCUL** USB stick with wMBUS firmware and an **im871a** USB stick connected to a Raspberry Pi 3B+. I receive the full set of data for the combination of Rossweiner water meters with an aventies radio add-on module (driver=**aventieswm**, data AES encrypted). I also get a reasonable amount of correct data (AES encrypted) from **SensoStar2** heat meters, but not the current energy value, only the current consumed volume. wmbusmeters comes to the conclusion that microclima 27/27 is the best driver.

(wmbusmeters) version: 1.14.0
(config) using device: /dev/ttyUSB0:im871a:c1   
(For aventieswm and SensoStar2: t1 instead of c1)
The results are the same for nanoCUL.

### SensoStar2

`wmbusmeters --analyze**=sensostar 3E44C5141703586300047A182030252F2F_046D1131E42804065C44000001FD17000415F3270000043B00000000042B00000000025F16000461220100002F2F`
Auto driver  : sensostar
Best driver  : microclima 27/27
Using driver : sensostar 00/00
000   : 3e length (62 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : c514 dll-mfct (EFE)
004   : 17035863 dll-id (63580317)
008   : 00 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 18 tpl-acc-field
012   : 20 tpl-sts-field ()
013   : 3025 tpl-cfg 2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
015   : 2f2f already decrypted check bytes
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 6D vif (Date and time type)
019 C!: 1131E428 ("meter_timestamp":"2023-08-04 17:17")
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 06 vif (Energy kWh)
025 C!: 5C440000 ("total_kwh":17500)
029   : 01 dif (8 Bit Integer/Binary Instantaneous value)
030   : FD vif (Second extension FD of VIF-codes)
031   : 17 vife (Error flags (binary))
032 C!: 00 ("current_status":"OK")
033   : 04 dif (32 Bit Integer/Binary Instantaneous value)
034   : 15 vif (Volume 10⁻¹ m³)
035 C!: F3270000 ("total_water_m3":1022.7)
039   : 04 dif (32 Bit Integer/Binary Instantaneous value)
040   : 3B vif (Volume flow l/h)
041 C!: 00000000 ("flow_water_m3h":0)
045   : 04 dif (32 Bit Integer/Binary Instantaneous value)
046   : 2B vif (Power W)
047 C!: 00000000 ("power_kw":0)
051   : 02 dif (16 Bit Integer/Binary Instantaneous value)
052   : 5F vif (Return temperature °C)
053 C!: 1600 ("return_c":22)
055   : 04 dif (32 Bit Integer/Binary Instantaneous value)
056   : 61 vif (Temperature difference 10⁻² K/°C)
057 C!: 22010000 ("difference_c":2.9)
061   : 2F skip
062   : 2F skip

[... 394 lines cut ...]
051   : 32 dif (16 Bit Integer/Binary Value during error state)
052   : 6C vif (Date type G)
053 C!: FFFF ("device_error_date":"2127-15-31")
055   : 04 dif (32 Bit Integer/Binary Instantaneous value)
056   : 6D vif (Date and time type)
057 C!: 1C15E528 ("device_date_time":"2023-08-05 21:28")

{
    "media":"heat",
    "meter":"qheat",
    "name":"",
    "id":"67627441",
    "status":"OK",
    "total_energy_consumption_kwh":0,
    "last_month_date":"2023-07-31",
    "last_month_energy_consumption_kwh":0,
    "last_year_date":"2022-12-31",
    "last_year_energy_consumption_kwh":0,
    "device_date_time":"2023-08-05 21:28",
    "device_error_date":"2127-15-31",
    "timestamp":"2023-08-05T21:41:20Z"
}

**What do I have to do to get the correct values instead of these 00000000?**
Thank you in advance!

HubUser0, 2023-08-10:
I have to apologize, but the current total energy value for the SensoStar2 heat meater is is actually given, for whatever reason I thought it was the value for 31.12.2022: 
(sensostar) 025 C!: 5C440000 ("total_kwh":17500)

That means that I receive the correct values for my **aventies water** meters and **sensostar2** heat meters, but not for for **Q Module 5.5** water and also not for **Q Heat 5**. The question is still: What do I have to do to receive the correct values instead of 00000000 for these meters?

HubUser0, 2023-08-10:
I guess that the structure of the telegrams from my Q module and Q heat meters is ok. It is just that the volume and energy values are not received/interpreted correctly or not even sent by the radio modules.
For example if I compare the telegrams for Q module 5.5 water from issue https://github.com/wmbusmeters/wmbusmeters/issues/352 the 00000000 values in the telegram of my Q module 5.5 are obvious:

[image]

HubUser0, 2023-10-25:
Is there nobody else who has ever experienced this problem?

HubUser0, 2023-10-25:
Good luck. So far I have not received any other volume or energy values than 0 from the Qundis meters and and modules.

kxn25, 2023-10-26:
Sorry but how did you install wmbusmeters?
i always get this error?
`Building 1.14.0-37-g6f0773a
g++ -I/usr/include/libxml2 -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/xmq.c -c -E > build/xmq.o.src
src/xmq.c:39:9: fatal error: libxml/tree.h: Datei oder Verzeichnis nicht gefunden
 #include<libxml/tree.h> `

HubUser0, 2023-10-26:
I did that several month ago and can't remember exactly, but just followed the instructions. Perhaps I did it using sudo.

HubUser0, 2023-10-26:
I don't think that I did anything else than:  
./configure; make; sudo make install will install wmbusmeters as a daemon.

kxn25, 2023-10-29:
Ok Thank you is installed now.
how did you get the AES Key?

HubUser0, 2023-10-29:
The seller activated the meter/module and deleted the AES key. He asked me if I wanted a key or not. I guess that your only hope is that the seller gives you the key (and is allowed to do it).

kxn25, 2023-10-29:
I got it working with the Wireless Adapter in Iobroker.
No neeed for AES keys.
It´s finding the Meter and the W-module. I have two Meters.

[image]

[image]

HubUser0, 2023-10-29:
Good!
Are the consumption values in IObroker the same as on the devices?

HubUser0, 2023-10-29:
What does it say in the log file for the interpretation of the telegrams?

HubUser0, 2023-10-29:
Which WMBUS USB stick are you using?

HubUser0, 2023-10-29:
[image]
I had several of these heat meters, but got only 0 m³ and 0 kWh values for all of them via wmbusmeters, the same for all Qmodule 5.5 water devices. I exchanged all the Qundis heat meters for Engelmann heat meters get all values delivered. The Qundis Qmodule 5.5 water devices are still not giving me any meaningful volume values. I am really wondering now, where my installation of wmbusmeters 1.14.0 might be different from yours.

kxn25, 2023-10-29:
The reading are corrosponding with the meters.
i have ordert from Piotek.de 
I have not setup the wmbusmeters i only use the [https://github.com/lvogt/ioBroker.wireless-mbus/blob/master/README.md](url)
I cant even see any log!!!
[image]

HubUser0, 2023-10-29:
OK, I might need to give that a try as well. Thanks for the info.

kxn25, 2023-10-29:
It's updating ever half a hour.

You are Welcome, good luck.

HubUser0, 2023-10-29:
Thanks!

I still hope that somebody can give a hint for the current Q module 5.5 heat or water devices regarding wmbusmeters.

HubUser0, 2023-10-29:
Unfortunately I get the same results with ioBroker and iM871a: positive values for the Engelmann heat meters and always 0 m³ for Qundis Qmodule 5.5 water.
[image]
I assume that the nanoCUL USB stick will not change that.

HubUser0, 2023-10-29:
[image]

HubUser0, 2023-10-29:
nanoCUL USB Stick, ioBroker and another Raspberry Pi 3B+ deliver also only 0 m³ values for the Qmodule 5.5 water devices. I have 16 of these Qmodule radio add-on devices. All of them show the same 0 m³, but the water meters show positive values.

kxn25, 2023-10-30:
Are the meters new? (Batterie?)
What chip is on that board?
I got the IM871A from Piotek.de.

Is ther something configured in the meters menue?
In the discription of the meter there is a parameter which can be changed if this option is unlocked!

HubUser0, 2023-10-30:
I get the energy and volume values from the Qmodule 5.5 **heat** devices, but from the Qmodule 5.5 **water** devices I do not get any meaningful volume data. There is nothing to be configured from me as a retail customer. I would need special programming hard and software to change anything.
[image]
The water meters themselves count the volume properly, but I also tend to assume that it is a setting issue of the Q modules.

HubUser0, 2023-10-30:
If AES keys were required the telegrams could not be decrypted without the keys. (My old meters all require AES keys. I have most of those keys.)

kxn25, 2023-10-30:
Sorry i dont know more than that.

weetmuts, 2023-10-31:
@HubUser0 Yes, it is obvious that the meter sends proper telegrams that you receive fine, but the meter sends zero values. You can also use the online analyze: https://wmbusmeters.org/analyze/3C449344228622372337724174626793442304460000200C05000000004C0500000000426CDF2CCC080500000000C2086CFF27326CFFFF046D1C15E528

@HubUser0  > also tend to assume that it is a setting issue of the Q modules.
Yes, I think so too.

@kxn25 I am working on loadable drivers from textfiles. So that we can maintain a driver database separate without having to recompile wmbusmeters. This is why the libxml2 dependency has been added. The format for the driver files looks like this:
https://libxmq.org/resources/examples/wmbusmeters_driver.html

If you run ./configure again it should recommend: sudo apt install libxml2-dev

HubUser0, 2023-10-31:
Yes, I checked several telegrams from the Qmodules with online tools, wmbusmeters and iobroker and compared the telegrams with other telegrams. It all points to a configuration issue of the Qmodules.

Thanks you for your answer.

kxn25, 2023-11-01:
The Answer from the Shop i got my Quindis from:

They have no AES Keys! 

[image]

HubUser0, 2023-11-07:
Finally it turned out that all 16 Qundis radio add-on modules were not activated in such a way that they were counting the water flow. The little Torx screw needs to be turned so much that a red light appears for 30 s and then at most half a turn more. Then the red light flashes periodically. That was not the case so far.
The radio modules need to be close enough to a turning magnetic disk in the water meters, otherwise the counting process is not transferred to the radio modules.

HubUser0, 2023-11-07:
The red lights are flashing on for about the 1st hour.

DevSecNinja, 2023-11-08:
> I got it working with the Wireless Adapter in Iobroker. No neeed for AES keys. It´s finding the Meter and the W-module. I have two Meters.
> 
> [image]
> 
> [image]

I've ordered the same stick as well but I can't get any telegrams yet. What mode do you run (C1/T1/S1?) and how often do you notice the Qmodule 5.5 Heat sends out a telegram? I can get unknown telegrams on S1 though, but they are unknown by wmbusmeters. Thanks!

kxn25, 2023-11-08:
I run this in C-mode (type A) with the W Mbus Adapter, i dont have wmbusmeters installed.
They update every 7 1/2min. 
I read somewhere that diffrend Mode have diffrend update times!

[image]

HubUser0, 2023-11-08:
You might need to run some litres (about 50) through the meters before they activate themselves. If the temperature difference between the sensors the energy flow should show up quickly.

DevSecNinja, 2023-11-08:
Hmm interesting, thanks. I'm trying to run wmbusmeters for my existing sensors and they don't show up in the Home Assistant add-on logs when using C1 and removing the dummy sensor. Any hints on troubleshooting?

HubUser0, 2023-11-08:
I only have Q module 5.5 water (not heat anymore). C1 works for them with wmbusmeters using nanoCUL or im871a. 
Are you sure that you don't need AES keys?
Have you tried the debug mode?

DevSecNinja, 2023-11-08:
Hmm, I have seen some encrypted messages on T1. I'll have a look at those. Thanks! I'm not getting any messages on c1 unfortunately.

HubUser0, 2023-11-08:
Have you asked the seller/installer for the mode and AES keys (if any in use)?

DevSecNinja, 2023-11-09:
No I did not as the meters have been pre-installed in the apartment. Can I somehow obtain the required mode? AES key would be difficult I guess.

HubUser0, 2023-11-09:
That is certainly not from a Qundis meter/module as it is from a Techem heat cost allocator. 
AES keys: Qundis does not interact with retail customers. The service company or the owner who has installed the meters should have the AES key.

If you try C or S mode you should receive sooner or later telegrams from the Q modules. 
https://qundis.de/wp-content/uploads/2017/10/Onepager_Qmodule5.5heat.pdf. 
https://qundis.de/wp-content/uploads/2023/01/DST5-00MO-DE0-QMH55-A_180123.pdf
The log file will tell you if they are encrypted or not.

HubUser0, 2023-11-09:
I guess C1 mode. Let it run for a couple of hours, preferably during day time, it might be restricted to Mo-Fr and 8-18 hours. If you do not see anything try S1.

DevSecNinja, 2023-11-09:
> I guess C1 mode. Let it run for a couple of hours, preferably during day time, it might be restricted to Mo-Fr and 8-18 hours. If you do not see anything try S1.

Will do, thanks very much for your help! 👍

closed 2024-02-09 by github-actions[bot]