wmbusmeters/wmbusmeters #1017: Lansen LAN-WMBUS-SMK2

hngfngzl, 2023-08-21:
### Type of request

Existing device improvement

### Meter description

Hello,

i bought an Lansen LAN-WMBUS-SMK2. 
The system gets its telegram, but I guess I need the decryption key (from Lansen). 
Ist it normal, that he says driver: unkown! ? In the diver list there is a Lansen Smoke detector enclosed. 

Best Regards
Daniel Wolf

### Logged telegrams from meter in decrypted format

```shell
Received telegram from: 01002751
          manufacturer: (LAS) Lansen Systems, Sweden (0x3033)
                  type: Smoke detector (0x1a) encrypted
                   ver: 0x0a
                device: rtlwmbus[00000001]
                  rssi: 130 dBm
                driver: unknown!
telegram=|_4E443330512700010A1A7A43804005AC981DDFBCA0BA3F30683A67B2C56FEB4C397816C0CD09C7FE59C7126C1628BBC48B7C33971B9039E1C1DF1BBE5A738B5EDD71088991123D038DD3F2806F5044|+106
```

weetmuts, 2023-08-21:
Yep, you need a decryption key. Since you bought the meter, you should be able to ask for the key.

There a lot of different versions of meters. The driver is unknown if there is not an exact match with the known mfct/version/media combo. However the existing driver lansensm might work just fine, we just have to add the new mfct/version/media combo to it. But we do not know until you have the decryption key.

https://wmbusmeters.org/analyze/4E443330512700010A1A7A43804005AC981DDFBCA0BA3F30683A67B2C56FEB4C397816C0CD09C7FE59C7126C1628BBC48B7C33971B9039E1C1DF1BBE5A738B5EDD71088991123D038DD3F2806F5044

hngfngzl, 2023-08-21:
Try this decrypted:
Auto driver  : not found!
Best driver  : unknown 00/00
Using driver : lansensm 00/00
000   : 4e length (78 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 3330 dll-mfct (LAS)
004   : 51270001 dll-id (01002751)
008   : 0a dll-version
009   : 1a dll-type (Smoke detector)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 6e tpl-acc-field
012   : 80 tpl-sts-field ()
013   : 4005 tpl-cfg 0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
015   : 2f2f already decrypted check bytes

{
    "media":"smoke detector",
    "meter":"lansensm",
    "name":"",
    "id":"01002751",
    "timestamp":"1970-01-01T00:00:00Z"
}

Using: wmbusmeters: 1.14.0-8-gd0f9ffc
d0f9ffc8206be4e5d18e13605218dde04f747eef

hngfngzl, 2023-08-21:
Next try: I took a decrypted telegram an used https://wmbusmeters.org/ with the key to decrypt.

Auto driver  : not found!
Best driver  : lansensm 02/20
Using driver : lansensm 02/20
000   : 4e length (78 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 3330 dll-mfct (LAS)
004   : 51270001 dll-id (01002751)
008   : 0a dll-version
009   : 1a dll-type (Smoke detector)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : f1 tpl-acc-field
012   : 80 tpl-sts-field ()(UNKNOWN_80)
013   : 4005 tpl-cfg 0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 7C vif (VIF in following string (length in first byte))
019   : 03 viflen (3)
020   : 33 vif (3)
021   : 53 vif (S)
022   : 44 vif (D)
023 C?: A0800400
027   : 02 dif (16 Bit Integer/Binary Instantaneous value)
028   : FD vif (Second extension FD of VIF-codes)
029   : 3A vife (Dimensionless / no VIF)
030 C!: 8202 ("minutes_since_last_manual_test_counter":642)
032   : 82 dif (16 Bit Integer/Binary Instantaneous value)
033   : 40 dife (subunit=1 tariff=0 storagenr=0)
034   : FD vif (Second extension FD of VIF-codes)
035   : 3A vife (Dimensionless / no VIF)
036 C?: 1D00
038   : 02 dif (16 Bit Integer/Binary Instantaneous value)
039   : FD vif (Second extension FD of VIF-codes)
040   : 46 vife (10^-3 Volts)
041 C?: 850C
043   : 82 dif (16 Bit Integer/Binary Instantaneous value)
044   : 40 dife (subunit=1 tariff=0 storagenr=0)
045   : FD vif (Second extension FD of VIF-codes)
046   : 46 vife (10^-3 Volts)
047 C?: FC0D
049   : 02 dif (16 Bit Integer/Binary Instantaneous value)
050   : 23 vif (On time days)
051 C?: 0100
053   : 82 dif (16 Bit Integer/Binary Instantaneous value)
054   : 80 dife (subunit=0 tariff=0 storagenr=0)
055   : 40 dife (subunit=2 tariff=0 storagenr=0)
056   : FD vif (Second extension FD of VIF-codes)
057   : 3A vife (Dimensionless / no VIF)
058 C?: DA04
060   : 02 dif (16 Bit Integer/Binary Instantaneous value)
061   : 7C vif (VIF in following string (length in first byte))
062   : 03 viflen (3)
063   : 54 vif (T)
064   : 46 vif (F)
065   : 23 vif (#)
066 C?: 0500
068   : 02 dif (16 Bit Integer/Binary Instantaneous value)
069   : 7C vif (VIF in following string (length in first byte))
070   : 03 viflen (3)
071   : 4C vif (L)
072   : 41 vif (A)
073   : 23 vif (#)
074 C?: 0000
076   : 2F skip
077   : 2F skip
078   : 2F skip

{
    "media":"smoke detector",
    "meter":"lansensm",
    "name":"",
    "id":"01002751",
    "status":"UNKNOWN_80",
    "minutes_since_last_manual_test_counter":642,
    "timestamp":"2023-08-21T19:10:48Z"
}

Using: wmbusmeters: 1.14.0-8-gd0f9ffc
d0f9ffc8206be4e5d18e13605218dde04f747eef

hngfngzl, 2023-08-21:
I have a generell question of the design idea of /etc/wmbusmeters.d and its subfiles.

Is the idea to generate a file for each meter (or is the idea to to group it with wildcard * in the id with the same driver) or to put in several different meters in one file (what didn't work out)?

hngfngzl, 2023-10-21:
Can be closed. Driver should be new but I guess I take a different device.

closed 2023-10-21 by hngfngzl