wmbusmeters/wmbusmeters #1018: Json file does not tell anything interesting

Solid48, 2023-08-21:
### Type of request

Bug

### OS version

Ubuntu 23.04

### wmbusmeters version

it says "_", but i downloaded it 2 to 3 months ago

### Your message goes here

When using the command (similar to the example in the readme) 
`wmbusmeters --format=json rtlwmbus:c1 testdevice auto ANYID NOKEY`
I get a lot of telegrams from my Qundis heat cost allocators (I checked that they had no password). But the information is in the form 
`{"media":"heat_cost_allocation","meter":"qcaloric","name":"testdevice","id":"8962...","status":"OK","device_date_time":"...","timestamp":"...","device":"rtlwmbus[00000001]","rssi_dbm":30}`
which does not help me because i do not need to know the date and time but the actual readout of the heat cost allocator namely consumption during last calender year and consumption up to the current date

Thanks in advance

weetmuts, 2023-08-21:
Add --logtelegrams and post the telegram here.

Solid48, 2023-08-21:
Thanks for answering so fast. This is about ten minutes of data

`Started config rtlwmbus[00000001] listening on c1
(meter) newly created meter (testdevice 89624212 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 29536055 sontex868) did not handle telegram!
(meter) newly created meter (testdevice 89621519 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89624211 qcaloric) did not handle telegram!
telegram=|294468504002406476F0A0_009F2D122F00C0122F00805A000014006BA1007CB2008DC3009ED4000FE500|+137
{"media":"smoke detector","meter":"tsd2","name":"testdevice","id":"64400240","status":"OK","prev_date":"2022-12-31T02:00:00Z","timestamp":"2023-08-21T18:14:40Z","device":"rtlwmbus[00000001]","rssi_dbm":25}
(meter) newly created meter (testdevice 89624210 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89624208 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89621516 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89621518 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89621515 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89621521 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89621538 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89624209 qcaloric) did not handle telegram!
telegram=|294468508357377176F0A0_009F2D102F00C0102F01805A000014006BA1007CB2008DC3009ED4000FE500|+370
{"media":"smoke detector","meter":"tsd2","name":"testdevice","id":"71375783","status":"OK","prev_date":"2022-12-31T02:00:00Z","timestamp":"2023-08-21T18:18:33Z","device":"rtlwmbus[00000001]","rssi_dbm":20}
(meter) newly created meter (testdevice 89621508 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89621514 qcaloric) did not handle telegram!
(meter) newly created meter (testdevice 89621507 qcaloric) did not handle telegram!
telegram=|294468504002406476F0A0_009F2D122F00C0122F00805A000014006BA1007CB2008DC3009ED4000FE500|+621
{"media":"smoke detector","meter":"tsd2","name":"testdevice","id":"64400240","status":"OK","prev_date":"2022-12-31T02:00:00Z","timestamp":"2023-08-21T18:22:44Z","device":"rtlwmbus[00000001]","rssi_dbm":14}
(meter) newly created meter (testdevice 89621520 qcaloric) did not handle telegram!`

I also noticed the `newly created meter (testdevice ... qcaloric) did not handle telegram!` message which should as far as I am concerned only show up if you do not have the correct password. But this can't be the case because two of them (giving the same data as every other one) were bought fresh and all of them said they had no password when checking with 
`wmbusmeters rtlwmbus:c1`

Solid48, 2023-08-23:
I resolved the problem myself, sorry for the inconvinience