wmbusmeters/wmbusmeters #1023: Questions for first time usage

miloit, 2023-08-28:
### Type of request

Question

### OS version

debian 11

### wmbusmeters version

newest

### Your message goes here

Hello I get this message

```
wmbusmeters --verbose auto 
(wmbusmeters) version: _
(config) number of meters: 0
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
```

how to solve it?

testuser7, 2023-09-02:
@miloit how many dongles do you have?
If you only have one dongle, try configuration with the option:
```
device=auto:t1
donotprobe=all
```
And install rtl_wmbus.

miloit, 2023-09-03:
Where do I find rtl_wmbus?

testuser7, 2023-09-03:
> Where do I find rtl_wmbus?

https://github.com/xaelsouth/rtl-wmbus
I don't know if there is a debian package available.

Another option is to use rtl_433:
```
device=rtl433
donotprobe=all
```

closed 2023-12-03 by github-actions[bot]