wmbusmeters/wmbusmeters #1024: Multical21 wrong key

big23, 2023-08-30:
### Type of request

Question

### OS version

Debian 11

### wmbusmeters version

1.14.0

### Your message goes here

I checked my KEY with Python and HA-addon. Everthing OK.
Decrypted meter details:
id = 74685644
key = B91EAF561BEF664ADD91875125763485

...but I have still error Wrong key....

(serial) received ascii "C1;1;1;2023-08-30 17:35:03.000;29;35;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2023-08-30 17:35:03.000;29;35;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 36
(wmbus) parseELL @10 26
(wmbus) parseDLL @0 36
(telegram) DLL L=23 C=44 (from meter SND_NR) M=2c2d (KAM) A=74685644 VER=1b TYPE=16 (Cold water meter) (driver multical21) DEV=rtlwmbus[00000001] RSSI=29
(wmbus) parseELL @10 26
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=13 SN=c1875020 (AES_CTR session=1 time=329852) CRC=fabd
Received telegram from: 74685644
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: rtlwmbus[00000001]
                  rssi: 29 dBm
                driver: multical21
(wmbus) 000   : 23 length (35 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 2d2c dll-mfct (KAM)
(wmbus) 004   : 44566874 dll-id (74685644)
(wmbus) 008   : 1b dll-version
(wmbus) 009   : 16 dll-type (Cold water meter)
(wmbus) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : 13 ell-acc
(wmbus) 013   : c1875020 sn (AES_CTR)
(wmbus) 017   : fabd payload crc (calculated e6a9 ERROR)
(wmbus) **019 CE: F7ED2AF48698B8B1BBDE661C9BCCA3820F failed decryption. Wrong key?**
(wmbus) telegram from 74685644 ignored by all configured meters!
(serial) received ascii "C1;1;1;2023-08-30 17:35:03.000;33;28;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2023-08-30 17:35:03.000;33;28;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=36.

big23, 2023-08-30:
I found problem with Telegram.
When I analyze on wmbusmeters.org my telegram: 0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>
I have an CRC error.
But when I change telegram to 23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f - everything is ok....

weetmuts, 2023-10-31:
Yes, the decryption code is pretty well tested, so the most likely reason for this is that you have the wrong key.  :-)

closed 2024-01-31 by github-actions[bot]