wmbusmeters/wmbusmeters #1026: How to run UART over TCP inside of docker container?

PovilasID, 2023-09-10:
### Type of request

Question

### OS version

Ubuntu 22.04 

### wmbusmeters version

1.14.0

### Your message goes here

This is more of documentation request.

Context:
I have im871a plugged into my OpenWRT router closer to my meter and I have ser2net exposing the device on port 5001 on default router ip 192.168.1.1 (OpenWRT even has UI for this package) Here is my config https://i.imgur.com/ls0s65D.png

Question:
How do I set wmbusmeters to listen to it? aka what do I write in the `device=` line in the config?

I found a sample config at #421

> `rawtty:CMD(nc -lk 4444)`

However in the readme in the devices section you list that `CMD` is available for `rtlwmbus` but `rawtty` looks only like an option for `stdin` and `telegrams.bin` Almost all configuration I tried returned `Not a valid device` (except `Started config rtlwmbus listening on none using CMD(nc -lk 192.168.1.1:5001)`)

testuser7, 2023-09-12:
You can try: `im871a:CMD(nc -lk 192.168.1.1:5001)`

weetmuts, 2023-09-13:
I think that socat might work with the right setup.

The thing is that when you override the wmbus device with a CMD, it blocks outgoing messages to the dongle
also it ignores and serial speed settings etc.

This is because the override feature was developed to perform testing with simulated dongles.
It just happened to work for the rtlwmbus/rawtty since they a pure input feeds anyway.

However the im871a and the other hardware dongles are two way protocols. You tell the dongle what kind of radio messages you want to listen to, transmit a read-out message to C2 meters, etc.

The question of how to run wmbusmeters on one device and connect the dongle to another device has come up before. It is an interesting and valuable problem to be solved.

Please try socat first!

PovilasID, 2023-09-14:
Yah I tried socat first just hoping it will magically work. It did not, so I came here to check maybe somebody has already had this issue and solved it. 
Socat has a bunch of options and I successfully avoided learning how these dongles actually work, till now.  I am going to try and do some debugging but if you have any hints that could narrow my scope I would greatly  appreciate it though considering how I do not know I do not expect to be successful... It is quite likely I am just going to take some raspberry pi like device that can be USB powered plug that into router and try to run this container on it. Either way, this will not be fun.

spectrality-github, 2023-10-04:
I have similar setup just with wired Mbus.  My Mbus master USB adapter (from AliExpress) connected to OpenWrt device with ser2net running.  And waiting for native Mbus TCP support.  https://github.com/wmbusmeters/wmbusmeters/issues/46 

Solution how to access remote serial port using socat is described here: https://www.lucid-control.com/accessing-usb-io-module-network-linux-socat/

You just need run socat before wmbusmeters 
socat -d -d pty,group-late=tty,link=/root/ttyMBUS0,mode=660,rawer,echo=0,b2400,waitslave,ignoreeof tcp:192.168.x.x:2003 &
And use /root/ttyMBUS0 as device in wmbusmeters config.

But socat process will stop when ser2net is closing TCP connection by timeout or connection problems or when wmbusmeters is not sending data. And to restore connectivity you will need to restart socat and wmbusmeters 

my ser2net config is:
2003:raw:30:/dev/mbusUSB0:2400 8DATABITS EVEN 1STOPBIT -RTSCTS LOCAL -XONXOFF

PovilasID, 2023-10-16:
@spectrality-github Thanks for detailed description of working setup. I tried unsuccefully. I am still getting `socat[29] N read(5, 0xffff80ce5000, 8192): I/O error (probably PTY closed)` but I am also running everything in docker containers, so extra level of abstraction may not be helping my cause.

closed 2024-01-15 by github-actions[bot]

weetmuts, 2026-11-02:
Support now exists: https://github.com/wmbusmeters/wmbusmeters/issues/2162