wmbusmeters/wmbusmeters #1027: Integra Topas Sonic

MarkKsKs, 2023-09-12:
### Type of request

New device support

### Meter description

manufacturer: (IMT) INTEGRA METERING, Switzerland (0x25b4)
type: Water meter (0x07) encrypted
ver: 0x10
device: amb8465[00000000]
rssi: -136 dBm
driver: unknown!

(Telegrams below are already decrypted)

### Logged telegrams from meter in decrypted format

```shell
telegram=|4144B4257746151010078C20C27A671030252F2F_04130F00000084101300000000043B00000000046D0A2CEC2944130F000000426CFF2802FD17800802FD7451152F|

Auto driver  : not found!
Best driver  : hydrus 24/26
Using driver : topaseskr 00/00
000   : 41 length (65 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b425 dll-mfct (IMT)
004   : 77461510 dll-id (10154677)
008   : 10 dll-version
009   : 07 dll-type (Water meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : c2 ell-acc
013   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
014   : 67 tpl-acc-field
015   : 10 tpl-sts-field (TEMPORARY_ERROR)
016   : 3025 tpl-cfg 2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
018   : 2f2f decrypt check bytes (OK)
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : 13 vif (Volume l)
022 C!: 0F000000 ("total_m3":0.015)
026   : 84 dif (32 Bit Integer/Binary Instantaneous value)
027   : 10 dife (subunit=0 tariff=1 storagenr=0)
028   : 13 vif (Volume l)
029 C?: 00000000
033   : 04 dif (32 Bit Integer/Binary Instantaneous value)
034   : 3B vif (Volume flow l/h)
035 C!: 00000000 ("current_flow_m3h":0)
039   : 04 dif (32 Bit Integer/Binary Instantaneous value)
040   : 6D vif (Date and time type)
041 C?: 0A2CEC29
045   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
046   : 13 vif (Volume l)
047 C!: 0F000000 ("volume_year_period_m3":0.015)
051   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
052   : 6C vif (Date type G)
053 C!: FF28 ("meter_year_period_start_date":"2023-08-31")
055   : 02 dif (16 Bit Integer/Binary Instantaneous value)
056   : FD vif (Second extension FD of VIF-codes)
057   : 17 vife (Error flags (binary))
058 C?: 8008
060   : 02 dif (16 Bit Integer/Binary Instantaneous value)
061   : FD vif (Second extension FD of VIF-codes)
062   : 74 vife (Reserved)
063 C!: 5115 ("battery_y":14.940759)
065   : 2F skip

{
    "media":"water",
    "meter":"topaseskr",
    "name":"",
    "id":"10154677",
    "battery_y":14.940759,
    "current_flow_m3h":0,
    "total_m3":0.015,
    "volume_year_period_m3":0.015,
    "meter_year_period_start_date":"2023-08-31",
    "timestamp":"2023-09-12T12:45:43Z"
}

---

telegram=|4144B4251886201010078C20B27A4E1030252F2F_04130000000084101300000000043B00000000025A1E01046D2D2CEC297C130000000002FD17800802FD7423162F|

Auto driver  : not found!
Best driver  : hydrus 20/26
Using driver : topaseskr 00/00
000   : 41 length (65 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b425 dll-mfct (IMT)
004   : 18862010 dll-id (10208618)
008   : 10 dll-version
009   : 07 dll-type (Water meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : b2 ell-acc
013   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
014   : 4e tpl-acc-field
015   : 10 tpl-sts-field (TEMPORARY_ERROR)
016   : 3025 tpl-cfg 2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
018   : 2f2f decrypt check bytes (OK)
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : 13 vif (Volume l)
022 C!: 00000000 ("total_m3":0)
026   : 84 dif (32 Bit Integer/Binary Instantaneous value)
027   : 10 dife (subunit=0 tariff=1 storagenr=0)
028   : 13 vif (Volume l)
029 C?: 00000000
033   : 04 dif (32 Bit Integer/Binary Instantaneous value)
034   : 3B vif (Volume flow l/h)
035 C!: 00000000 ("current_flow_m3h":0)
039   : 02 dif (16 Bit Integer/Binary Instantaneous value)
040   : 5A vif (Flow temperature 10⁻¹ °C)
041 C!: 1E01 ("temperature_c":28.6)
043   : 04 dif (32 Bit Integer/Binary Instantaneous value)
044   : 6D vif (Date and time type)
045 C?: 2D2CEC29
049   : 7C dif (8 digit BCD Value during error state storagenr=1)
050   : 13 vif (Volume l)
051 C?: 00000000
055   : 02 dif (16 Bit Integer/Binary Instantaneous value)
056   : FD vif (Second extension FD of VIF-codes)
057   : 17 vife (Error flags (binary))
058 C?: 8008
060   : 02 dif (16 Bit Integer/Binary Instantaneous value)
061   : FD vif (Second extension FD of VIF-codes)
062   : 74 vife (Reserved)
063 C!: 2316 ("battery_y":15.515719)
065   : 2F skip

{
    "media":"water",
    "meter":"topaseskr",
    "name":"",
    "id":"10208618",
    "battery_y":15.515719,
    "current_flow_m3h":0,
    "temperature_c":28.6,
    "total_m3":0,
    "timestamp":"2023-09-12T12:46:38Z"
}

-- 

(driver=auto)

telegram=|4144B4251886201010078C20C27A5E1030252F2F_04130000000084101300000000043B00000000025A1E01046D312CEC297C130000000002FD17800802FD7423162F|

Auto driver  : not found!
Best driver  : hydrus 20/26
Using driver : hydrus 20/26
000   : 41 length (65 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b425 dll-mfct (IMT)
004   : 18862010 dll-id (10208618)
008   : 10 dll-version
009   : 07 dll-type (Water meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : c2 ell-acc
013   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
014   : 5e tpl-acc-field
015   : 10 tpl-sts-field (TEMPORARY_ERROR)(TEMPORARY_ERROR)
016   : 3025 tpl-cfg 2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
018   : 2f2f decrypt check bytes (OK)
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : 13 vif (Volume l)
022 C!: 00000000 ("total_m3":0)
026   : 84 dif (32 Bit Integer/Binary Instantaneous value)
027   : 10 dife (subunit=0 tariff=1 storagenr=0)
028   : 13 vif (Volume l)
029 C!: 00000000 ("total_tariff1_m3":0)
033   : 04 dif (32 Bit Integer/Binary Instantaneous value)
034   : 3B vif (Volume flow l/h)
035 C!: 00000000 ("flow_m3h":0)
039   : 02 dif (16 Bit Integer/Binary Instantaneous value)
040   : 5A vif (Flow temperature 10⁻¹ °C)
041 C!: 1E01 ("flow_temperature_c":28.6)
043   : 04 dif (32 Bit Integer/Binary Instantaneous value)
044   : 6D vif (Date and time type)
045 C!: 312CEC29 ("meter_datetime":"2023-09-12 12:49")
049   : 7C dif (8 digit BCD Value during error state storagenr=1)
050   : 13 vif (Volume l)
051 C?: 00000000
055   : 02 dif (16 Bit Integer/Binary Instantaneous value)
056   : FD vif (Second extension FD of VIF-codes)
057   : 17 vife (Error flags (binary))
058 C?: 8008
060   : 02 dif (16 Bit Integer/Binary Instantaneous value)
061   : FD vif (Second extension FD of VIF-codes)
062   : 74 vife (Reserved)
063 C!: 2316 ("remaining_battery_life_y":15.515719)
065   : 2F skip

{
    "media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"10208618",
    "flow_m3h":0,
    "flow_temperature_c":28.6,
    "remaining_battery_life_y":15.515719,
    "total_m3":0,
    "total_tariff1_m3":0,
    "meter_datetime":"2023-09-12 12:49",
    "status":"TEMPORARY_ERROR",
    "timestamp":"2023-09-12T12:48:18Z"
}
```

gh0stface33, 2023-09-15:
@MarkKsKs What is the key you use to decrypted your water meter ?
I have integra topas sonic too.

MarkKsKs, 2023-09-15:
> @MarkKsKs What is the key you use to decrypted your water meter ? I have integra topas sonic too.

I have asked my supplier the key, they're the ones who should give it to you.

gh0stface33, 2023-09-15:
> > @MarkKsKs What is the key you use to decrypted your water meter ? I have integra topas sonic too.
> 
> I have asked my supplier the key, they're the ones who should give it to you.

Ok, my supplier don't want to give me the key or he doesn't know the key. 
Can you give me your key to test with my meter ?

gh0stface33, 2023-09-16:
@MarkKsKs 

Hi, my supplier doesn't want to give me the key or they didn't know it

Can you give me your key to test with my meter ?

gh0stface33, 2023-09-21:
@MarkKsKs 
Hi can you help me ?

gh0stface33, 2023-09-22:
@MarkKsKs 

Hi can you test to decrypt my telegram with your decrypted key ?

telegram=|_4E44B4252178101010077A1A00402571B721D881E204544353B85C1E05267927264C75D8ABAB21F6D7EC8081CD4494C59D1F24FD590FB5B253F768FA3DFC8404AB6910ABC01C17DCCCC38C24E06449|+12

gh0stface33, 2023-10-05:
@MarkKsKs 

Please can you help me ?