wmbusmeters/wmbusmeters #1036: wmbusmeters randomly stop processing telegrams

raszuk, 2023-09-14:
### Type of request

Bug

### wmbusmeters version

1.14.0-21-g239a39d

### Your message goes here

wmbus randomly stops working in snap: 

(serialcmd) closed /bin/sh pid=439922 fd=5 (rtlwmbus)
Lost 00000001 closing rtlwmbus[00000001]
(serialcmd) closed /bin/sh pid=439922 fd=-1 (rtlwmbus)
No wmbus device detected, waiting for a device to be plugged in.
(shell) invoking /bin/pidof failed!
(serial) event loop stopped!
(wmbusmeters) version: 1.14.0-21-g239a39d

sudo snap restart wmbusmeters fixed the issue for some time, but this is not a solution 

Side question: Is there some way to add timestamp to wmbusmeters.log file ?

weetmuts, 2023-09-15:
--logtimestamps=always

Check the dongle hardware. Does it get hot and shut down itself? Is the usb plug unstable?

raszuk, 2023-09-15:
It used to get hot in the past so I have added two big radiators to it and fan next to it. Now it is completely cold. Dongle cable and hardware are all very stable and the setup is fixed so does not move. The dongle is also not hanging but mounted on a base and connected via USB short cable. 

I never had this problem when I used same setup directly in ubuntu. Only surfaced when I moved to snap. So not sure if this may play any role or how to debug it further. Will browse systems logs when I enable timestamps. 

Thx !

raszuk, 2023-09-18:
OK I think I know what happens ... timestamping the log helped.  out of the sudden ubuntu core with snap discovers Realtek RTL2823U as DVD and cut's wmbusmeters from connectivity to it. Please see below log: 

wmbusmeters.log:

(serialcmd) closed /bin/sh pid=11106 fd=5 (rtlwmbus)
[2023-09-18_18:24:18] Lost 00000001 closing rtlwmbus[00000001]
(serialcmd) closed /bin/sh pid=11106 fd=-1 (rtlwmbus)
No wmbus device detected, waiting for a device to be plugged in.
[2023-09-18_18:24:18] (shell) invoking /bin/pidof failed!

- - - 
dmesg: 

[Mon Sep 18 18:24:17 2023] usb 1-1.4: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[Mon Sep 18 18:24:17 2023] usb 1-1.4: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[Mon Sep 18 18:24:17 2023] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
[Mon Sep 18 18:24:17 2023] usb 1-1.4: media controller created
[Mon Sep 18 18:24:17 2023] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[Mon Sep 18 18:24:17 2023] i2c i2c-11: Added multiplexed i2c bus 12
[Mon Sep 18 18:24:17 2023] rtl2832 11-0010: Realtek RTL2832 successfully attached
[Mon Sep 18 18:24:17 2023] usb 1-1.4: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[Mon Sep 18 18:24:17 2023] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
[Mon Sep 18 18:24:17 2023] r820t 12-001a: creating new instance
[Mon Sep 18 18:24:17 2023] r820t 12-001a: Rafael Micro r820t successfully identified
[Mon Sep 18 18:24:17 2023] rtl2832_sdr rtl2832_sdr.0.auto: Registered as swradio0
[Mon Sep 18 18:24:17 2023] rtl2832_sdr rtl2832_sdr.0.auto: Realtek RTL2832 SDR attached
[Mon Sep 18 18:24:17 2023] rtl2832_sdr rtl2832_sdr.0.auto: SDR API is still slightly experimental and functionality changes may follow
[Mon Sep 18 18:24:17 2023] Registered IR keymap rc-empty
[Mon Sep 18 18:24:17 2023] rc rc0: Realtek RTL2832U reference design as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/rc/rc0
[Mon Sep 18 18:24:17 2023] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[Mon Sep 18 18:24:17 2023] input: Realtek RTL2832U reference design as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/rc/rc0/input3
[Mon Sep 18 18:24:17 2023] usb 1-1.4: dvb_usb_v2: schedule remote query interval to 200 msecs
[Mon Sep 18 18:24:17 2023] usb 1-1.4: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[Mon Sep 18 18:24:18 2023] audit: type=1400 audit(1695054258.762:33): apparmor="DENIED" operation="exec" profile="snap.wmbusmeters.wmbusmeters" name="/usr/sbin/killall5" pid=19190 comm="wmbusmeters" requested_mask="x" denied_mask="x" fsuid=0 ouid=0

Btw when running under snap the wmbusmeters does NOT timestamp important events by default :( 

Btw after reset it runs find for a day or two then suddenly stops as described above.

raszuk, 2023-09-18:
Log just after restarting: 

(serial) event loop stopped!
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
[2023-09-19_00:18:39] (shell) invoking /bin/pidof failed!
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
[2023-09-19_00:18:39] Started config rtlwmbus[00000001] listening on t1,c1 using fq 868.95M
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(serialcmd) opened /bin/sh pid 19545 fd 8 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(serialcmd) closed /bin/sh pid=19545 fd=8 (rtlwmbus)
(wmbusmeters) version: 1.14.0-31-gce6e282
(config) store meter files in: "/var/snap/wmbusmeters/common/logs/meter_readings"
(config) using device: rtlwmbus:868.95M:t1,c1
(config) number of meters: 233
[2023-09-19_00:19:10] Started config rtlwmbus[00000001] listening on t1,c1 using fq 868.95M
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(serialcmd) opened /bin/sh pid 19708 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(meter) created K2_P2_L024_CO_82867940 kamheat 82867940 encrypted
(meter) used meter template K2_P2_L024_CO_82867940 82867940 kamheat to match 82867940
(meter) started meter 1 (K2_P2_L024_CO_82867940 82867940 kamheat)
(meter) K2_P2_L024_CO_82867940(1) kamheat  handling telegram from 82867940
(telegram) DLL L=3e C=44 (from meter SND_NR) M=2c2d (KAM) A=82867940 VER=40 TYPE=0c (Heat volume at inlet meter) (driver kamheat) DEV=rtlwmbus[00000001] RSSI=141
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=6c SN=e0274a20 (AES_CTR session=0 time=303742) CRC=d940
(telegram) TPL CI=78

dmesg: 

[Tue Sep 19 00:18:39 2023] audit: type=1400 audit(1695075519.536:34): apparmor="DENIED" operation="exec" profile="snap.wmbusmeters.wmbusmeters" name="/usr/sbin/killall5" pid=19537 comm="wmbusmeters" requested_mask="x" denied_mask="x" fsuid=0 ouid=0
[Tue Sep 19 00:19:10 2023] r820t 12-001a: destroying instance
[Tue Sep 19 00:19:10 2023] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.4' successfully deinitialized and disconnected

weetmuts, 2023-09-22:
Thank you! This is a very important discovery! Apparently there are solutions for this:
https://sdr-enthusiasts.gitbook.io/ads-b/setting-up-rtl-sdrs/blacklist-kernel-modules

raszuk, 2023-09-22:
Ok very cool - I did see this page before when I started to play with wmbusmeters originally few years back and even applied this under normal ubuntu. 

Here I am under snap but I guess this is main ubuntu which is "stealing" the SDR ... so I presume you recommend to remove permanently those kernel modules from main host os manually ?

raszuk, 2023-09-22:
Tried and noticed those issues on Ubuntu Core snap

raszuk@ubuntu:~/log$ sudo depmod -a
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.dep.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.dep.bin.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.alias.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.alias.bin.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.softdep.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.symbols.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.symbols.bin.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.builtin.bin.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.builtin.alias.bin.18778.34601.1695390666, 301, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/5.15.0-1037-raspi, modules.devname.18778.34601.1695390666, 301, 644): Read-only file system

raszuk@ubuntu:~/log$ sudo update-initramfs -u
sudo: update-initramfs: command not found
raszuk@ubuntu:~/log$

raszuk, 2023-09-23:
Ok so apparently the provided solution will not work on snap Ubuntu Core. I did some research and found this statement:

closed 2023-12-24 by github-actions[bot]