wmbusmeters/wmbusmeters #1038: No telegrams from heat meters

FlavorFx, 2023-09-20:
### Type of request

Question

### OS version

Hass 

### wmbusmeters version

Home Assistant Add On

### Your message goes here

I setup wmbusmeters according to the documentation but I can't see any telegram in the log. But I can see telegrams when I use Wireless M-Bus Software. 

This is my hardware configuration:

Raspberry Pi 4
Dongle: IMST 871a (im871a) FW V1.6
Heat meters: Deltamess which is an OEM version of Qundis Q heat 5.5
Adapter Q module 5.5 heat (868MHz)
Home Assistant AddOn

I also changed the default configuration `device=auto:t1` to `device=auto:c1` but no telegrams

This is the log when I started Wmbusmeters HA AddOn:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
[14:47:26] INFO: CONFIG_CONF ...
[14:47:26] INFO: {"device":"auto:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[14:47:26] INFO: CONFIG_METERS ...
[14:47:26] INFO: [{"driver":"amiplus","id":"12345678","key":"NOKEY","name":"ExampleMeter"}]
[14:47:26] INFO: Syncing wmbusmeters configuration ...
[14:47:26] INFO: Registering meters ...
[14:47:26] INFO: Adding meter-0001 ...
[14:47:26] INFO: Generating MQTT configuration ... 
[14:47:28] INFO: Broker core-mosquitto will be used.
[14:47:28] INFO: 
MQTT Discovery ...
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
[14:47:30] INFO:  Adding meter: ExampleMeter ...
[14:47:30] INFO:   File /config/wmbusmeters/etc/mqtt_discovery/amiplus.json not found.
[14:47:30] INFO: MQTT Discovery cleanup...
[14:47:31] INFO: Starting web configuration service.
[14:47:31] INFO: Running wmbusmeters ...
Started auto im871a[00104277] on /dev/ttyUSB0 listening on t1
[14:47:33] INFO: Starting Nginx...
2023/09/20 14:47:33 [notice] 83#83: using the "epoll" event method
2023/09/20 14:47:33 [notice] 83#83: nginx/1.24.0
2023/09/20 14:47:33 [notice] 83#83: OS: Linux 6.1.21-v8
2023/09/20 14:47:33 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/09/20 14:47:33 [notice] 83#83: start worker processes
2023/09/20 14:47:33 [notice] 83#83: start worker process 250
```

This is the log when I use Wireless M-Bus Software

[image]

This is the log file

```
20.09.2023;12:08:25.33;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2E 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x09 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0x8B 0x00 ;;
20.09.2023;12:08:55.33;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2E 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x0A 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0x6D 0x00 ;;
20.09.2023;12:09:25.34;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2E 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x0A 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0x4F 0x00 ;;
20.09.2023;12:09:55.32;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2E 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x0B 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0x31 0x00 ;;
20.09.2023;12:10:35.34;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2E 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x0C 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0x09 0x00 ;;
20.09.2023;12:11:05.34;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2F 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x0C 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0xB0 0x01 ;;
20.09.2023;12:11:35.34;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2F 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x0D 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0x92 0x01 ;;
20.09.2023;12:12:05.33;n.a.;n.a.;0x32 (50);0x46 (70);0x93 0x44;0x76 0x48 0x17 0x37;0x23 (35);0x37 (55);0x72 0x64 0x95 0x24 0x67 0x93 0x44 0x23 0x04 0x2F 0x00 0x08 0x00 0x04 0xED 0x39 0x06 0x0B 0xF4 0x29 0x01 0xFD 0x0C 0x10 0x04 0x6D 0x0D 0x0A 0xF4 0x29 0x02 0xFD 0x3C 0xC2 0x01 0x02 0xFD 0xAC 0x7E 0x74 0x01 ;;
```
What's wrong?

BIBOLV, 2023-09-22:
You need to remove demo meter, then you will see all received telegrams or enable debug loglevel.

FlavorFx, 2023-09-24:
Thanks for the hint. I removed the demo meter and enabled the debug log level. Unfortunately, still no telegrams. The communication to my M-Bus USB adapter seems to work properly, but why I'm not able to receive any massages? I appreciate any help.

kxn25, 2023-09-26:
Hello,
Which one should i buy? 
iam using the qundius heat 5.5!

FlavorFx, 2023-09-26:
Finally, it works. Thanks for the hint. I replaced the demo meter, chanced the device to im871a:c1,t1 and determine the driver, id, and key, traced the mqtt messages and setup a mqtt sensor in HA. Great!

closed 2023-09-26 by BIBOLV