wmbusmeters/wmbusmeters #1039: segfault

raszuk, 2023-09-20:
### Type of request

Bug

### OS version

snap @ Ubuntu 20.04.4 LTS

### wmbusmeters version

1.14.0-31

### Your message goes here

[wto wrz 19 22:35:44 2023] rtl_wmbus[1711230]: segfault at 7fff925da0ff ip 00007fe48d1a9aab sp 00007ffe925d9f68 error 4 in libc.so.6[7fe48d031000+195000]

[wto wrz 19 22:35:44 2023] Code: 00 00 00 48 3b 15 1d fb 07 00 0f 87 dc 01 00 00 4c 8d 04 11 49 31 c8 49 c1 e8 3f 81 e1 00 0f 00 00 44 01 c1 0f 84 8a 00 00 00 <c5> fe 6f 6c 16 e0 c5 fe 6f 74 16 c0 48 89 f9 48 83 cf 1f c5 fe 6f

[wto wrz 19 22:35:45 2023] audit: type=1400 audit(1695155707.155:451): apparmor="DENIED" operation="exec" profile="snap.wmbusmeters.wmbusmeters" name="/usr/sbin/killall5" pid=1911730 comm="wmbusmeters" requested_mask="x" denied_mask="x" fsuid=0 ouid=0

After the segfault wmbus does not self recover under snap.

raszuk, 2023-10-04:
The problem continues ... 

8 48 8d b3
[wto paz  3 04:55:07 2023] wmbusmeters[159506]: segfault at 0 ip 000055c76cb74fca sp 00007f6a8f1da3c0 error 4 in wmbusmeters[55c76cb52000+ec000]
[wto paz  3 04:55:07 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 04:56:42 2023] wmbusmeters[159812]: segfault at 0 ip 000055dd8bbc8fca sp 00007f544179a3c0 error 4 in wmbusmeters[55dd8bba6000+ec000]
[wto paz  3 04:56:42 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 04:58:17 2023] wmbusmeters[160109]: segfault at 0 ip 0000558e73370fca sp 00007f10023023c0 error 4 in wmbusmeters[558e7334e000+ec000]
[wto paz  3 04:58:17 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 04:59:52 2023] wmbusmeters[160410]: segfault at 0 ip 00005633ee0d8fca sp 00007fb67ec2f3c0 error 4 in wmbusmeters[5633ee0b6000+ec000]
[wto paz  3 04:59:52 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:01:27 2023] wmbusmeters[160694]: segfault at 0 ip 00005599de0cefca sp 00007f115e3223c0 error 4 in wmbusmeters[5599de0ac000+ec000]
[wto paz  3 05:01:27 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:03:02 2023] wmbusmeters[160996]: segfault at 0 ip 0000562e5a024fca sp 00007f6bb37653c0 error 4 in wmbusmeters[562e5a002000+ec000]
[wto paz  3 05:03:02 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:06:12 2023] wmbusmeters[161299]: segfault at 0 ip 000055ad57c22fca sp 00007fdd6d9e93c0 error 4 in wmbusmeters[55ad57c00000+ec000]
[wto paz  3 05:06:12 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:07:47 2023] wmbusmeters[161850]: segfault at 0 ip 0000558b85d92fca sp 00007f3d350273c0 error 4 in wmbusmeters[558b85d70000+ec000]
[wto paz  3 05:07:47 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:09:22 2023] wmbusmeters[162148]: segfault at 0 ip 000055b07abaefca sp 00007fa4e968e3c0 error 4 in wmbusmeters[55b07ab8c000+ec000]
[wto paz  3 05:09:22 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:10:57 2023] wmbusmeters[162454]: segfault at 0 ip 000055abc5e5efca sp 00007f8a9fb5f3c0 error 4 in wmbusmeters[55abc5e3c000+ec000]
[wto paz  3 05:10:57 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:12:32 2023] wmbusmeters[162766]: segfault at 0 ip 0000555e532cbfca sp 00007f1064d8f3c0 error 4 in wmbusmeters[555e532a9000+ec000]
[wto paz  3 05:12:32 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:14:08 2023] wmbusmeters[163062]: segfault at 0 ip 0000559f1ef04fca sp 00007f2c2c03e3c0 error 4 in wmbusmeters[559f1eee2000+ec000]
[wto paz  3 05:14:08 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:15:43 2023] wmbusmeters[163351]: segfault at 0 ip 0000564569eccfca sp 00007f310f2b33c0 error 4 in wmbusmeters[564569eaa000+ec000]
[wto paz  3 05:15:43 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:17:18 2023] wmbusmeters[163655]: segfault at 0 ip 00005581d7fb5fca sp 00007f99e57d23c0 error 4 in wmbusmeters[5581d7f93000+ec000]
[wto paz  3 05:17:18 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[wto paz  3 05:20:29 2023] wmbusmeters[163960]: segfault at 0 ip 000055cecf153fca sp 00007f7ab7d973c0 error 4 in wmbusmeters[55cecf131000+ec000]
[... cut ...]
[sro paz  4 03:49:14 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3
[sro paz  4 03:50:53 2023] wmbusmeters[410041]: segfault at 0 ip 0000555a5bfa5fca sp 00007f0c174213c0 error 4 in wmbusmeters[555a5bf83000+ec000]
[sro paz  4 03:50:53 2023] Code: 83 83 80 00 00 00 10 eb 12 4c 8d 43 78 48 89 f2 48 89 fe 4c 89 c7 e8 57 fe ff ff 48 8b 7d 00 48 8b b3 80 00 00 00 48 2b 73 78 <48> 8b 07 48 c1 fe 04 ff 50 08 4c 8b 65 00 48 8d 6c 24 08 48 8d b3

raszuk, 2023-10-04:
And what is new now - even restarting (sudo snap restart wmbusmeters) does not help ... 

Looks to me like there are too many issues with Ubuntu Core and snap based services. Far from production ready IMO.

weetmuts, 2023-10-14:
Problematic. I need the telegram that caused the segfault. Can you build from scratch instead of using snap? (snapiscrap :-/ )

If you can build from scratch, do "make DEBUG=true"
then install the build_debug/wmbusmeters and build_debug/wmbusmetersd into /usr/bin
these will print a nice stack trace when they crash.

closed 2024-01-13 by github-actions[bot]