wmbusmeters/wmbusmeters #1040: Wired Mbus TCP support

spectrality-github, 2023-09-30:
### Type of request

Enhancement

### OS version

HA addon

### wmbusmeters version

Latest

### Your message goes here

   I have my Mbus master USB adapter (from AliExpress) connected to OpenWrt device with `ser2net` running.
And  I'm interested in native Mbus TCP support in wmbusmeters, as it is implemented in libmbus.
`mbus-tcp-request-data`
http://www.rscada.se/libmbus/

 Currently trying to workaround lack of Mbus TCP support by adding and starting `socat` inside wmbusmeters-ha-addon container. Like described here: https://www.lucid-control.com/accessing-usb-io-module-network-linux-socat/ 
 
 Next line is added before starting wmbusmeters in run.sh:
`socat -d -d pty,group-late=tty,link=/root/ttyMBUS0,mode=660,rawer,echo=0,b2400,waitslave,ignoreeof tcp:192.168.x.x:2003 &`
And using `/root/ttyMBUS0` as device in wmbusmeters config.
ser2net config is:
`2003:raw:30:/dev/mbusUSB0:2400 8DATABITS EVEN 1STOPBIT -RTSCTS LOCAL -XONXOFF`

But socat process is stopping and not restarting when ser2net is closing TCP connection by timeout or connection problems and whole container need to be restarted to restore connectivity. Also in this case ser2net timeout should be alwai configured longer that pool interval. 

Also there is problem with MQTT discovery form wired Mbus devices
 As colons are required in driver name (amiplus:mbus) for wired Mbus devices, but should be removed to match discovery template files.
```
MQTT Discovery ...
 Adding meter: MyMeter ...
  File /config/wmbusmeters/etc/mqtt_discovery/amiplus:mbus.json not found.
MQTT Discovery cleanup...
```
 I fixed that in my private fork with additional line in mqtt_discovery.sh
`aryKV["driver"]=${aryKV["driver"]%%:*}`
https://github.com/spectrality-github/wmbusmeters-ha-addon/blob/9351b966d42ccfe838aa140dc314b67b0fb45570/wmbusmeters-ha-addon-edge/mqtt_discovery.sh#L42
It word be good if this will be fixed in Main branch.

 I need to create new driver and discovery template for my NZR EcoCount Compact electricity meters as amiplus driver is not fully matching. But I don't know how to wmbusmeters to read two Mbus telegrams in a row as meter splits response after SND_NKE it should be read two times with  REQ_UD2

`68 22 22 68 08 0B 72 59 14 91 07 52 3B 00 02 58 00 00 00 8C 10 05 85 76 49 00 84 00 29 2F 6A 01 00 01 FD 17 00 1F D1 16` 
As I understood last '1F' record means 'More records follow' as libmbus decodes it.
and second telegram contains additional parameters. 
`68 4F 4F 68 08 0B 72 59 14 91 07 52 3B 00 02 59 00 00 00 84 00 29 21 74 01 00 84 01 29 9B 5B 00 00 84 02 29 5C DF 00 00 84 03 29 2A 39 00 00 82 01 FD 49 EC 00 82 02 FD 49 E9 00 82 03 FD 49 E9 00 82 01 FD 5A 78 00 82 02 FD 5A 03 01 82 03 FD 5A 4D 00 CC 16 `

[image]
[image]
[image]

`mbus-tcp-request-data -d 192.168.x.x 2003 11
[2023-09-30 20:34:53Z] SEND (005): 10 5B 0B 66 16
[2023-09-30 20:34:53Z] RECV (040): 68 22 22 68 08 0B 72 59 14 91 07 52 3B 00 02 22 00 00 00 8C 10 05 65 80 49 00 84 00 29 42 45 01 00 01 FD 17 00 1F 73 16
mbus_frame_print: Dumping M-Bus frame [type 4, 40 bytes]: 68 22 22 68 08 0B 72 59 14 91 07 52 3B 00 02 22 00 00 00 8C 10 05 65 80 49 00 84 00 29 42 45 01 00 01 FD 17 00 1F 73 16
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>7911459</Id>
        <Manufacturer>NZR</Manufacturer>
        <Version>0</Version>
        <ProductName></ProductName>
        <Medium>Electricity</Medium>
        <AccessNumber>34</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>1</Tariff>
        <Device>0</Device>
        <Unit>Energy (100 Wh)</Unit>
        <Value>498065</Value>
        <Timestamp>2023-09-30T20:34:53Z</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>0</Tariff>
        <Device>0</Device>
        <Unit>Power (1e-2 W)</Unit>
        <Value>83266</Value>
        <Timestamp>2023-09-30T20:34:53Z</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Error flags</Unit>
        <Value>0</Value>
        <Timestamp>2023-09-30T20:34:53Z</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>More records follow</Function>
        <Value></Value>
        <Timestamp>2023-09-30T20:34:53Z</Timestamp>
    </DataRecord>

</MBusData>`

BIBOLV, 2023-10-05:
Please create PR with changes to have fixes in main branch.

spectrality-github, 2023-10-08:
I'm not very good with GIT (just learning) and I don't know how to create PR just for one line, as all other changes in my fork is just for my experiments which are not working properly yet.
Maybe it will be easier to include this one line in mqtt_discovery.sh to someone else PR.
`aryKV["driver"]=${aryKV["driver"]%%:*}`
https://github.com/spectrality-github/wmbusmeters-ha-addon/blob/9351b966d42ccfe838aa140dc314b67b0fb45570/wmbusmeters-ha-addon-edge/mqtt_discovery.sh#L42

Also initially this issue was opened to request native Wired Mbus TCP support as it implemented in libmbus, which for sure I will not be able do myself.
But if you think that integrating socat to HA addon container is good idea, I can try to implement this and create PR.

 Maybe I should create separate issue regarding new driver creation for NZR EcoCount Compact electricity meters?
Currently I'm stuck and don't know what should be changed in driver to read multiply responses from meter after sending NKE.
Is it possible modify driver to read additional message from meter if last message is ending with '1F' record, which means 'More records follow' as libmbus decodes it.
  Also is it any way in driver first to send message to meter and then read response, as for example with wired MBUS you can  sent to meter request for specific data and then read response.

closed 2024-01-07 by github-actions[bot]

weetmuts, 2026-11-02:
Support now exists: https://github.com/wmbusmeters/wmbusmeters/issues/2162