wmbusmeters/wmbusmeters #1042: How to disable "regular reset"

raszuk, 2023-10-04:
### Type of request

Bug

### OS version

Description:    Ubuntu 20.04.4 LTS

### wmbusmeters version

1.14.0-31

### Your message goes here

How to disable regular reset: 

(main) regular reset of rtlwmbus  will happen every 82800 seconds****

Problems with wmbusmeters occur when this is triggered.

BIBOLV, 2023-10-05:
Try to set `--resetafter=<time>` to something very high - year or something, but dongles sometimes hangs and this was implemented for that reason. You can also play around with `--alarmtimeout=<time>` and `--alarmshell=<cmdline>`

raszuk, 2023-10-05:
Many thx - very helpful. 

How should I see such alarms in the log which would trigger alarmshell action ? 

Any hints on solving it properly with snap based OS (ubuntu core)  where all rootfs is read only and I can not delete permanently or disable any kernel modules ? 

Any plans to support wmbusmeters in conda ?

BIBOLV, 2023-10-06:
alarmtimeout is time interval where no telegrams are received, say, if you know that your meter reports every 10 mins, then you can set it to 20 mins - that means that you are not receiving two cycles of telegrams, then you can set alarmshell - what action to do. for that you need to think how to achieve what is needed. One option to try is to install usb-reset snap and try to call it, but no idea if it will work - you need to test.

conda is not in plans currently, but if you are willing to create and maintain it, you are more than welcome to create PR 😄

raszuk, 2023-10-06:
Ok many thx I will play with alarm stuff over the weekend. I am thinking to simply reset the wmbusmeters (sudo snap restart wmbusmeters) as this always fixes the issue manually.

raszuk, 2023-10-10:
Unfortunately wmbusmeters in snap ubuntu core is still failing in spite of all above recommendations. 

I see this in log: 

(serialcmd) closed /bin/sh pid=8579 fd=5 (rtlwmbus)
[2023-10-09_17:29:48] Lost 00000001 closing rtlwmbus[00000001]
(serialcmd) closed /bin/sh pid=8579 fd=-1 (rtlwmbus)
No wmbus device detected, waiting for a device to be plugged in.
[2023-10-09_17:29:48] (shell) invoking /bin/pidof failed!

And this in dmesg: 

[Mon Oct  9 17:29:47 2023] usb 1-1.2: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[Mon Oct  9 17:29:47 2023] usb 1-1.2: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[Mon Oct  9 17:29:47 2023] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
[Mon Oct  9 17:29:47 2023] usb 1-1.2: media controller created
[Mon Oct  9 17:29:47 2023] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[Mon Oct  9 17:29:47 2023] i2c i2c-11: Added multiplexed i2c bus 12
[Mon Oct  9 17:29:47 2023] rtl2832 11-0010: Realtek RTL2832 successfully attached
[Mon Oct  9 17:29:47 2023] usb 1-1.2: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[Mon Oct  9 17:29:47 2023] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
[Mon Oct  9 17:29:47 2023] r820t 12-001a: creating new instance
[Mon Oct  9 17:29:47 2023] r820t 12-001a: Rafael Micro r820t successfully identified
[Mon Oct  9 17:29:47 2023] rtl2832_sdr rtl2832_sdr.0.auto: Registered as swradio0
[Mon Oct  9 17:29:47 2023] rtl2832_sdr rtl2832_sdr.0.auto: Realtek RTL2832 SDR attached
[Mon Oct  9 17:29:47 2023] rtl2832_sdr rtl2832_sdr.0.auto: SDR API is still slightly experimental and functionality changes may follow
[Mon Oct  9 17:29:47 2023] Registered IR keymap rc-empty
[Mon Oct  9 17:29:47 2023] rc rc0: Realtek RTL2832U reference design as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/rc/rc0
[Mon Oct  9 17:29:47 2023] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[Mon Oct  9 17:29:47 2023] input: Realtek RTL2832U reference design as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/rc/rc0/input12
[Mon Oct  9 17:29:47 2023] usb 1-1.2: dvb_usb_v2: schedule remote query interval to 200 msecs
[Mon Oct  9 17:29:47 2023] usb 1-1.2: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[Mon Oct  9 17:29:48 2023] audit: type=1400 audit(1696865388.918:54): apparmor="DENIED" operation="exec" profile="snap.wmbusmeters.wmbusmeters" name="/usr/sbin/killall5" pid=5565 comm="wmbusmeters" requested_mask="x" denied_mask="x" fsuid=0 ouid=0

Any further hints ?

raszuk, 2023-10-10:
Upon restart wmbusmeters in snap I see this log: 

(serial) event loop stopped!
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
[2023-10-10_15:21:47] (shell) invoking /bin/pidof failed!
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
[2023-10-10_15:21:47] Started config rtlwmbus[00000001] listening on t1,c1 using fq 868.95M
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(serialcmd) opened /bin/sh pid 6404 fd 8 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 36000000 seconds
(serialcmd) closed /bin/sh pid=6404 fd=8 (rtlwmbus)
(wmbusmeters) version: 1.14.0
(config) store meter files in: "/var/snap/wmbusmeters/common/logs/meter_readings"
(config) using device: rtlwmbus:868.95M:t1,c1
(config) number of meters: 233
[2023-10-10_15:22:18] Started config rtlwmbus[00000001] listening on t1,c1 using fq 868.95M
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(serialcmd) opened /bin/sh pid 6570 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 36000000 seconds

closed 2024-01-10 by github-actions[bot]