wmbusmeters/wmbusmeters #1044: Multi-component telegram?

marcelstoer, 2023-10-06:
### Type of request

Question

### OS version

RPi/Debian bullseye

### wmbusmeters version

latest, Git revision ce6e282

### Your message goes here

The utility installed a new gas meter this morning. I now see that the telegram seems to contain two components: a gas meter and a "Bus/System component".

All good with that? Is that to be expected?

```
pi@white-rose:~ $ wmbusmeters/build/wmbusmeters --logtelegrams /dev/ttyUSB0:im871a:t1,c1
Started config im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 20343320
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07) encrypted
                   ver: 0x68
                device: im871a[00103852]
                  rssi: -76 dBm
                driver: iperl
telegram=|_2E44AE4C2033342068077A3E00200505DF53A73990A9DAAD01965B75EC7006BDC7965CE0826D21D41E94AC7214E18D|+2
Received telegram from: 00607474
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Bus/System component (0x0e) encrypted
                   ver: 0x02
      Concerning meter: 06011112
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Gas meter (0x03) encrypted
                   ver: 0x36
                device: im871a[00103852]
                  rssi: -77 dBm
                driver: unknown!
telegram=|_3644E61E74746000020E7212110106E61E3603303020657BB1926F5922ED14FCE36FA97AE21A54D2A47C164D6E145D29EA4A5F5AEE5EFE|+14
```

marcelstoer, 2023-11-05:
@weetmuts @BIBOLV I am a little lost with this. According to the [product brochure page 2](https://products.gwf.ch/wp-content/uploads/2018/10/RCM_EPe40232.pdf) the transfer protocol is OMS. Yet, wmbusmeters reports the driver as unknown. The data is encrypted and I didn't get the key yet. I'm wondering if the key would even help given I didn't find a generic OMS driver (I also checked #347).

marcelstoer, 2024-01-05:
Hey bot, please keep this open.

BIBOLV, 2024-01-05:
Drivers are not recognized because data is encrypted. Once you have keys and have decrypted telegrams - check if  then they are recognized.

closed 2024-02-06 by marcelstoer