wmbusmeters/wmbusmeters #1049: amb 8465 usb dongle detected but no meters found

Gharlyk, 2023-10-11:
### Type of request

Question

### OS version

Ubuntu 22.04.3 LTS

### wmbusmeters version

1.12.0-31-gce6e282

### Your message goes here

Hello,

I am trying to read my heat meter using an amb8465 (I tried with two usb dongle (amb8462 Metis-I and amb8465 Metis -II)

It seems that wmbusmeters detects the usb dongle device without problem:

I am running the following command:
`wmbusmeters --debug /dev/ttyUSB0:amb8465:c1,t1`

> (wmbusmeters) version: 1.14.0-31-gce6e282
> (config) using device: /dev/ttyUSB0:amb8465:t1,c1 
> (config) number of meters: 0
> (lookup) with file/hex "/dev/ttyUSB0" amb8465
> (serial) check if /dev/ttyUSB0 can be accessed
> (serial) tty /dev/ttyUSB0 can be accessed
> (serial) EVENT thread interrupted
> (serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
> (serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
> (serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465/amb3665)
> (serial) received binary ""
> (amb8465/3665) sending 6 bytes attempt 0
> (serial /dev/ttyUSB0) sent "FF0A02008077"
> (amb8465/3665) sent 6 bytes OK
> (serial) EVENT thread interrupted
> (amb8465/3665) reading response... 1
> (serial) received binary "29FF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF01030000FFFFFFFFFF"
> (amb8465) not enough data for decoding ConfigAMB8465!
> (amb8465/3665) reading response... 2
> (serial) received binary "FFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF69"
> (amb8465) proprely decoded ConfigAMB8465 response. Content: uart_cmd_out_enable=01 id=00000000 media=00 version=00 c_field=44 auto_rssi=01 b1_add_disable=01
> (amb8465) found response at offset 1
> (serialtty) closed /dev/ttyUSB0 (detect amb8465/amb3665)
> (amb8465) detect uart_cmd_out_enable=01 id=00000000 media=00 version=00 c_field=44 auto_rssi=01 b1_add_disable=01
> (amb8465) are you there? yes 00000000
> (serial) EVENT thread interrupted
> (main) opening /dev/ttyUSB0:amb8465:t1,c1
> Started config amb8465[00000000] on /dev/ttyUSB0 listening on t1,c1
> (amb8465) on /dev/ttyUSB0
> (serial) EVENT thread interrupted
> (serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
> (serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
> (serialtty) opened /dev/ttyUSB0 fd 3 (amb8465)
> (amb8465) get config
> (serial /dev/ttyUSB0) sent "FF0A02008077"
> (serial) EVENT thread interrupted
> (serial) received binary "FF8A7A007800807102000100"
> (amb8465) checkAMB8465Frame "FF8A7A007800807102000100"
> (amb8465) not enough bytes yet, partial command response 12 126.
> (serial) received binary "00FFFFFA00FFFFFFFFFFFFFFFFFFFF"
> (amb8465) chunk time 16 msec
> (amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFF"
> (amb8465) not enough bytes yet, partial command response 27 126.
> (serial) received binary "FFFFFFFFFFFF0C3200026400FFFFFF"
> (amb8465) chunk time 15 msec
> (amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFF"
> (amb8465) not enough bytes yet, partial command response 42 126.
[... 209 lines cut ...]
> ^C(serial) stopping manager
> (serial) MAIN thread interrupted
> (serial) EVENT thread interrupted
> (serial) event loop stopped!
> (serial) TIMER thread interrupted
> (serial) closing 1 devices
> (wmbus) disconnected /dev/ttyUSB0 amb8465
> (serialtty) closed /dev/ttyUSB0 (amb8465)
> (wmbus) deleted amb8465
> (serial) closing 1 devices

However I don't manage to receive the message saying that a meter as been found : 
> Received telegram from: 12345678
>           manufacturer: (MMC) My Meter Company (0x1234)
>                   type: Cold water meter (0x16)
>                    ver: 0x01
>                   device: amb8465[00000000]
>                   rssi: -39 dBm
>                   driver: unknown

I don't get why I am not having messages saying that he has found a heatmeter, I am using the same dongle on windows with the dedicated program from the company and I see many devices...

Any suggestions to solve my issue ?

Many thanks for the support

weetmuts, 2023-10-11:
Aha! Can you try to change the source code in wmbus_amb8465.c with this diff:

-        int rssi_len = (rssi_expected_ && data[1] == (0x80|CMD_DATA_IND)) ? 1 : 0;
+        int rssi_len = (rssi_expected_ && data[1] == CMD_DATA_IND) ? 1 : 0;
 
@@ -816,6 +817,7 @@ void WMBusAmber::handleMessage(int msgid, vector<uchar> &frame, int rssi_dbm)
 {
     switch (msgid) {
     case (0):
+    case (CMD_DATA_IND):
     {

Your dongle might be in the command mode for receiving data, ie the payload is prefixed with ff03.
It seems like my code assumes the command prefix to be ff83, but the high bit is apparently
only set if its an actual response to a command. The incoming telegrams are not responses per se.

The fix should make it eat the wrapper prefix correctly. I have never actually tested this code
(which might explain the bug) since changing from transparent to command mode is something that has to be configured using writing to flash in the dongle.

Please test!

Gharlyk, 2023-10-11:
Hello,
Thanks for your quick reply, I am modifying it but I don't get what you want with:

> @@ -816,6 +817,7 @@ void WMBusAmber::handleMessage(int msgid, vector<uchar> &frame, int rssi_dbm)
>  {
>      switch (msgid) {
>      case (0):
> +    case (CMD_DATA_IND):
>      {

Should I remove case (0) with case (CMD_DATA_IND)? As I do not see any - symbol before case 0, I prefer to ask.

Or should I add case(CMD_DATA_IND) ? In this case what is to put inside ?

Thanks :)

weetmuts, 2023-10-11:
Add the new case and keep the old case (0) in the source. It must be there since transparent telegrams without ff03 will have msgid==0. It should looke like:
```
case (0):
case (CMD_DATA_IND):
```

weetmuts, 2023-10-11:
CMD_DATA_IND is defined to be 3.

Gharlyk, 2023-10-11:
Hello,
It works, I am receiving now messages from different meters ! (Still did not saw the one I am looking for ;)

This is the code modification I did as you requested :
```
        // Only response from CMD_DATA_IND has rssi
        //int rssi_len = (rssi_expected_ && data[1] == (0x80|CMD_DATA_IND)) ? 1 : 0;
        int rssi_len = (rssi_expected_ && data[1] == CMD_DATA_IND) ? 1 : 0;
```
```
void WMBusAmber::handleMessage(int msgid, vector<uchar> &frame, int rssi_dbm)

{
    switch (msgid) {
    case (0):
    case (CMD_DATA_IND):
    {
        AboutTelegram about("amb8465["+cached_device_id_+"]", rssi_dbm, FrameType::WMBUS);
        handleTelegram(about, frame);
        break;
    }
```

Many thanks for the support!

weetmuts, 2023-10-11:
Thanks for the report and testing help! Permanent fix push to github.
https://github.com/wmbusmeters/wmbusmeters/commit/6e3bac97d480748e6d548b47f7728246828c8e66

Gharlyk, 2023-10-16:
Hello again ;)

The program is working perfectly with the dongle amb8465 METIS-I, I have different meters configured I can manage to receive them all, it's really great !
However I wanted to try with my other dongle the amb8465 METIS-II (I have a bigger antenna, which allows me to stay at my desk and receive all telegrams) and I see with the debug that the METIS-II is receiving telegrams but nothing is decoded hence no meters readings available.

log file :
```
(config) number of meters: 7
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 1
(shell) 5920 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(serial) EVENT thread interrupted
(shell) output: >>>5916
<<<
(shell) return code 0
(shell) 5921 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465/3665) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465/3665) reading response... 1
(serial) received binary "FF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF01030000FFFFFFFFFFFFFF0000"
(amb8465) not enough data for decoding ConfigAMB8465!
(amb8465/3665) reading response... 2
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF69"
(amb8465) proprely decoded ConfigAMB8465 response. Content: uart_cmd_out_enable=01 id=00000000 media=00 version=00 c_field=44 auto_rssi=01 b1_add_disable=01
(amb8465) found response at offset 0
(serialtty) closed /dev/ttyUSB0 (detect amb8465/amb3665)
(serial) EVENT thread interrupted
(amb8465) detect uart_cmd_out_enable=01 id=00000000 media=00 version=00 c_field=44 auto_rssi=01 b1_add_disable=01
(amb8465) are you there? yes 00000000
(main) opening c1
Started auto amb8465[00000000] on /dev/ttyUSB0 listening on c1
(amb8465) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 6 (amb8465)
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "FF8A7A00780080710200010000FF"
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FF"
(amb8465) not enough bytes yet, partial command response 14 126.
(serial) received binary "FFFA00FFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 30 126.
(serial) received binary "FFFFFF0C3200026400FFFFFFFFFF01"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF01"
(amb8465) not enough bytes yet, partial command response 45 126.
[... 74 lines cut ...]
(amb8465) checkAMB8465Frame "FF034A44C5143455052500047A3C003025E39041"
(amb8465) not enough bytes yet, partial command response 20 79.
(serial) received binary "6A9290FE30838D864C7FC58E55CCF0"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF034A44C5143455052500047A3C003025E390416A9290FE30838D864C7FC58E55CCF0"
(amb8465) not enough bytes yet, partial command response 35 79.
(serial) received binary "CF31EFE3E236A045D774743605BDD399"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF034A44C5143455052500047A3C003025E390416A9290FE30838D864C7FC58E55CCF0CF31EFE3E236A045D774743605BDD399"
(amb8465) not enough bytes yet, partial command response 51 79.
(serial) received binary "69A614BD0C177037CE5A08305ABF03"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF034A44C5143455052500047A3C003025E390416A9290FE30838D864C7FC58E55CCF0CF31EFE3E236A045D774743605BDD39969A614BD0C177037CE5A08305ABF03"
(amb8465) not enough bytes yet, partial command response 66 79.
(serial) received binary "FD0C05010002FD0B201112BA"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF034A44C5143455052500047A3C003025E390416A9290FE30838D864C7FC58E55CCF0CF31EFE3E236A045D774743605BDD39969A614BD0C177037CE5A08305ABF03FD0C05010002FD0B201112BA"
(amb8465) not enough bytes yet, partial command response 78 79.
(serial) received binary "FF03"
(amb8465) rx long delay (15s), drop incomplete telegram
```

Is it linked to my previous post ? do you have a idea how to solve it ?

Many thanks for the support!

Gharlyk, 2023-12-19:
Hello @weetmuts , any suggestions how to solve the issue ?

Many thanks for your help.

weetmuts, 2024-02-18:
It seems like there is a one of bug here:

 checkAMB8465Frame "FF034A44C5143955052500047A31003025EE5793BF2A3D278935AFBD10B1A84F3D96268900C9EEB858FE5E3C2CABBE80BE2BFFEB9A2388771D44227773BA5A0CC503FD0C05010002FD0B2011122E"
(amb8465) not enough bytes yet, partial command response 78 79.

It thinks it is missing a single byte! :-)

Gharlyk, 2024-02-20:
Hi weetmuts,
Yes exactly, this is the issue, it never gets the full frame. Any idea where this could come from?
I stay available if you need more data 😉