wmbusmeters/wmbusmeters #1050: Please improve driver hydrus for Diehl Water Meter Hydrus 2.0

AnathosN, 2023-10-12:
Please see attached telegram analysis.
Meter is a Diehl Hydrus 2.0 using the hydrus driver. Unfortunately it cannot read all of the infos in the telegram.
If I unserstand it correctly it does only count full m3 while the meter should deliver 3 digits after the comma, allowing daily consumption measurement.
Thanks a lot!

https://wmbusmeters.org/analyze/5344a5111412757276078c0062900f002c25e92231003116a99e98826d477ae9803107102f2f0c163802000004fd17000008080a5a250202fd7448114c1633010000426cdf2c84046d3B37fe298c041632020000:hydrus

weetmuts, 2023-10-12:
From the analyze you can see that it only sends full m3 values. (Volume m³)
contrary to the temperature for which it sends 1 decimal. (Flow temperature 10⁻¹ °C)

weetmuts, 2023-10-12:
The meter itself tracks smaller values, my guess is they made a mistake configuring the wmbus telegrams this hydrus meter. This config should be for huge industries consuming multiple m3 per day. :-) Perhaps you can call them and ask?

Normally water meters use 0.001 resolution for households.

AnathosN, 2023-10-12:
> The meter itself tracks smaller values, my guess is they made a mistake configuring the wmbus telegrams this hydrus meter. This config should be for huge industries consuming multiple m3 per day. :-) Perhaps you can call them and ask?
> 
> Normally water meters use 0.001 resolution for households.

I have the mobile number of a guy at Diehls service, but he doesn't answer the phone atm. Will check later.
Or should I ask my water supplier directly if this is intentional?

AnathosN, 2023-10-12:
> From the analyze you can see that it only sends full m3 values. (Volume m³) contrary to the temperature for which it sends 1 decimal. (Flow temperature 10⁻¹ °C)

There are some other values in the output. I think one of them is m³ since last time the meter was read.
077   : 8C dif (8 digit BCD Instantaneous value)
078   : 04 dife (subunit=0 tariff=0 storagenr=8)
079   : 16 vif (Volume m³)
080 C!: 32020000 ("consumption_at_history_1_m3":232)
This can be deciphered only when I use driver evo868, which is suggested as best driver but doesn't provide the remaining battery life value.

Is there a chance we can integrate the additional values into the hydrus driver?

I will talk to the guy from the water supplier, if they can braodcast additional data, e.g. daily and monthly consumption if this is supported. Otherwise I think I can calculate it from the liters.

In order to calculate it, I would need to save the meter reading which lead to an error as I use the HA addon.

When I changed following values:
meterfiles=/var/lib/wmbusmeters/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name-id
logfile=/var/log/wmbusmeters/wmbusmeters.log
the addon stopped working.
I tried to put the files into the /wmbusmeters/ folder then which is set to be the source folder for wmbusmeters and is within the config folder of HA but I get a write error for the meter-readings folder.
AFAIK the folders within /config/ should be accessible for r/w without addition config.

Any suggestions?

Thanks a lot for your support! Really appreciated!

AnathosN, 2023-10-25:
https://wmbusmeters.org/analyze/6344a5111412757276078c0050900f002c25083232005dced0164a554f817a08c04107102f2f0c164102000004fd17000000080a5a640102fd74e5104c1633010000426cdf2c84046d3B37fe298c0416320200000B3B0000000e91715314468797412f2f:hydrus

So, the technical support of my water supplier activated the high detail volume output, but it seems to be a seperate field.
0E dif (12 digit BCD Instantaneous value)
090   : 91 vif (Volume 10⁻⁵ m³)
091   : 71 combinable vif (MultiplicativeCorrectionFactorForValue)
092 C?: 531446879741
Unfortunately the Hydrus driver cannot decode it.
Could you help me with this, please?

AnathosN, 2023-11-23:
Any update on this request? I would highly appreciate some help. Even if someone could tell me how to change the driver myself with the Home Assistant addon, I saw the driver Wiki for wmBUS Meters but I cannot find the driver files in my HassOS.

brunomelorp, 2026-11-04:
@AnathosN  How did you manage to read the data from the Diehl Metering meters? How did you obtain the encryption key? Would you be able to share the encryption key?

brunomelorp, 2026-11-04:
@AnathosN  Which module did you use to receive data from the Diehl Metering meter? An RTL-SDR? A CC1101?