wmbusmeters/wmbusmeters #1054: Device Support Itron EquaScan h-Koax RF I

huhndomi, 2023-10-14:
### Type of request

New device support

### Meter description

I would like to read four Itron EquaScan h-Coax RF I heat meters using wmbusmeters. These are not recognized in the current version 1.14.0. These heat meters are also sold in identical form by Engelmann in the SensoStar series. I assume that Engelmann is the manufacturer. I have attached the manufacturer's data sheets in the appendix.

I read the data using a Raspberry Pi3 and a nano CUL 868 MBUS with the firmware nanoCUL_r568_mbus_c1t1_bufsize300.hex

[P0413 EquaScan hKoaxRF_D_0720_II.pdf](https://github.com/wmbusmeters/wmbusmeters/files/12908106/P0413.EquaScan.hKoaxRF_D_0720_II.pdf)
[1080619044_2020_07_03_BA_S3_Itron_de.pdf](https://github.com/wmbusmeters/wmbusmeters/files/12908107/1080619044_2020_07_03_BA_S3_Itron_de.pdf)

### Logged telegrams from meter in decrypted format

```shell
(wmbusmeters) logging started 2023-10-14 23:13:26 1.14.0-32-g6e3bac9
(wmbusmeters) version: 1.14.0-32-g6e3bac9
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: /dev/ttyUSB0:cul:c1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" cul
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (detect cul)
(serial /dev/ttyUSB0) sent "560D0A"
(serial) EVENT thread interrupted
(serial) received binary "0A5620312E3637206E616E6F43554C3836385F723536380D0A"
(cul) probe response "<0A>V 1.67 nanoCUL868_r568<0D><0A>"
(serialtty) closed /dev/ttyUSB0 (detect cul)
(serial) EVENT thread interrupted
(cul) are you there? yes
(main) opening /dev/ttyUSB0:cul:c1
[2023-10-14_23:13:27] Started config cul on /dev/ttyUSB0 listening on c1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (cul)
(main) regular reset of cul  on /dev/ttyUSB0 will happen every 82800 seconds
(cul) set link mode c
(serial /dev/ttyUSB0) sent "6272630D0A"
(serial) EVENT thread interrupted
(serial) received binary "434D4F44450D0A"
(cul) checkCULFrame "CMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "CMODE"(serial /dev/ttyUSB0) sent "5832310D0A"
(wmbus) no alarm (expected activity) for cul
(serial) EVENT thread interrupted
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received binary "62593439343433373243393634"
(cul) checkCULFrame "bY4944372C964"
(cul) no eol found yet, partial frame
(serial) received binary "30303132333341313638443230423835333630433432313234364132453732304630363834413438453931423745303030323142343242433139"
(cul) checkCULFrame "bY4944372C964001233A168D20B85360C421246A2E720F0684A48E91B7E00021B42BC19"
(cul) no eol found yet, partial frame
(serial) received binary "39373843374133304443463938324631343634434631314633444539313235424434363733303142324134383945313530414641344545353841314437364638383836453534444137373739334645383532330D0A"
(cul) checkCULFrame "bY4944372C964001233A168D20B85360C421246A2E720F0684A48E91B7E00021B42BC19978C7A30DCF982F1464CF11F3DE9125BD467301B2A489E150AFA4EE58A1D76F8886E54DA77793FE8523<0D><0A>"
(cul) checkCULFrame RSSI_RAW=35
(cul) checkCULFrame LQI=66
(wmbus) trimming frame B "4944372C964001233A168D20B85360C421246A2E720F0684A48E91B7E00021B42BC19978C7A30DCF982F1464CF11F3DE9125BD467301B2A489E150AFA4EE58A1D76F8886E54DA77793FE"
(wmbus) ff b dll crc first 0-72 93fe ok
(wmbus) trimming frame B "4944372C964001233A168D20B85360C421246A2E720F0684A48E91B7E00021B42BC19978C7A30DCF982F1464CF11F3DE9125BD467301B2A489E150AFA4EE58A1D76F8886E54DA77793FE"
(wmbus) trimmed 2 dll crc bytes from frame b and ignored 0 suffix bytes.
(wmbus) trimmed frame B "4744372C964001233A168D20B85360C421246A2E720F0684A48E91B7E00021B42BC19978C7A30DCF982F1464CF11F3DE9125BD467301B2A489E150AFA4EE58A1D76F8886E54DA777"
[... 567 lines cut ...]
(telegram) DLL L=32 C=44 (from meter SND_NR) M=2c37 (KAW) A=23014096 VER=3a TYPE=16 (Cold water meter) (driver flowiq2200) DEV=cul RSSI=-56
(wmbus) parseELL @10 41
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=c6 SN=9260c421 (AES_CTR session=2 time=1852937) CRC=6ea9
Received telegram from: 23014096
          manufacturer: (KAW) Kamstrup, Denmark (0x2c37)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x3a
                device: cul
                  rssi: -56 dBm
                driver: flowiq2200
(wmbus) 000   : 32 length (50 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 372c dll-mfct (KAW)
(wmbus) 004   : 96400123 dll-id (23014096)
(wmbus) 008   : 3a dll-version
(wmbus) 009   : 16 dll-type (Cold water meter)
(wmbus) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : c6 ell-acc
(wmbus) 013   : 9260c421 sn (AES_CTR)
(wmbus) 017   : 6ea9 payload crc (calculated c36f ERROR)
(wmbus) 019 CE: 8A8D61516AA9D01A1ECFE01D4F7E6DB8DC9EE958CA4DCDFA3D047265EEAFA326 failed decryption. Wrong key?
telegram=|_3244372C964001233A168D20C69260C4216EA98A8D61516AA9D01A1ECFE01D4F7E6DB8DC9EE958CA4DCDFA3D047265EEAFA326|+225
(wmbus) telegram from 23014096 ignored by all configured meters!
```

huhndomi, 2023-10-16:
I have now also been able to record the telegrams from the heat meter using the IM871A USB stick. However, these do not seem to differ from that of the CUL stick. Add that anyway.

[Itron EquaScan log.txt](https://github.com/wmbusmeters/wmbusmeters/files/12921386/Itron.EquaScan.log.txt)

huhndomi, 2023-10-16:
I have now followed the instructions to create a new driver. I had the telegram analyzed as described. The following result came out.

```
telegram=|404497A66057751100047AA40000A0_04065C120000041353592200066D0A1836F02A004406680C0000426CDF2C0D7FEB0700000006040C5C80034E0C7937502321|+9

Auto driver  : not found!
Best driver  : enercal 20/36
Using driver : enercal 20/36
000   : 40 length (64 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 97a6 dll-mfct (ITW)
004   : 60577511 dll-id (11755760)
008   : 00 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : a4 tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 00a0 tpl-cfg a000 (bidirectional synchronous )
015   : 04 dif (32 Bit Integer/Binary Instantaneous value)
016   : 06 vif (Energy kWh)
017 C!: 5C120000 ("total_kwh":4700)
021   : 04 dif (32 Bit Integer/Binary Instantaneous value)
022   : 13 vif (Volume l)
023 C!: 53592200 ("total_m3":2251.091)
027   : 06 dif (48 Bit Integer/Binary Instantaneous value)
028   : 6D vif (Date and time type)
029 C!: 0A1836F02A00 ("meter_datetime":"2023-10-16 22:24:10")
035   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
036   : 06 vif (Energy kWh)
037 C!: 680C0000 ("target_kwh":3176)
041   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
042   : 6C vif (Date type G)
043 C!: DF2C ("target_date":"2022-12-31")
045   : 0D dif (variable length Instantaneous value)
046   : 7F vif (Manufacturer specific)
047   : EB varlen=235
048 C?: 0700000006040C5C80034E0C79375023
064   : 21 dif (8 Bit Integer/Binary Minimum value)

{
    "media":"heat",
    "meter":"enercal",
    "name":"",
    "id":"11755760",
    "target_kwh":3176,
    "total_kwh":4700,
    "total_m3":2251.091,
    "meter_datetime":"2023-10-16 22:24:10",
    "status":"OK",
    "target_date":"2022-12-31",
    "timestamp":"2023-10-16T21:28:25Z"
}
root@wmbus:~# wmbusmeters --analyze 404497A66057751100047AA40000A0_04065C120000041353592200066D0A1836F02A004406680C0000426CDF2C0D7FEB0700000006040C5C80034E0C7937502321

``` 

However, the entire telegram is never logged. At the end there is always +XX.
How can I have the entire telegram logged?

I have already logged the following telegrams:

telegram=|_404497A65857751100047A160000A00406744700000413FF152100066D343234F02A0044061E2F0000426CDF2C0D7FEB0700000006040C5C80034E0C7941502321|+64

telegram=|_404497A65657751100047AE20000A00406A505000004135EF80100066D133434F02A0044061B050000426CDF2C0D7FEB0700000006040C5C80034E0C7938502321|+110

telegram=|_404497A65757751100047A280000A00406764C00000413F6C48000066D293634F02A0044069A360000426CDF2C0D7FEB0700000006040C5C80034E0C7942502321|+129

telegram=|404497A65857751100047A290000A0_0406754700000413BA162100066D2F1936F02A0044061E2F0000426CDF2C0D7FEB0700000006040C5C80034E0C7941502321|+195

huhndomi, 2023-10-19:
I found a suitable driver for my heat meters. The “sensostar” driver worked immediately for this model. I get the following values from the counter every 5 minutes:

```
{"media":"heat","meter":"sensostar","name":"Waerme-EG-Fussboden","id":"11755757","consumption_1_months_ago_kwh":13978,"energy_consumption_at_reporting_date_kwh":13978,"total_kwh":19601,"total_water_m3":8461.322,"meter_timestamp":"2023-10-18 00:04:54","reporting_date":"2022-12-31","timestamp":"2023-10-17T23:03:48Z","device":"im871a[78563412]","rssi_dbm":-44}
{"media":"heat","meter":"sensostar","name":"Waerme-EG-Fussboden","id":"11755757","consumption_1_months_ago_kwh":13978,"energy_consumption_at_reporting_date_kwh":13978,"total_kwh":19601,"total_water_m3":8461.496,"meter_timestamp":"2023-10-18 00:09:52","reporting_date":"2022-12-31","timestamp":"2023-10-17T23:08:48Z","device":"im871a[78563412]","rssi_dbm":-44}
{"media":"heat","meter":"sensostar","name":"Waerme-EG-Fussboden","id":"11755757","consumption_1_months_ago_kwh":13978,"energy_consumption_at_reporting_date_kwh":13978,"total_kwh":19601,"total_water_m3":8461.496,"meter_timestamp":"2023-10-18 00:14:52","reporting_date":"2022-12-31","timestamp":"2023-10-17T23:13:47Z","device":"im871a[78563412]","rssi_dbm":-44}
``` 

The following values are read from the telegram:

- consumption_1_months_ago_kwh
- energy_consumption_at_reporting_date_kwh
- total_kwh
- total_water_m3
- meter_timestamp
- reporting_date

However, I would be interested in the following values:

- power
- flow-water
- forward
- return
- difference

Does anyone know how I can get these values?

Andi252, 2023-11-07:
> However, the entire telegram is never logged. At the end there is always +XX.

The '|+XX' at the end of the telegram is not a missing block of the telegram, it is some kind of time diff see:
https://github.com/wmbusmeters/wmbusmeters/blob/22d6880385a5b1c35aea43833b263e403e5bfe2d/src/util.cc#L1061

Andi252, 2024-01-12:
hello #1091 should now detect your heat meter correctly as ITW meter.

huhndomi, 2024-03-25:
Thank you. I can't test it because we have install a new headpump and renew the Energy Meters.

closed 2024-03-25 by huhndomi

weetmuts, 2024-06-19:
@huhndomi FYI @Andi252 wrote a new itronheat driver that is now integrated. Just so you know. :-)