wmbusmeters/wmbusmeters #1055: receive no Telegrams with "RWE Smart Home" which is in principle a IMST iM871A

huhndomi, 2023-10-14:
### Type of request

Question

### OS version

raspbean - Debian version: 12 (bookworm)

### wmbusmeters version

1.14.0

### Your message goes here

I do not receive any telegrams with the "RWE Smart Home" which is in principle a IMST iM871A USB receiver. With the same Raspberry Pi3 and a CUL 868 MBUS USB receiver I can receive telegrams (see issue "Device Support Itron EquaScan h-Koax RF I #1054").

Thanks to the tip on [iobroker](https://www.npmjs.com/package/iobroker.wireless-mbus?activeTab=readme) I was able to activate the stick:

IMST iM871A variant: There exists a "RWE Smart Home" USB receiver which is in principle a IMST iM871A, but the kernel will not automatically load the corresponding driver. This is a one-liner to create a udev rule to fix that:

`sudo bash -c "echo \$'ACTION==\"add\", ATTRS{idVendor}==\"10c4\", ATTRS{idProduct}==\"87ed\", RUN+=\"/sbin/modprobe cp210x\" RUN+=\"/bin/sh -c \\'echo 10c4 87ed > /sys/bus/usb-serial/drivers/cp210x/new_id\\'\"' > /etc/udev/rules.d/99-imst.rules"`

The stick has firmware 14 and is available in the system:
```
lsusb
Bus 001 Device 007: ID 10c4:87ed Silicon Labs IMST USB-Stick for Smart Meter
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

log:
```
(wmbusmeters) logging started 2023-10-14 23:39:10 1.14.0-32-g6e3bac9
(wmbusmeters) version: 1.14.0-32-g6e3bac9
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: /dev/ttyUSB0:im871a:c1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" im871a
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A5811008330014017F190000F1AA"
(im871a) checkIM871AFrame "A5811008330014017F190000F1AA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 aaf1 expected aaf1
(im871a) received full frame
(device info bytes) "330014017F190000"
(im871a/im170a) info: type=im871a mode=collector firmware=14 hci=01 uid=0000197f
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000700B32555150001010001FF07020000010000001524"
(im871a) checkIM871AFrame "A5810616FF000700B32555150001010001FF07020000010000001524"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 2415 expected 2415
(im871a) received full frame
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(device config bytes) "FF000700B32555150001010001FF0702000001000000"
(im871a/im170a) config: other link_mode=cab id=01001555 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 01001555 im871a firmware: 14
(main) opening /dev/ttyUSB0:im871a:c1
[2023-10-14_23:39:10] Started config im871a[01001555] on /dev/ttyUSB0 listening on c1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000700B32555150001010001FF07020000010000001524"
(im871a) checkIM871AFrame "A5810616FF000700B32555150001010001FF07020000010000001524"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 2415 expected 2415
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 01001555
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A5811008330014017F190000F1AA"
(im871a) checkIM871AFrame "A5811008330014017F190000F1AA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 aaf1 expected aaf1
(im871a) received full frame
(im871a) device info completed
(im871a) device info: type=im871a mode=collector firmware=14 hci=01 uid=0000197f
(im871a) set config to set link mode 07
(serial /dev/ttyUSB0) sent "A5010306000207300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
[2023-10-14_23:39:12] (memory) rss 1921024 peak 1.83 MiB
```

huhndomi, 2023-10-16:
I was able to solve the problem myself. An update of the stick's firmware to version 1.6 solved the problem. Now I can receive telegrams with this stick. Therefore the ticket can be closed.

I proceeded as follows:
1. Download the firmware and the Windows update tool from the [wireless-solutions.de](https://wireless-solutions.de/downloadfile/wireless-m-bus-software/) website.
2. Simply create a free account to download. The download is then possible
3. Download the necessary Windows drivers from [Silicon Labs](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads)
4. Since the stick uses a different hardware ID than the original IMST sticks, the driver must be assigned via the Windows Device Manager. To do this, select “Update drivers” and in the next point “Select from a list of available drivers on my computer”. Then select the path to the unpacked driver and confirm again that you are sure to actually use this driver.
5. Install the Windows tool “WM-Bus Firmware Updater”. After the driver has been successfully installed, the corresponding COM port can be selected and the stick can be read. If that works, you've done everything right. Then select and flash the appropriate firmware file “WMBusUSB iM871A-USB V1_6 BC13.hex”.
6. You can then test the stick using the Windows tool “Wireless M-Bus Studio”. I used the tool to reset the configuration of the stick to factory settings.
7. Stick connected to the Raspberry Pi again and started wmbusmeters. Telegrams finally arrive.

closed 2023-10-16 by huhndomi

weetmuts, 2023-10-17:
Oh, that is interesting. What is the change in the firmware, a bug fix?
Should we write something in the README about this?

huhndomi, 2023-10-19:
I can't say whether a bug fix solved the problem or whether the RWE stick had adapted firmware.
I think it makes sense to include this information in the Readme. It took me some time to figure it out.

Andi252, 2023-11-07:
I can confirm, the standard RWE stick does not receive any telegram.

With the update to the firmware 1.6 the stick immediately receives telegrams.

The stick was sold as a combo with a reading head for a power meter, therefor I think the firmware is restricted to telegrams from such a "reading head".

It was also quite a challange to update the firmware, because RWE changed the USB ID of the stick, thererfor the serial interface is not detected by WIndows 10. I had to create a custom driver for Windows 7 and update the stick there.