wmbusmeters/wmbusmeters #1056: ista sensonic 3 heat meter

avandorp, 2023-10-16:
### Type of request

New device support

### Meter description

ista sensonic 3 heat meter sending on c1, product number 4030020. (Can also be used for cold metering, I don't think the device I own is configured for that use case.)
(Attention device owners: Do not click randomly on the single button of the meter. If you are in the wireless service loop and click wrongly, you will activate istas proprietary wireless protocol. Only ista will be able to reset the device and activate the wireless m-bus. To activate the wireless m-bus you have to long click until 2A is shown, then wait shortly, click *once", wait until 2B is shown, *wait* again, single click until 2C is shown, then double click. Do not double click in 2A or 2B.)

If I find time  I will provide a new driver. Currently "microclima" works well enough (judging from the results I get on a meter showing 0 kWh). I will be able to tell more when the heating season starts.

### Logged telegrams from meter in decrypted format

```shell
5344742669315033A9048C2070900F002C25961200009D6949E80EB1E2707A96003007102F2F_0C0500000000426C00004C050000000082016CFE298C0105000000008C0115000000002F2F2F2F2F2F2F2F2F2F2F
```

morremeyer, 2023-11-13:
Note for others with this meter: If its managed by e.g. your landlord and 2A shows `read`, you will not be able to turn on MBus since the meter has been configured to use the proprietary "Ista Funk" mode.

xeonixo, 2024-01-02:
> Note for others with this meter: If its managed by e.g. your landlord and 2A shows `read`, you will not be able to turn on MBus since the meter has been configured to use the proprietary "Ista Funk" mode.

Our meters are read by ISTA dudes. Someone comes once a year and read it wireless with it's device. Is that already the proprietary mode? We have no further gateway or anything else here. I am really interested in the data from this device.

morremeyer, 2024-01-02:
@xeonixo Most likely yes.

If you want to check, run wmbusmeters without a meter list and save the telegrams to analyze them later. However, I have low hopes :/

xeonixo, 2024-01-03:
@morremeyer Unfortunately, I don't have any hardware available. I would have to buy everything, so unfortunately the hurdle is too high for me with so little success.

avandorp, 2024-03-04:
> @morremeyer Unfortunately, I don't have any hardware available. I would have to buy everything, so unfortunately the hurdle is too high for me with so little success.

You can long click until 2A is shown. If it says "read" it's the proprietary mode (just don't double click in 2A or 2B).

weetmuts, 2024-06-19:
Thanks for the telegram! Great work in writing the driver in c++, but nowadays the drivers should be written in the configuration file format. Which is just a text file, ie no need to re-compile wmbusmeters. I have added your driver here:

https://github.com/wmbusmeters/wmbusmeters/blob/master/drivers/src/istaheat.xmq

I changed the names of the fields. total is the latest total reading. target is commonly used to mean the previous billing period, typically a month, and last_year is what you expect. Selecting field names are not that easy, there are too many options to choose from. :-) 

Please test!

weetmuts, 2024-06-19:
@avandorp For the next driver you develop you can simply do this:

wmbusmeters --analyze=drivers/src/istaheat.xmq 5344742669315033A9048C2070900F002C25961200009D6949E80EB1E2707A96003007102F2F_0C0500000000426C00004C050000000082016CFE298C0105000000008C0115000000002F2F2F2F2F2F2F2F2F2F2F

I.e. just edit the xmq file and rerun the analyze command. :-)

avandorp, 2024-06-19:
Thanks. I wasn't sure whether the xmq files were ready for production yet.
I will test it, but probably not very soon. My installation is somewhat annoying to reach and update.

grinco, 2024-09-14:
Hey there. Does anyone know whether there is a proprietary protocol decoder for ista? I have confirmed with the vendor that my meters are using their proprietary mesh. Would be curious if anyone reverse engineered it...

avandorp, 2024-09-17:
> Hey there. Does anyone know whether there is a proprietary protocol decoder for ista?

I have not heard of such a project...

closed 2024-12-17 by github-actions[bot]