wmbusmeters/wmbusmeters #1058: Wired MBUS devices with a key setting of NOKEY lead to frame decoding issue

mrtnkhl, 2023-10-17:
### Type of request

Bug

### OS version

Home Assistant Core 11

### wmbusmeters version

ha-addon-edge containter

### Your message goes here

When setting the meter attribute "key" to "NOKEY" the code tries to use "NOKEY" as a key on frames received via wired M-BUS with the following config and fails to decode frames:

`MAIN=/dev/ttyUSB2:mbus:2400`

`driver=sensostar:MAIN:mbus`

weetmuts, 2023-10-18:
Can you provide a debug log?

mrtnkhl, 2023-11-09:
> Can you provide a debug log?

We are also seeing the M-BUS to drop and become unavailable over time. A restart of the add-on fixes that. 
@weetmuts could you provide details on how to access the full logs via the Home Assistant frontend or terminal since what is displayed in the add on log view in the UI is limited to only recent messages unfortunately?

closed 2024-02-09 by github-actions[bot]