wmbusmeters/wmbusmeters #1064: Sensostar E wired M-Bus meter reading gets stuck in loop over "no 0x68 byte found"

mrtnkhl, 2023-10-21:
### Type of request

Bug

### HAOS version

Version 9

### wmbusmeters version

edge

### Your message goes here

After several hours operating with no issues and reading correct data from the meter, the Home Assistant Edge Plugin displays the following in the log. Only a restart of the plugin solves the issue and the meter is read correctly for a couple more hours until the issue reappears:

(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(meter) waermemengenmesser 24051579 did not send a response!
(meter) polling waermemengenmesser 24051579 (secondary) with req ud2 bus MAIN
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) no 0x68 byte found, clearing

mrtnkhl, 2023-10-21:
Home Assistant 2023.10.3
Supervisor 2023.10.0
Operating System 11.0
Frontend 20231005.0 - latest

mrtnkhl, 2023-10-21:
Add On version [edge] 1.14.0-31-4

weetmuts, 2024-01-24:
Yes, It probably gets out of sync with the incoming slow serial data. The test is too narrow, it expects the first byte
to be 0x68, if there is some garbage (ie end of the previous telegram) first then it will clear the buffer.

I am working on a fix. Can you run with --debug and capture the serial data when this happens?

closed 2024-04-25 by github-actions[bot]