wmbusmeters/wmbusmeters #1066: multical602 w. upgraded printboard

drspliph, 2023-10-22:
Explain the problem..... 

https://wmbusmeters.org/analyze/31442d2c143713651c0c8d20fca87394257f1Bc90937Bcc93fe24aec374B7463df84ef7311684a829244805f73fBaB198162:kamheat:A307A3226B904BF7AEA55669B436890F 

And from wmbusmeters.log
 (meter) MyHeatingMeter: for me? 65133714 in 65133714
[2023-10-22_16:17:08] (meter) MyHeatingMeter: meter detection did not match the selected driver kamheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 65133714 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Heat volume at inlet meter (0x0c) ver: 0x1c
[2023-10-22_16:17:08] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2023-10-22_16:17:08] (meter) to add support for this unknown mfct,media,version combination
(meter) MyHeatingMeter: yes for me
(meter) MyHeatingMeter(1) kamheat  handling telegram from 65133714
(meter) MyHeatingMeter 65133714 "31442D2C143713651C0C8D2034487994253B924C298C8570F29A65F29FFD370920A36F36E290AD4697C90C906E904816691E"
(wmbus) parseDLL @0 50
(telegram) DLL L=31 C=44 (from meter SND_NR) M=2c2d (KAM) A=65133714 VER=1c TYPE=0c (Heat volume at inlet meter) (driver unknown!) DEV=cul RSSI=-41
(wmbus) parseELL @10 40
(ELL) decrypting "3B924C298C8570F29A65F29FFD370920A36F36E290AD4697C90C906E904816691E"
(ELL) IV 2D2C143713651C0C2048799425000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "0ADEE7E8E35389C01F3205BCC3A7355F"
(ELL) block 1 block_size 16 offset 16
(ELL) decrypted "76409AD17ACF2638EF03FBA36B2526F6"
(ELL) block 2 block_size 1 offset 32
(ELL) decrypted "8F"
(ELL) decrypted "0ADEE7E8E35389C01F3205BCC3A7355F76409AD17ACF2638EF03FBA36B2526F68F"
[2023-10-22_16:17:08] (wmbus) WARNING! decrypted payload crc failed check, did you use the correct decryption key? 0ade payload crc (calculated fba0) Permanently ignoring telegrams from id: 65133714 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Heat volume at inlet meter (0x0c) ver: 0x1c
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=34 SN=48799425 (AES_CTR session=8 time=5851028) CRC=0ade
[2023-10-22_16:17:08] (meter) newly created meter (MyHeatingMeter 65133714 kamheat) did not handle telegram!

closed 2023-10-24 by drspliph

nagydavid, 2024-05-22:
hey @drspliph  , could you solve the issue?, I am facing the same on the ha addon

drspliph, 2024-05-25:
It's a bit fuzzy now - and I'm not one for notetaking.

But I believe I added a line to driver_kamheat.cc with the version number given by a telegram.log I gathered for the meter.

The working theory being that the new board would likely behave as all the others.

'It worked for me - but the raspberry is too flaky so I railoraded a colleague to participate in an embedded adventure and we now have a fairly solid C3 with a c1101 solution.