wmbusmeters/wmbusmeters #1067: Hydrus Flow value seems wrong

pschwager, 2023-10-24:
### Type of request

Existing device improvement

### Meter description

Hi,

I find sometimes "extremely" high values in the flow field of my hydrus device logs of wmbusmeters. Most of the time it contains values from 0 till 0.5 m³/h, which sounds reasonable, but from time to time it jumps up to values at the 2000 m³/h. I provided here one example:

https://wmbusmeters.org/analyze/6344A5112018646470078C00D7900F002C256AB59B00F0F13032019092DE7A6A004007102F2F0C13896729004C1323462400CC101300000000CC201323462400426CDF2C0B3B0200F002FD742F0D025AC100C4016D3B17FE29CC01132841290001FD089F:hydrus

Could it be that the flow is sometimes getting negative and thus a overflow is happening or something like that?

### Logged telegrams from meter in decrypted format

```shell
6344A5112018646470078C00D7900F002C256AB59B00F0F13032019092DE7A6A004007102F2F0C13896729004C1323462400CC101300000000CC201323462400426CDF2C0B3B0200F002FD742F0D025AC100C4016D3B17FE29CC01132841290001FD089F
```

Stefan8615, 2023-11-12:
Have exactly the same issue here…

weetmuts, 2023-11-12:
Yeah, its negative values. Normally you have to say AutoSigned in the driver when values can be negative for a field. This is necessary when the values are sent using binary, ie there is no flag indicating if the binary should be interpreted as signed or unsigned. However for bcd values it IS visible since a negative bcd value according the (w)mbus standard has the highest nybble set to F. So I made both Auto and AutoSigned detect such
bcd values and properly negate the value.

wmbusmeters now also detects bcd values with all F which means that there is no value.
So either it is now not printed in the json or printed as null in the fields.

Now your meter reports "flow_m3h":-0.002
:-)

Push fixed! Please test!

weetmuts, 2023-11-12:
By the way! There is a value in the telegram which is not mapped correctly.
 In your telegram above:
"total_m3":296.789
then there is an unused value at the end:
294.128

Is this the yesterdays value?

pschwager, 2023-11-16:
I just updated and will wait for the next negative value.

> In your telegram above: "total_m3":296.789 then there is an unused value at the end: 294.128
That's the value of the start of month / end of last month, so in this example 01.11.2023 00:00

weetmuts, 2023-11-16:
Thanks! I rename woot_m3/at_datetime to target_m3 and target_datetime.  This is a naming convention that is used for several other meters for the best read-out value when doing drive by reading of meters for regular billing.

pschwager, 2023-11-17:
Just checked about the negative values: They are parsed correct, now! Thanks for fixing it

weetmuts, 2024-05-06:
https://github.com/wmbusmeters/wmbusmeters/issues/1238