wmbusmeters/wmbusmeters #1068: kaloBLUE water meter with radio plug-in module

fwmarcel, 2023-10-25:
### Type of request

New device support

### Meter description

Hello all,
thanks a lot for your efforts! I have started using the wmbusmeters on my Home Assistant instance to get more data in my home. 
We have both heat cost allocators and water meters in the house.
However, these do not seem to be supported. Can you guys help with this?

Thanks

meter detection could not find driver for id: 01823153 mfct: (LSE) Unknown (0xb265) type: Water meter (0x07) ver: 0x51

### Logged telegrams from meter in decrypted format

```shell
I am currently still collecting telegrams. How can I tell if they are encrypted?
```

weetmuts, 2023-10-31:
If you run just `wmbusmeters --logtelegrams auto:t1`
you will get lines like:

```
Received telegram from: 72727272
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x70
                driver: hydrus
telegram=|............................|
``` 

where you see encrypted. If it does not say encrypted, then it is probably not encrypted. :-)

fwmarcel, 2023-12-31:
I had checked the messages are unfortunately encrypted

closed 2024-04-01 by github-actions[bot]