wmbusmeters/wmbusmeters #1074: curl to REST api

Flearz, 2023-10-31:
### Type of request

Question

### OS version

Raspberry PI OS

### wmbusmeters version

latest

### Your message goes here

Hello,
I am using a Texas Instruments board to simulate a meter.
My goal is to post the json data to a flask server.

QUESTION 1:
I am struggling to format the shell command in order to send this : 
[image]
This is the what i tried:
[image]

Please help me get the right curl command...

QUESTION 2 
This is what I transmit:
```
/* Check if a specific manufacturer id should be used */
#ifndef MANUFACTURER_ID
  /*! Default manufacturer ID (STACKFORCE) MSB */
  #define MANUFACTURER_ID_H      0xCEU
  /*! Default manufacturer ID (STACKFORCE) LSB */
  #define MANUFACTURER_ID_L      0x86U
  #warning MANUFACTURER_ID is not defined. Default manufacturer ID 0xCE86 (STF) is used
#else
  /* Set the ID of the manufacturer */
  #define MANUFACTURER_ID_H      (uint8_t)((MANUFACTURER_ID & 0xFF00U)>>8)
  #define MANUFACTURER_ID_L      (uint8_t)(MANUFACTURER_ID & 0x00FFU)
#endif /* MANUFACTURER_ID */

/*! Max length of the payload:
    MAX_PAYLOAD_LEN = 255-9(Dll-Header->L field not counted)-1(CI-Field) */
#define MAX_PAYLOAD_LEN (uint8_t)245U

/*==============================================================================
                            VARIABLES
==============================================================================*/

/*! Example meter address (own address) */
s_wmbus_addr_t gs_addr = {{MANUFACTURER_ID_H, MANUFACTURER_ID_L},
                    {0x80,0x00,0x00,0x01},    /* ident number            */
                    0x23,                     /* version                 */
                    WMBUS_DEV_TYPE_OTHER};    /* type, here water        */

/*! Example collector address */
s_wmbus_addr_t gs_collector = {{MANUFACTURER_ID_H, MANUFACTURER_ID_L},
                    {0x80,0x00,0x00,0x02},    /* ident number            */
                     0x23,                    /* version                 */
                    WMBUS_DEV_TYPE_OTHER};    /* type, here other        */

/*! Example meter key. */
ENCRYPTION_KEY_t gt_key[] = {0x00,0x11,0x22,0x33,0x44,0x55,0x66,0x77,
                    0x88,0x99,0xAA,0xBB,0xCC,0xDD,0xEE,0xFF};

/*! Local storage to store telegram data in. */
uint8_t gac_telegramStorage[MAX_PAYLOAD_LEN];

/*! Init start strucutre for meter device */
s_apl_mtrInitAttr_t gs_meterInitAttr =
{
  {
  /* Frequency offset for the carrier. */
  CUSTOMER_FREQ_OFFSET,
  /* Device address */
  &gs_addr,
  /* Telegram storage */
  gac_telegramStorage,
  /* Size of the telegram storage */
  MAX_PAYLOAD_LEN
  },
  /* Encryption key */
  gt_key,
  /* Collector address. */
  &gs_collector,
  /* Set the device to connected. Only if the device is a meter device.
     Otherwise this field is ignored. */
  true,
  /* Periodical interval for sending data. Only if the device is a meter
     device. [s] */
  8U
};

/*! HJ user data.  */
uint8_t gpc_userData[]={ 0xA1,  /* Record 1: DIB: Data=8 digit BCD, Function=Instantaneous*/
                         0xA2, 
                         0xA3,  
                         0xA4,  
                         0xA5,  
                         0xA6,  
                         0xA7, 
                         0xA8,  
                         0xA9, 
                         0xB1, 
                         0xB2}; 
```
And this is what i get when I log with debug:
[image]

I would like to have the datas of the first picture + my payload formatted as json and transmit it using curl.

What should I do ?

weetmuts, 2023-10-31:
I think this might give you the answer.  https://reqbin.com/req/c-dwjszac0/curl-post-json-example

What do you want to encode in the difvif data? Ie, from your byte A1 and onwards?

Flearz, 2023-11-01:
Hello,
Thanks for you answer.
The problem is that the curl is working if I just use it alone:
[image]

but if I try to do it with the "--shell" option (without the meter data for now):
[image]

Once this will for I would like to transmit this (to begin with)
[image]

I guess I would have to replace my dummy data with '$METER_JSON'... but I can't get this working.

Once these problems are fixed, I would ideally, end up trasmitting something like this: 
{"media":"other","meter":"auto","name":"TI_MCU","id":"80000001","timestamp":"2023-10-31T16:15:54Z","device":"amb8465[00000000]","rssi_dbm":-113, **"payload":"A1A2A3A4A5A6A7A8A9B1B2"**}

THanks in advance

closed 2024-02-01 by github-actions[bot]