wmbusmeters/wmbusmeters #1076: 2 Devices Configuration Home Assistant

alexmaria8, 2023-11-01:
### Type of request

Question

### HAOS version

2023.9.2

### wmbusmeters version

1.14.0

### Your message goes here

Hello!
Running the wmbusmeters addon on home assistant already with 1 device.
I got another im871a-USB device because i want to run 2 of them at the same time. The first one is working fine,
but I fail at the confifuration respectively it won't save 2 device after a restart of the wmbusmeter-addon neither a system-restart of Home Assistant.

What i tried:
**1) At the wmbusmeters HA addon configuration i clicked on 'Add a new parameter' named it: device = /dev/ttyUSB3:im871a:s1
The first one i got in use is already named: device = /dev/ttyUSB2:im871a:c1:t1
After i clicked on 'Save Changes' the addon restarts but than theres only 1 device (the newly added) and the other one is gone.**

**2) I edited the wmbusmeter.conf file from this:**
_device=/dev/ttyUSB2:im871a:c1,t1
donotprobe=/dev/ttyAMA0
format=json
logfile=/dev/stdout
loglevel=normal
logtelegrams=false
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"_
**to this:**
_device=/dev/ttyUSB2:im871a:c1,t1
device=/dev/ttyUSB3:im871a:s1
donotprobe=/dev/ttyAMA0
format=json
logfile=/dev/stdout
loglevel=normal
logtelegrams=false
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"_
**saved the file (i closed the file and reopnened it to check) but after a full restart of the system it looked again like before (only 1 device):**
_device=/dev/ttyUSB2:im871a:c1,t1
donotprobe=/dev/ttyAMA0
format=json
logfile=/dev/stdout
loglevel=normal
logtelegrams=false
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"_

Can you plz help me?

alexmaria8, 2023-11-02:
> Try to use the identifiers of your im871a-USB's to separate them
> 
> * the ID is the last 8 digits of the serial number printed on the stick
> 
> device=/dev/ttyUSB0:im871a[00102759]:t1
> 
> wmbusmeters log shows you the IDs:
> 
> Started config im871a[00102759] on /dev/ttyUSB0 listening on t1

Hey lindsand!
Thank you for your reply! Sadly it gives me the same results as before, so only 1 device stays the other one not.
Here are some screenshots (First variant I tried to change it in the wmbusmeter configuration UI and 2nd try to change the wmbusmeter.conf file).

Variant 1 (1st pic is initial situation, 2nd what i added and 3rd the result after saving and restarting the addon):
[image]
[image]
[image]
[image]

Variant 2 (1st pic is initial situation, 2nd what i added and saved and 3rd the result after a restart of the system):
[image]
[image]
[image]

weetmuts, 2023-11-02:
Perhaps HA cannot handle multiple identical parameters? @BIBOLV

lindsand, 2023-11-02:
Put them on the same line separated with ;

>>wmbusmeters configuration
>>Content of this section will be used as wmbusmeters.conf. If two values for configuration parameter is needed, like:

>>donotprobe: /dev/ttyAMA0
>>donotprobe: /dev/ttyUSB0
>>**In web configuration interface it should be provided in one line** using ; delimiter between values, like: >>donotprobe=/dev/ttyAMA0;/dev/ttyUSB0

I made a mistake in the configuration some days ago and set up a logfile that didn't exist preventing wmbusmeters from starting. I tried to correct it by editing wmbusmeters.conf but it didn't help, the same wrong value appeared in wmbusmeters.conf after each restart. So I came to the conclusion that the UI-content is stored somewhere else and wmbusmeters.conf is overwritten with UI-data after each restart. So I had to uninstall wmbusmeters totally and start from scratch.

BIBOLV, 2023-11-02:
Yes, two parameters can be provided in one line using ; as delimiter.
Yes, manually edited config file will be reseted on each restart. Config is stored in HA internal db and written to config on each restart.
@lindsand - you could have reseted config without reinstall trough addon configuration tab.

lindsand, 2023-11-02:
>>you could have reseted config without reinstall trough addon configuration tab

No, wmbusmeter kept on crashing before addon configuration tab became available. So I was stuck.

alexmaria8, 2023-11-02:
@lindsand @BIBOLV @weetmuts

Thank u all for ur help! **Using the delimiter  ;  was the key.** Now i can go on.

Just for people who encounter the same issue (in the future) i post the screenshot with the working HA addon config below. (I checked the wmbusmeters.conf later too and it keeps the second device. Getting it from the UI configuration as u already mentioned)
THX again!

[image]
[image]

closed 2023-11-02 by alexmaria8

BIBOLV, 2023-11-02:
@lindsand - you are talking about config tab at sidebar or one within HA addon related configuration alongside documentation tab?

lindsand, 2023-11-03:
Config tab at sidebar, I made a mistake in the logfile name:

[image]

End result was that wmbusmeters crashed at startup (because it couldn't write to the log) and again after each restart, watchdog was on. So it was looping. And editing wmbusmeters.conf didn't change anything, old values popped back.

I noticed this possibility only afterwards:
[image]

I wanted to log some meter readings and after some trials I found out it can be done by adding a line in  wmbusmeters configuration:

meterfiles=/config/wmbusmeters/logs/meter_readings