wmbusmeters/wmbusmeters #1078: Got unexpected response from the API: Service not enabled

sonnstein, 2023-11-04:
### Type of request

Question

### HAOS version

11.1

### wmbusmeters version

1.14.0

### Your message goes here

My MQTT connection does not work.
In the logfile of the Wmbusmeters there is a error message about the MQTT Connection
```
[12:55:03] INFO: Adding meter-0001 ...
[12:55:03] INFO: Generating MQTT configuration ... 
[12:55:04] INFO: Broker 192.168.2.1 will be used.
[12:55:05] ERROR: Got unexpected response from the API: Service not enabled
[12:55:05] INFO: MQTT Discovery cleanup...
[12:55:06] INFO: Starting web configuration service.
[12:55:06] INFO: Running wmbusmeters ...
```

In the logfile of the MQTT broker, I can see a successfull connection,that is closed two seconds later.

Can anybody give me a hint, what can cause the problem?

BIBOLV, 2023-11-04:
Did you configure some meter and observed in logs that message is being sent to MQTT but is not received? 
I have noticed similar False Positive error about MQTT in logs previously.
Adding to plans to check what actually causes this error and if we can get rid of it.

BIBOLV, 2024-12-25:
Updated to remove error message - checks are done to verify MQTT availability or configuration and properly logged.

closed 2025-03-26 by github-actions[bot]