wmbusmeters/wmbusmeters #1079: MQTT entity name starts with the device name in your config

karmat75, 2023-11-04:
### Type of request

Bug

### HAOS version

11.1

### wmbusmeters version

1.14.0

### Your message goes here

the following messages appear in the last times in the core logs:

```
Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:1314
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 2. November 2023 um 22:00:41 (4 occurrences)
Last logged: 2. November 2023 um 22:00:41

MQTT entity name starts with the device name in your config {'device': {'identifiers': ['wmbusmeters_13188703'], 'manufacturer': 'Qundis', 'model': 'lse_07_17', 'name': 'BZ_WaterMeter_Cold', 'sw_version': '13188703', 'connections': []}, 'enabled_by_default': False, 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'device_class': <SensorDeviceClass.SIGNAL_STRENGTH: 'signal_strength'>, 'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>, 'name': 'BZ_WaterMeter_Cold rssi', 'state_topic': 'wmbusmeters/BZ_WaterMeter_Cold', 'unique_id': 'wmbusmeters_13188703_rssi_dbm', 'unit_of_measurement': 'dbm', 'value_template': Template<template=({{ value_json.rssi_dbm }}) renders=0>, 'icon': 'mdi:signal', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'force_update': False, 'availability_mode': 'latest', 'payload_not_available': 'offline'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Rssi'
MQTT entity name starts with the device name in your config {'device': {'identifiers': ['wmbusmeters_13188703'], 'manufacturer': 'Qundis', 'model': 'lse_07_17', 'name': 'BZ_WaterMeter_Cold', 'sw_version': '13188703', 'connections': []}, 'enabled_by_default': True, 'json_attributes_topic': 'wmbusmeters/BZ_WaterMeter_Cold', 'device_class': <SensorDeviceClass.WATER: 'water'>, 'state_class': <SensorStateClass.TOTAL: 'total'>, 'name': 'BZ_WaterMeter_Cold total', 'state_topic': 'wmbusmeters/BZ_WaterMeter_Cold', 'unique_id': 'wmbusmeters_13188703_total_m3', 'unit_of_measurement': 'm³', 'value_template': Template<template=({{ value_json.total_m3 }}) renders=0>, 'icon': 'mdi:gauge', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'force_update': False, 'availability_mode': 'latest', 'payload_not_available': 'offline'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Total'
MQTT entity name starts with the device name in your config {'device': {'identifiers': ['wmbusmeters_13186226'], 'manufacturer': 'Qundis', 'model': 'lse_07_17', 'name': 'BZ_WaterMeter_Hot', 'sw_version': '13186226', 'connections': []}, 'enabled_by_default': False, 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'device_class': <SensorDeviceClass.SIGNAL_STRENGTH: 'signal_strength'>, 'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>, 'name': 'BZ_WaterMeter_Hot rssi', 'state_topic': 'wmbusmeters/BZ_WaterMeter_Hot', 'unique_id': 'wmbusmeters_13186226_rssi_dbm', 'unit_of_measurement': 'dbm', 'value_template': Template<template=({{ value_json.rssi_dbm }}) renders=0>, 'icon': 'mdi:signal', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'force_update': False, 'availability_mode': 'latest', 'payload_not_available': 'offline'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Rssi'
MQTT entity name starts with the device name in your config {'device': {'identifiers': ['wmbusmeters_13186226'], 'manufacturer': 'Qundis', 'model': 'lse_07_17', 'name': 'BZ_WaterMeter_Hot', 'sw_version': '13186226', 'connections': []}, 'enabled_by_default': True, 'json_attributes_topic': 'wmbusmeters/BZ_WaterMeter_Hot', 'device_class': <SensorDeviceClass.WATER: 'water'>, 'state_class': <SensorStateClass.TOTAL: 'total'>, 'name': 'BZ_WaterMeter_Hot total', 'state_topic': 'wmbusmeters/BZ_WaterMeter_Hot', 'unique_id': 'wmbusmeters_13186226_total_m3', 'unit_of_measurement': 'm³', 'value_template': Template<template=({{ value_json.total_m3 }}) renders=0>, 'icon': 'mdi:gauge', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'force_update': False, 'availability_mode': 'latest', 'payload_not_available': 'offline'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Total'
```
Maybe the following links will help:
- https://www.home-assistant.io/integrations/mqtt#naming-of-mqtt-entities
- https://developers.home-assistant.io/docs/core/entity/#entity-naming

lindsand, 2023-11-05:
Looks like you are using the stable version and driver 'lse_07_17' with MQ-discovery so this file:

https://github.com/wmbusmeters/wmbusmeters-ha-addon/blob/main/wmbusmeters-ha-addon/mqtt_discovery/lse_07_17.json

with these lines for sensor name:
            "name": "**{name}** total",
            "name": "**{name}** rssi",

End result is that the device name is repeated in the sensor name:

{'device': {'identifiers': ['wmbusmeters_13188703'],
 'manufacturer': 'Qundis',
 'model': 'lse_07_17',
 **'name': 'BZ_WaterMeter_Cold',**
 'sw_version': '13188703',
 'connections': []},
 'enabled_by_default': False,
 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>,
 'device_class': <SensorDeviceClass.SIGNAL_STRENGTH: 'signal_strength'>,
 'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>,
 'name': '**BZ_WaterMeter_Cold rssi'**,

and

 'name': '**BZ_WaterMeter Cold** total',

It should be:
**"name": "Total",** 
**"name": "rssi",**
for the sensor names
            
It's already corrected here for the edge-version:

[wmbusmeters-ha-addon-edge/mqtt_discovery/lse_07_17.json](https://github.com/wmbusmeters/wmbusmeters-ha-addon/blob/main/wmbusmeters-ha-addon-edge/mqtt_discovery/lse_07_17.json)
(Device name prefix from entity name removed)

Either switch to the edge-version or update your mqtt_discovery file  lse_07_17.json
(or copy the edge version to replace your /config/wmbusmeters/etc/mqtt_discovery/lse_07_17.json)

This works for me without warnings:
[image]

karmat75, 2023-11-06:
Hello Anders,

when it is stil fixed in the edge ring, i will wait for the deployment in the stable ring.

Thanks for your time!

BR

closed 2023-11-06 by karmat75