wmbusmeters/wmbusmeters #1080: Build on MACOSX not running

Anas92230, 2023-11-04:
Hello,

I try to install it on MACOSX but I'm not able to do it.
I have this error after launch ./configure :
checking build system type... aarch64-apple-darwin23.0.0
checking host system type... aarch64-apple-darwin23.0.0
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for rtlsdr_get_device_count in -lrtlsdr... no
configure: error: Could  not find rtlsdr library. Try: sudo apt install librtlsdr-dev

Any suggestion?

Thanks

Anas92230, 2023-11-04:
I see an another issue about that
I add line on Makefile but we have to launch configure before. I have the same error.

Anas92230, 2023-11-10:
no more chance :-(

jetmcquack, 2024-05-27:
same problem here the issue is still OPEN

jetmcquack, 2024-05-27:
checking build system type... aarch64-apple-darwin23.5.0
checking host system type... aarch64-apple-darwin23.5.0
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for rtlsdr_get_device_count in -lrtlsdr... no
configure: error: Could  not find rtlsdr library. Try: sudo apt install librtlsdr-dev

weetmuts, 2024-05-27:
brew install librtlsdr libusb

jetmcquack, 2024-05-27:
done, unfortunately I get this:
`jetmcquack@jetminim2-eth wmbusmeters % brew install librtlsdr libusb
==> Downloading https://formulae.brew.sh/api/formula.jws.json
############################################################################################################################################################# 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
############################################################################################################################################################# 100.0%
Warning: librtlsdr 2.0.2 is already installed and up-to-date.
To reinstall 2.0.2, run:
  brew reinstall librtlsdr
Warning: libusb 1.0.27 is already installed and up-to-date.
To reinstall 1.0.27, run:
  brew reinstall libusb`

jetmcquack, 2024-05-27:
I have also tried on another Sonoma macOS machine running 14.5 and I get the same exact error:

> jetmcquack@jetminim1-eth wmbusmeters % ./configure                  
checking build system type... aarch64-apple-darwin23.5.0
checking host system type... aarch64-apple-darwin23.5.0
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for rtlsdr_get_device_count in -lrtlsdr... no
configure: error: Could  not find rtlsdr library. Try: sudo apt install librtlsdr-dev

weetmuts, 2024-05-27:
Yeah, homebrew stuffs the libraries in a different directory when building on arm macs compared to intel macs.
The github build started to fail, so I had to debug this. 

Fix pushed! Please test!

Anas92230, 2024-05-28:
@weetmuts great news 😎
I'll test it 👍🏽

Anas92230, 2024-05-28:
hi,
./ configure is ok
when I launch make I have this errors:
New version number generates new build/version.h
Building 1.16.1-33-g8748a2d
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/address.cc -c -E > build/address.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/address.cc -MMD -c -o build/address.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/aes.cc -c -E > build/aes.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/aes.cc -MMD -c -o build/aes.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/aescmac.cc -c -E > build/aescmac.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/aescmac.cc -MMD -c -o build/aescmac.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/bus.cc -c -E > build/bus.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/bus.cc -MMD -c -o build/bus.o
In file included from src/bus.cc:18:
src/bus.h:78:10: warning: private field 'rtl433_found_' is not used [-Wunused-private-field]
    bool rtl433_found_ = false;
         ^
1 warning generated.
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/cmdline.cc -c -E > build/cmdline.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/cmdline.cc -MMD -c -o build/cmdline.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/config.cc -c -E > build/config.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/config.cc -MMD -c -o build/config.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/drivers.cc -c -E > build/drivers.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/drivers.cc -MMD -c -o build/drivers.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/dvparser.cc -c -E > build/dvparser.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/dvparser.cc -MMD -c -o build/dvparser.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/formula.cc -c -E > build/formula.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/formula.cc -MMD -c -o build/formula.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/mbus_rawtty.cc -c -E > build/mbus_rawtty.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/mbus_rawtty.cc -MMD -c -o build/mbus_rawtty.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/metermanager.cc -c -E > build/metermanager.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/metermanager.cc -MMD -c -o build/metermanager.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/meters.cc -c -E > build/meters.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/meters.cc -MMD -c -o build/meters.o
src/meters.cc:2152:27: warning: adding 'int' to a string does not append to the string [-Wstring-plus-int]
    if (bps != 0) r += bps+":";
                       ~~~^~~~
src/meters.cc:2152:27: note: use array indexing to silence this warning
1 warning generated.
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/manufacturer_specificities.cc -c -E > build/manufacturer_specificities.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/manufacturer_specificities.cc -MMD -c -o build/manufacturer_specificities.o
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/printer.cc -c -E > build/printer.o.src
c++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild -I/opt/homebrew/include src/printer.cc -MMD -c -o build/printer.o
[... cut ...]
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/ios:1004:1: note: candidate found by name lookup is 'std::hex'
hex(ios_base& __str)
^
src/util.cc:322:20: warning: passing an object of reference type to 'va_start' has undefined behavior [-Wvarargs]
    va_start(args, fmt);
                   ^
src/util.cc:317:34: note: parameter of type 'const string &' (aka 'const basic_string<char> &') is declared here
string tostrprintf(const string& fmt, ...)
                                 ^
1 warning and 8 errors generated.
make: *** [build/util.o] Error 1

jetmcquack, 2024-05-28:
Hello i've checked and same problem here, but there's an easy fix. in util.cc you have to change look for the const hex and rename it to something like hexChar, there are occurences at lines 
252
259
269
270
282
283
296
297

i don't know how to make edit here on github but fixing these lines will fix the ambiguous error.

jetmcquack, 2024-05-28:
However after this edit it will not compile, as there's an error in sudo make install. 
In the script install.sh it's called another one (install_binaries.sh) which uses a parameter not found in macOS.
at line 14 the code is

> install -D -m 755 "$SRC" "$wmbusmeters_path" || exit $?

parameter -D (which creates directories if missing) is not present on macos. 

i don't know ho to fix this

Anas92230, 2024-05-28:
> Hello i've checked and same problem here, but there's an easy fix. in util.cc you have to look for and change the const hex -> rename it to something like hexChar; there are occurences at lines 252 259 269 270 282 283 296 297
> 
> i don't know how to make edit here on github but fixing these lines will fix the ambiguous error.

Thanks I have to add line 258 too!

Anas92230, 2024-05-28:
> However after this edit it will not compile, as there's an error in `sudo make install`. In the script install.sh it's called another one (install_binaries.sh) which uses a parameter not found in macOS. at line 14 the code is
> 
> > install -D -m 755 "$SRC" "$wmbusmeters_path" || exit $?
> 
> parameter -D (which creates directories if missing) is not present on macos.
> 
> i don't know how to fix this

on my side make finished wit success.
but when I try sudo make install I have this error (-D parameter :-():
 sudo make install
su: illegal option -- c
usage: su [-] [-flm] [login [args]]
su: illegal option -- c
usage: su [-] [-flm] [login [args]]
su: illegal option -- c
usage: su [-] [-flm] [login [args]]
su: illegal option -- c
usage: su [-] [-flm] [login [args]]
su: illegal option -- c
usage: su [-] [-flm] [login [args]]
su: illegal option -- c
usage: su [-] [-flm] [login [args]]
New version number generates new build/version.h
Building _
Installing build/wmbusmeters
install: illegal option -- D
usage: install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
               [-o owner] file1 file2
       install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
               [-o owner] file1 ... fileN directory
       install -d [-v] [-g group] [-m mode] [-o owner] directory ...
make: *** [install] Error 64

jetmcquack, 2024-05-28:
yes, that's what i meant to say, it gives you error 64 but if you read a few lines above it says `illegal option -- D`

jetmcquack, 2024-05-28:
@weetmuts i have created a pull request for the "hex" ambiguity  fix 
#1292

jetmcquack, 2024-05-28:
great glad to help!! unfortunately I don't know how to fix the problem with invalid parameter --D

closed 2024-08-27 by github-actions[bot]

BrianDK17, 2024-09-27:
Hey everybody,

Anas92230, 2024-09-28:
> Hey everybody, 
> 
> I got the same error when trying to run the install script. 
> 
> `install: illegal option -- D`
> 
> Does anybody know how to fix it? 

No news on my side 😔.

nieeck, 2024-12-28:
Am also having the same issue on OSX: "install: illegal option -- D" (on top of another error)

```
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Building _
Installing build/wmbusmeters
install: illegal option -- D
usage: install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
               [-o owner] file1 file2
       install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
               [-o owner] file1 ... fileN directory
       install -d [-v] [-g group] [-m mode] [-o owner] directory ...
make: *** [install] Error 64

```