wmbusmeters/wmbusmeters #1082: Not receiving Diehl/Izar water meter data

MarkusMit, 2023-11-05:
### Type of request

Bug

### HAOS version

11.1

### wmbusmeters version

1.14.0/1.14.0-40-2

### Your message goes here

Neither of the two Home Assistant (HA) wmbusmeters plugin variants (normal, edge) is receiving data from my Diehl water meter.
Running HA on Raspberry Pi 4 B, 8G (Home Assistant image).

I also installed rtl-sdr, rtl_wmbus and wmbusmeters in WSL (Ubuntu) on Win10, there it is working like a charm. Data is received every few seconds. (using `wmbusmeters --logtelegrams --format=json --debug rtlwmbus:868.95M:t1`)

I noticed, that the HA plugin is using rtl_sdr version 0.8, while my own setup is using v2.0.1 (Installed according to [this](https://osmocom.org/projects/cellular-infrastructure/wiki/Latest_Builds)). Could this be a possible reason?
(Versions https://github.com/steve-m/librtlsdr/tags)

Log-files:
[c8a990ad_wmbusmeters-ha-addon_2023-11-05T15-30-15.110Z.log](https://github.com/wmbusmeters/wmbusmeters/files/13260125/c8a990ad_wmbusmeters-ha-addon_2023-11-05T15-30-15.110Z.log)
[c8a990ad_wmbusmeters-ha-addon-edge_2023-11-05T15-26-54.989Z.log](https://github.com/wmbusmeters/wmbusmeters/files/13260126/c8a990ad_wmbusmeters-ha-addon-edge_2023-11-05T15-26-54.989Z.log)
[wsl_wmbusmeters.log.txt](https://github.com/wmbusmeters/wmbusmeters/files/13260127/wsl_wmbusmeters.log.txt)

Skotniki17, 2023-11-24:
Same issue , did you find solution

chpego, 2023-12-26:
https://github.com/wmbusmeters/wmbusmeters/issues/989

closed 2024-03-27 by github-actions[bot]