wmbusmeters/wmbusmeters #1083: Segmentation fault

rstephan, 2023-11-06:
### Type of request

Bug

### OS version

Ubuntu

### wmbusmeters version

master

### Your message goes here

A segmentation fault is happening with a typo in the driver name.

e.g.:
```
./wmbusmeters 434493157856341233038c2075900f002c25b30a000021924d4f2fb66e017a75002007109058475f4bc91df878b80a1b0f98b629024aac727942bfc549233c0140829b93 MyTapWater q40 12345678 NOKEY
```
**q400 -> q40**

It looks like a, not handled 0-ptr in `DriverDynamic::load()`.

weetmuts, 2023-11-06:
Yikes, thanks! Yes, the new code for loading drivers from text files was used for non-existant drivers. 
Test added that verified that the load is only attempted if the driver name ends with .xmq and the file actually exists. 

If you are curious this is what the future text base driver files will look like.
I.e. we will not need to recompile the whole source to add new meters.

https://libxmq.org/resources/examples/wmbusmeters_driver.html

weetmuts, 2023-11-06:
Fix pushed! Please test!

rstephan, 2023-11-07:
Works for me. Thanks!