wmbusmeters/wmbusmeters #1085: Add support for WURTH ELEKTRONIK 2607056281001

Anas92230, 2023-11-07:
### Type of request

New device support

### Meter description

Properties Test conditions Value Unit
Communication Protocol/
Standard RFprot proprietary (P2P, star, mesh)
Microcontroller μC MSP430
Radio Chipset RF-IC TI - CC1125
Memory Size (Flash) 65536 B
Memory Size (RAM) 6144 B
Radio channel min. 100
Radio channel max. 140
Interface 1 IO1 USB
Interface 1 Bitrate Range IO1 Bitrate latest FW revision 115.2 kbps
Number of I/O Ports IOPorts latest FW revision 

### Logged telegrams from meter in decrypted format

```shell
device not detected:
-wmbusmeters /dev/ttyUSB0:amb8465:c1,t1
Expected amb8465 on /dev/ttyUSB0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!

-wmbusmeters auto:t1
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!

-wmbusmeters --debug /dev/ttyUSB0:amb8465:c1,t1
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0:amb8465:t1,c1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" amb8465
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) before config: /dev/ttyUSB0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465/3665) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465/3665) reading response... 1
(serial) received binary ""
(amb8465/3665) reading response... 2
(serial) received binary ""
(amb8465/3665) reading response... 3
(serial) received binary ""
(amb8465/3665) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465/amb3665)
Expected amb8465 on /dev/ttyUSB0 but did not find it! Ignoring tty!
(verbose) ignoring device t1,c1
No wmbus device detected, waiting for a device to be plugged in.
(serial) EVENT thread interrupted
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(shell) return code 1
(shell) 17343 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(serial) EVENT thread interrupted
(shell) output: >>>17340
<<<
(shell) return code 0
(shell) 17344 exited
(serial) EVENT thread interrupted
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
```

weetmuts, 2023-11-07:
I think that the dongle/chip PROPRIETARY RADIO USB 868MHZ cannot listen to wmbus signals. It can only listen to radio traffic sent by itself, ie it is a proprietary radio protocol which, as far as I know is not used for wmbus compatible meters.

You might want to buy the Metis-II dongle instead which is already supported by wmbusmeters if you want to listen to wmbus-compatible meters.

Anas92230, 2023-11-07:
thank you for your quicly response :-)

Anas92230, 2023-11-07:
Best Regards

Anas92230, 2023-11-07:
this one it's ok : Metis-II WIRL-WMB8 USB Radio Module wMBus, Würth Elektronik?

weetmuts, 2023-11-07:
Yes  2607056283001 - Metis-II WIRL-WMB8 USB Radio Module wMBus

Anas92230, 2023-11-07:
i ordered this one :-)
thanks a lot

Anas92230, 2023-11-09:
Hello @weetmuts i trceived Metis-ii but no telegram received in compare with my rtl-sdr dongle:
command launched : wmbusmeters --logtelegrams --debug auto:t1
(wmbusmeters) version: _
(config) number of meters: 0
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465/3665) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465/3665) reading response... 1
(serial) received binary "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFF"
(amb8465) not enough data for decoding ConfigAMB8465!
(amb8465/3665) reading response... 2
(serial) received binary "FFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF69"
(amb8465) proprely decoded ConfigAMB8465 response. Content: uart_cmd_out_enable=00 id=00000000 media=00 version=00 c_field=44 auto_rssi=00 b1_add_disable=01
(amb8465) found response at offset 0
(serialtty) closed /dev/ttyUSB0 (detect amb8465/amb3665)
(amb8465) detect uart_cmd_out_enable=00 id=00000000 media=00 version=00 c_field=44 auto_rssi=00 b1_add_disable=01
(amb8465) are you there? yes 00000000
(main) opening t1
Started auto amb8465[00000000] on /dev/ttyUSB0 listening on t1
(amb8465) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (amb8465)
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "FF8A7A0078008071020000"
(amb8465) checkAMB8465Frame "FF8A7A0078008071020000"
(amb8465) not enough bytes yet, partial command response 11 126.
(serial) received binary "0000FFFFFA00FFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 26 126.
(serial) received binary "FFFFFFFFFFFFFF0C3200026400FFFFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFF"
(amb8465) not enough bytes yet, partial command response 42 126.
(serial) received binary "FFFF010004000000FFFFFF01440000"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000"
(amb8465) not enough bytes yet, partial command response 57 126.
(serial) received binary "000000000000FFFF0B060100FFFFFFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFF"
(amb8465) not enough bytes yet, partial command response 73 126.
(serial) received binary "FF00030000FFFFFFFFFFFFFF0000FF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FF"
(amb8465) not enough bytes yet, partial command response 88 126.
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 103 126.
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 119 126.
(serial) received binary "FFFFFFFFFFFF69"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF69"
[... 1 lines cut ...]
(amb8465) get config completed
(amb8465) get config response "7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[... cut ...]
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) EVENT thread interrupted
(serial) received binary "FF8B0453005AB0C9"
(amb8465) checkAMB8465Frame "FF8B0453005AB0C9"
(amb8465) received full command frame
(amb8465) get device id completed
(amb8465) get device id response "0453005AB0"
(amb8465) unique device id 53005ab0
(wmbus) no alarm (expected activity) for amb8465
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
Any idea?

Anas92230, 2023-11-10:
I open a new one issue with metis-ii

closed 2023-11-10 by Anas92230