wmbusmeters/wmbusmeters #1086: Qundis Q heat 5.5 R mismatch of value and unit of measurement in Home Assistant

hubikj, 2023-11-07:
When I analyze the telegram (see link below), it shows the correct value in kWh (2651.5), which corresponds to ~9.5 GJ, which is what the display of the meter shows. In Home Assistant, the value of total_energy_consumption shows up with unit of measurement kWh, but the value is ~9500 which would be correct if it was in MJ. Changing the unit of measurement in Home Assistant does not fix the problem, because the value is also converted to ~34000.

https://wmbusmeters.org/analyze/41449344533743674637725337436793444604620000200c0d545409004c0d36840300426cdf2ccc080d37330900c2086cff2a02fd170000326cffff046d130Be72B

hubikj, 2024-01-07:
This issue is still present. I imagine it should be easy to fix. Let me know if I can do something to help. Thanks.

weetmuts, 2024-05-11:
Strange, I am looking at the ./wmbusmeters-ha-addon/mqtt_discovery/qheat.json file.
I am not an expert, but it looks ok.

Do you have some extra setting in the wmbusmeters.conf  or HA-addon. Such a big problem should have
been seen by others.

hubikj, 2024-05-15:
I did not change anything from the defaults. This is how I have the meter configured:
[image]

hubikj, 2026-02-24:
> Fix merged and wmbusmeters.org update, please verify if it gives the right values now.

I think the online decoder was always giving the right values (as I wrote in the OP). I am using the Home Assistant addon which is what was giving me the problems and I had to [parse the values with units manually](https://github.com/wmbusmeters/wmbusmeters/issues/1086#issuecomment-2112823883). I do not see any update of the HA addon (yet?) to verify the fix.