wmbusmeters/wmbusmeters #1089: Metis-ii not working

Anas92230, 2023-11-10:
### Type of request

Bug

### OS version

Ubuntu 20.04.6 LTS

### wmbusmeters version

wmbusmeters --version give me wmbusmeters: _

### Your message goes here

Hello,

I recentrly get a Metis-ii usb dongle.
this usb dongle is on MSP430 with radio chipset CC1125.
wmbusmeters detecte it like amb8465 device:
`
wmbusmeters --logtelegrams  --verbose --format=json --t1 auto
`
``
(wmbusmeters) version: _
(config) number of meters: 0
(serial) tty /dev/ttyUSB0 can be accessed
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465/amb3665)
(serialtty) closed /dev/ttyUSB0 (detect amb8465/amb3665)
(amb8465) detect uart_cmd_out_enable=00 id=00000000 media=00 version=00 c_field=44 auto_rssi=00 b1_add_disable=01
(amb8465) are you there? yes 00000000
Started auto amb8465[00000000] on /dev/ttyUSB0 listening on t1
(amb8465) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0 fd 3 (amb8465)
(amb8465) get config
(amb8465) get config completed
(main) regular reset of amb8465  on /dev/ttyUSB0 will happen every 82800 seconds
(amb8465) set link mode 08
(amb8465) set link mode completed
(amb8465) get device unique id
(amb8465) get device id completed
(amb8465) unique device id 53005ab0
No meters configured. Printing id:s of all telegrams heard!
``
i receive nothing!
with RTL2832u with the same command i received messages.
Can you help me to understand where is the failure :-)

weetmuts, 2023-11-10:
Is your meter transmitting on t1,c1 or s1? Remember that rtlwmbus listens to all automatically
but the other dongles must be configured: try:

wmbusmeters --logtelegrams  --format=json amb8465:c1,t1

Anas92230, 2023-11-10:
hello
when i do that i'm not able to receive messages:
``
wmbusmeters --logtelegrams  --format=json amb8465:c1,t1
``
``
Started config amb8465[00000000] on /dev/ttyUSB0 listening on t1,c1
No meters configured. Printing id:s of all telegrams heard!
``
but when i plug my other dongle i receive message from other device(Interlogix-security and other tpms for example).

Anas92230, 2023-11-10:
O surprise i just receive message!!!
``
Received telegram from: 94653196
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: amb8465[00000000]
                  rssi: -53 dBm
                driver: qcaloric
telegram=|_314493449631659435087A2F0000200B6E0028004B6E172600426CDF2CCB086E002800C2086CFF2A326CFFFF046D2612EA2B|+280
``
Great!
now for my watermeter is on T1, when i search some information about my watermeter itron i found different frequencies on reception . How to modify frequencies on wmbusmeters?

Anas92230, 2023-11-11:
@weetmuts you think that i have to test S1 mode?
i'll wait 1 day capture before change anything.

Anas92230, 2023-11-11:
@weetmuts How it's possible to get the same telegram with:
-**my nooelec dongle**:
``
wmbusmeters --logtelegrams  --format=json auto:s1
``
``
Started auto rtlwmbus[77771111153705700] listening on s1
Received telegram from: 94653196
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: rtlwmbus[77771111153705700]
                  rssi: 15 dBm
                driver: qcaloric
telegram=|_49449344963165943508780DFF5F350082DE00000F0007B06EFFFF00280000DF2C17260000FF2A00280000450097005100200025001E000300000000000000000000002F046D1A10EB2B|+150
``
**and metis-ii dongle**:
``
wmbusmeters --logtelegrams --format=json  amb8465:c1,t1
``
``
Started config amb8465[00000000] on /dev/ttyUSB0 listening on t1,c1
Received telegram from: 94653196
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: amb8465[00000000]
                  rssi: -53 dBm
                driver: qcaloric
telegram=|_49449344963165943508780DFF5F350082DF0000F00007B06EFFFF00280000DF2C17260000FF2A00280000450097005100200025001E000300000000000000000000002F046D1D10EB2B|+8668
``
for me C1 and T1 is on 868,950MHz and S1 on 868,300MHz.
How it's possible???

weetmuts, 2023-11-11:
Check the README. Wmbusmeters tells rtlwmbus to tune the rtlsdr to 868.625. Rtl-wmbus then peforms software radio stuff and shifts the frequency up to 868.95 and down to 868.300. Listening to both at the same time. Do not ask me how it works.... :-) But please read the whole README. I think you will find a lot of info there.

Anas92230, 2023-11-11:
Yes with rtl_wmbus -s.
Finally when you specify t1 or s1 I suppose that you not add "-s" parameter?
I'm on the way to close my long Day capture. My itron Cyble HRF watermeter don't speak for the moment.

Anas92230, 2023-11-14:
I listen t1 c1 and s1 and...nothing! What do you think about that? If you have another idea to help me ?
My Watermeter if you or someone else ever seen :
[image]

weetmuts, 2023-11-14:
Are you sure it is transmitting wmbus? https://www.itron.com/na/solutions/product-catalog/cyble-5
It seems that meter comes in different configurations, wmbus, lora and sigfox.
These protocols all use ~868MHz.

Anas92230, 2023-11-14:
> Are you sure it is transmitting wmbus? https://www.itron.com/na/solutions/product-catalog/cyble-5
> It seems that meter comes in different configurations, wmbus, lora and sigfox.
> These protocols all use ~868MHz.

With this information I don't believe 🤣. Maybe someone create a tools for that 😁.
Thank you @weetmuts 👍🏽

closed 2024-02-14 by github-actions[bot]