wmbusmeters/wmbusmeters #1093: LSE meter (Siemens WFH36/WFC36?)

DevSecNinja, 2023-11-13:
### Type of request

New device support

### Meter description

Hello,

Thanks for this awesome project! I'm running a im871a in S1 mode and I'm getting the following unknown meters. I have one Siemens WFH36 and one Siemens WFC36 meter and some Qundis Qheat 5s. I suspect these are the Siemens ones. Could be related to #416 or not. (Looks like a different telegram)

Please let me know if you need more telegrams and I'll add them!

Thanks!

### Logged telegrams from meter in decrypted format

```shell
Received telegram from: 00d02470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xed)
                   ver: 0xe4
                device: im871a[00104389]
                  rssi: -89 dBm
                driver: unknown!
telegram=|_09C465B27024D000E4ED|+57
Received telegram from: 00202570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x6c)
                   ver: 0xe5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_09C465B270252000E56C|+75
Received telegram from: 00302570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x6c)
                   ver: 0xe5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_09C465B270253000E56C|+195
Received telegram from: 00e02470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xed)
                   ver: 0xe4
                device: im871a[00104389]
                  rssi: -89 dBm
                driver: unknown!
telegram=|_09C465B27024E000E4ED|+537
Received telegram from: 32402570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xb0)
                   ver: 0xf5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_4BC465B270254032F5B0A0830300620D77003944653270779609D8067AF3000000046D1206ED2B0C13051612004C1366931000426CDF2C326CFFFF02BB56000082046CFF2A8C041393111200|+767
Received telegram from: 00402570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x6c)
                   ver: 0xe5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_09C465B270254000E56C|+795
Received telegram from: 5af02470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x99)
                   ver: 0xf4
                device: im871a[00104389]
                  rssi: -90 dBm
                driver: unknown!
telegram=|_4BC465B27024F05AF499A08303009B0EC4013944653218379909D8077A7D000000046D3216EC2B0C13825748004C1390684200426CDF2C326CFFFF02BB56000082046CFF2A8C041387354800|+1115
Received telegram from: 00f02470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xed)
                   ver: 0xe4
                device: im871a[00104389]
                  rssi: -89 dBm
                driver: unknown!
telegram=|_09C465B27024F000E4ED|+1137
Received telegram from: 3c502570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xb0)
                   ver: 0xf5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_4BC465B27025503CF5B0A0830300620D77003944653270779609D8067AF3000000046D1206ED2B0C13051612004C1366931000426CDF2C326CFFFF02BB56000082046CFF2A8C041393111200|+1370
Received telegram from: 00502570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x6c)
                   ver: 0xe5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_09C465B270255000E56C|+1394
Received telegram from: 00002470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x9a)
                   ver: 0xf4
                device: im871a[00104389]
                  rssi: -90 dBm
                driver: unknown!
telegram=|_4CC465B270240000F49AA0831000530E22003A4465B273730566230C7A1C0000000C05087132004C0560212800426CDF2C326CFFFF046D1E06ED2B0C789418033782046CFF2A8C040585303200|+1717
Received telegram from: 00002470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xed)
                   ver: 0xe4
                device: im871a[00104389]
                  rssi: -89 dBm
                driver: unknown!
telegram=|_09C465B270240000E4ED|+1737
Received telegram from: 04602570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x9a)
                   ver: 0xf4
                device: im871a[00104389]
                  rssi: -73 dBm
                driver: unknown!
telegram=|_4CC465B270256004F49AA0831000530E22003A4465B273730566230C7A1C0000000C05087132004C0560212800426CDF2C326CFFFF046D1E06ED2B0C789418033782046CFF2A8C040585303200|+1977
Received telegram from: 00602570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x6c)
                   ver: 0xe5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_09C465B270256000E56C|+1994
Received telegram from: 00102470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xed)
                   ver: 0xe4
                device: im871a[00104389]
                  rssi: -90 dBm
                driver: unknown!
telegram=|_09C465B270241000E4ED|+2338
Received telegram from: 00702570
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x6c)
                   ver: 0xe5
                device: im871a[00104389]
                  rssi: -72 dBm
                driver: unknown!
telegram=|_09C465B270257000E56C|+2593
Received telegram from: 0a202470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0x71)
                   ver: 0xf6
                device: im871a[00104389]
                  rssi: -89 dBm
                driver: unknown!
telegram=|_4BC465B27024200AF671A0830300460AAB013944653278169709D8067AF2000000046D0B02ED2B0C13284124004C1338072100426CDF2C326CFFFF02BB56000082046CFF2A8C041302302400|+2907
Received telegram from: 00202470
          manufacturer: (LSE) Unknown (0xb265)
                  type: Unknown (0xed)
                   ver: 0xe4
                device: im871a[00104389]
                  rssi: -89 dBm
                driver: unknown!
telegram=|_09C465B270242000E4ED|+2938
```

Anas92230, 2023-11-13:
Hello,
How did you listen specific in S1 mode?
Because i have some trouble with that.

DevSecNinja, 2023-11-13:
> Hello, How did you listen specific in S1 mode? Because i have some trouble with that.

Hmm, under device config in the WMBus Meter addon, I just configured device to be `im871a:s1`?

Anas92230, 2023-11-13:
ok i have a discussion with @weetmuts about that.
when you launch wmbusmeters, you launch this command rtl_sdr.....| rtl_wmbus **-s**, and this option add a shift frequency between 868.3 and 868.95 (around 868.625). On my side I received nothing :-( . 
Thank you for your feedback. I don't pollue anymore your post.

DevSecNinja, 2023-11-13:
Updated the opening post. I received the correct telegrams but sometimes the meter sends a telegram that's unrecognized.

@Anas92230 Oh that's weird. It worked out of the box for me. No worries.

DevSecNinja, 2024-01-14:
Not stale

DevSecNinja, 2024-04-15:
Not stale

weetmuts, 2024-04-15:
Fix pushed. Please test. https://github.com/wmbusmeters/wmbusmeters/commit/d90dbb196e28122e42f0f3394a697570dc54b42d

DevSecNinja, 2024-04-15:
Thank you! To test this on Home Assistant, can I just backup & disable the stable add-on, enable edge, test & revert? Thanks!

weetmuts, 2024-04-15:
I am not an expert on the ha-addon stuff, but it sounds reasonable. :-)

DevSecNinja, 2024-04-16:
Thanks! I'm now running the edge version (1.16.1-27) and still seem to be getting the errors. I'll run it with logging the telegrams and get back to you.

```
Started config im871a[00104389] on /dev/ttyUSB1 listening on s1
(meter) 66057373: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 66057373 mfct: (LSE) Unknown (0xb265) type: Heat volume at inlet meter (0x0c) ver: 0x23
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) neighbor_water_warm_floor_heating: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 66057393 mfct: (LSE) Unknown (0xb265) type: Heat volume at inlet meter (0x0c) ver: 0x1a
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) 09971639: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09971639 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) unknown_warm_water_meter_2: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09971629 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) unknown_warm_water_meter_4: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09971636 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) unknown_warm_water_meter_1: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09967781 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(wmbus) resetting /dev/ttyUSB1:im871a[00104389]
(wmbus) reset completed /dev/ttyUSB1:im871a[00104389]
```

DevSecNinja, 2024-04-16:
Got the telegrams too - hope these help!

```
Started config im871a[00104389] on /dev/ttyUSB1 listening on s1
(meter) 66057373: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 66057373 mfct: (LSE) Unknown (0xb265) type: Heat volume at inlet meter (0x0c) ver: 0x23
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|2E4465B273730566230C7ABF000000_0C05965135004C0520123400426CFF2C326CFFFF046D1D0910340C7894180337|+2227
(meter) unknown_warm_water_meter_1: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09967781 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|2D44653281779609D8067A96000000_046D040C10340C13910009004C1307630800426CFF2C326CFFFF02BB560000|+3581
telegram=|2D44653223369909D8077A22000000_046D0F0B10340C13702634004C1365823200426CFF2C326CFFFF02BB560000|+4859
(meter) unknown_warm_water_meter_2: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09971629 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|2D44653229169709D8067A97000000_046D130D10340C13933008004C1396960700426CFF2C326CFFFF02BB560000|+7832
(meter) unknown_warm_water_meter_4: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09971636 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|2D44653236169709D8067A97000000_046D330D10340C13065720004C1326751900426CFF2C326CFFFF02BB560000|+8050
(meter) neighbor_water_warm_floor_heating: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 66057393 mfct: (LSE) Unknown (0xb265) type: Heat volume at inlet meter (0x0c) ver: 0x1a
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|334465B2937305661A0C7A9D000000_046D390B10340C0576230300426CFF2C4C0540230300326CFFFF02FD1700000C7834989959|+11276
(meter) 09971639: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 09971639 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xd8
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination

```

DevSecNinja, 2024-06-16:
Not stale.

DevSecNinja, 2024-08-16:
Not stale.

closed 2024-11-15 by github-actions[bot]