wmbusmeters/wmbusmeters #1094: Looking for help how to set up properly - apator162, MQTT

DoktorPrase, 2023-11-15:
### Type of request

Question

### HAOS version

2023.11.2

### wmbusmeters version

1.14.0

### Your message goes here

FIrst of all - thanks for this addon!
Second of all - I am not a linux guy, just started with HA, so i might be "stupid". Sorry

Now: rather fresh install, see config and log below:
mqtt section of configuration.yaml
```
mqtt:
  sensor:
    - state_topic: "wmbusmeters/KoupelnaTepla"
      json_attributes_topic: "wmbusmeters/KoupelnaHot"
      unit_of_measurement: "m³"
      value_template: "{{ value_json.total_m3 }}"
      name: Water usage
      icon: "mdi:gauge"
      state_class: total_increasing
      device_class: water
    - state_topic: "wmbusmeters/KoupelnaStudena"
      json_attributes_topic: "wmbusmeters/KoupelnaCold"
      unit_of_measurement: "m³"
      value_template: "{{ value_json.total_m3 }}"
      name: Water usage
      icon: "mdi:gauge"
      state_class: total_increasing
      device_class: water
    - state_topic: "wmbusmeters/WCTepla"
      json_attributes_topic: "wmbusmeters/ToiletHot"
      unit_of_measurement: "m³"
      value_template: "{{ value_json.total_m3 }}"
      name: Water usage
      icon: "mdi:gauge"
      state_class: total_increasing
      device_class: water
    - state_topic: "wmbusmeters/WCStudena"
      json_attributes_topic: "wmbusmeters/ToiletCold"
      unit_of_measurement: "m³"
      value_template: "{{ value_json.total_m3 }}"
      name: Water usage
      icon: "mdi:gauge"
      state_class: total_increasing
      device_class: water
    - state_topic: "wmbusmeters/KuchynTepla"
      json_attributes_topic: "wmbusmeters/KitchenHot"
      unit_of_measurement: "m³"
      value_template: "{{ value_json.total_m3 }}"
      name: Water usage
      icon: "mdi:gauge"
      state_class: total_increasing
      device_class: water
    - state_topic: "wmbusmeters/KuchynStudena"
      json_attributes_topic: "wmbusmeters/KitchenCold"
      unit_of_measurement: "m³"
      value_template: "{{ value_json.total_m3 }}"
      name: Water usage
      icon: "mdi:gauge"
      state_class: total_increasing
      device_class: water
[... 131 lines cut ...]
172.30.32.2 - - [15/Nov/2023:23:13:29 +0100] "GET / HTTP/1.1" 200 1288 "http://192.168.1.215:8123/c8a990ad_wmbusmeters-ha-addon" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/119.0.0.0 Safari/537.36"
172.30.32.2 - - [15/Nov/2023:23:13:29 +0100] "GET /bootstrap.min.css HTTP/1.1" 200 221221 "http://192.168.1.215:8123/api/hassio_ingress/lNBH2KwcyyKAl1KCl0ukhufvx5bbxFGi2I30kKgbmrw/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/119.0.0.0 Safari/537.36"
172.30.32.2 - - [15/Nov/2023:23:13:29 +0100] "GET /main.js HTTP/1.1" 200 9981 "http://192.168.1.215:8123/api/hassio_ingress/lNBH2KwcyyKAl1KCl0ukhufvx5bbxFGi2I30kKgbmrw/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/119.0.0.0 Safari/537.36"
172.30.32.2 - - [15/Nov/2023:23:13:29 +0100] "GET /bootstrap.bundle.min.js HTTP/1.1" 200 80762 "http://192.168.1.215:8123/api/hassio_ingress/lNBH2KwcyyKAl1KCl0ukhufvx5bbxFGi2I30kKgbmrw/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/119.0.0.0 Safari/537.36"
172.30.32.2 - - [15/Nov/2023:23:13:29 +0100] "GET /wmbusmeters.png HTTP/1.1" 200 55338 "http://192.168.1.215:8123/api/hassio_ingress/lNBH2KwcyyKAl1KCl0ukhufvx5bbxFGi2I30kKgbmrw/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/119.0.0.0 Safari/537.36"
172.30.32.2 - - [15/Nov/2023:23:13:29 +0100] "GET /get_json HTTP/1.1" 200 939 "http://192.168.1.215:8123/api/hassio_ingress/lNBH2KwcyyKAl1KCl0ukhufvx5bbxFGi2I30kKgbmrw/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/119.0.0.0 Safari/537.36"
telegram=|6E4401063957370405077AC90060852F2F_0FC0AF6B97080200430C02830018D65C88CB01B8291210225100007B01F94C000065450000AC3D0000923A0000DE350000A5300000F6280000C5200000481C000045170000570F000040070000A0CF259304FFFFFFFFFFFFFFFFFFFFB9C1|+148
/bin/sh: -h: not found
(shell) /bin/sh exited with non-zero return code: 127
telegram=|6E4401060579380405077AD10060852F2F_0FC0AF6B97080200430C02830018D66C9CF80399D21310781900007B01C6170000CB140000501200009A11000060100000C10E0000300C0000B2090000C208000020070000C80400000B020000A0C9239304FFFFFFFFFFFFFFFFFFFFE98B|+158
/bin/sh: -h: not found
(shell) /bin/sh exited with non-zero return code: 127
telegram=|6E4401064410720405077A6D0060852F2F_0FCFAF6B9708020043A301830008DA2288CB01F92E0410609300007B01B38C0000EF7E00004C710000BA690000315F0000D755000025480000E83900005931000005270000281A0000990A0000A034EF9A04FFFFFFFFFFFFFFFFFFFF6A28|+189
(shell) /bin/sh exited with non-zero return code: 127
/bin/sh: -h: not found
telegram=|6E4401063508720405077A2F0060852F2F_0FD4AF6B9708000043A301830008DA8A68CB01982B0410AB6300007B019E5E00000B5600003F4D00009A48000025420000BA3A00006F30000026270000C82100000B1A0000AD10000045070000A0F5EE9A04FFFFFFFFFFFFFFFFFFFF739F|+288
/bin/sh: -h: not found
(shell) /bin/sh exited with non-zero return code: 127
telegram=|6E4401066052700405077ADA0060852F2F_0FCFAF6B9718020043A401847361F8444EFA0508000010882100007B018D1F0000271C0000051800005716000012140000C4110000780E0000960B0000440A0000090800003E05000052020000A037349B04FFFFFFFFFFFFFFFFFFFF1415|+299
/bin/sh: -h: not found
(shell) /bin/sh exited with non-zero return code: 127
```

Can you please point me into the right direction?
Thanks.

DoktorPrase, 2023-11-16:
I have copied from the sourcecode the apator162.json into /homeassistant/wmbusmeters/etc/mqtt_discovery and the "suspicious secodn thing" is gone.

weetmuts, 2023-11-16:
shell=shell=/usr/bin/mosquitto_pub
looks like one shell= too many? :-)

DoktorPrase, 2023-11-16:
oh. my. Sorry for being stupid (but I have warned you :D ).

In the meantime I have restarted a couple times and since the MQTT discovery started working (I guess after coyping the apator162.json to the MQTT discovery folder) I deleted the mqtt section of the configuration.yaml - lo and behold, the sensors finally appeared, but they are there twice - one as a "sensor integration" and one as a "MQTT integration":
[image]

DoktorPrase, 2023-11-16:
After yet another restart and the duplicate sensors (those integrated as "Senzor") are gone.
The double "shell" was a mistake.

Lesson learned: doublecheck the code, restart often :D

closed 2023-11-16 by DoktorPrase