wmbusmeters/wmbusmeters #1097: Water meter missing in HA

MichalSasinski, 2023-11-17:
### Type of request

Question

### HAOS version

11,1

### wmbusmeters version

1.14.0

### Your message goes here

Hello,
I think I am very close to see water meter in HA, just missing last step. I can see that messages and values are visible in logs, still I cannot see it in HA nor add to energy dashboard.
Thank you for any suggestions,
Michal

**wmbusmeters.conf:**

device=auto:t1
donotprobe=/dev/ttyAMA0
format=json
logfile=/dev/stdout
loglevel=debug
logtelegrams=false
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"

**meter-0001:**

id=41229638
key=
name=Licznik_wody

**Current entry in configuration.yaml:**

mqtt:
  sensor:
    - name: "Licznik_Wody_Total"
      state_topic: "wmbusmeters/Licznik_wody"
      unit_of_measurement: "m³"
      device_class: water
      state_class: total_increasing
      value_template: "{{ value_json.total_m3 }}"

**Logs**

(shell) env "METER_MANUFACTURE_YEAR=2019"
(shell) waiting for child 8223 to complete.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh: return code 1
(shell) /bin/sh exited with non-zero return code: 1
(serial) received binary "A5C2031944304C38962241CC01A2511600139C63DD5B494294C43DCA5170C071"
(im871a) checkIM871AFrame "A5C2031944304C38962241CC01A2511600139C63DD5B494294C43DCA5170C071"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 112 (-66 dBm)
(im871a) got crc16 71c0 expected 71c0
(im871a) received full frame
(diehl) preprocess necessary SAP_PRIOS
(diehl) Pre-processing: setting device type to water meter for SAP PRIOS
(wmbus) parseDLL @0 26
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(meter) Licznik_wody: for me? 41229638 in 41229638
(meter) Licznik_wody: yes for me
(meter) Licznik_wody(1) izar  handling telegram from 41229638
(meter) Licznik_wody 41229638 "1944304C38962241CC01A2511600139C63DD5B494294C43DCA51"
(diehl) preprocess necessary SAP_PRIOS
(diehl) Pre-processing: setting device type to water meter for SAP PRIOS
(wmbus) parseDLL @0 26
(telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=41229638 VER=00 TYPE=07 (Water meter) (driver izar) DEV=im871a[00104084] RSSI=-66
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) TPL CI=a2
(meters) skipping field without matcher prefix(Text)[0]...
(meters) skipping field without matcher serial_number(Text)[1]...
(meters) skipping field without matcher total(Volume)[2]...
(meters) skipping field without matcher last_month_total(Volume)[3]...
(meters) skipping field without matcher last_month_measure_date(Text)[4]...
(meters) skipping field without matcher remaining_battery_life(Time)[5]...
(meters) skipping field without matcher current_alarms(Text)[6]...
(meters) skipping field without matcher previous_alarms(Text)[7]...
(meters) skipping field without matcher transmit_period(Time)[8]...
(meters) skipping field without matcher manufacture_year(Text)[9]...
(izar) Decoded PRIOS data: 4B28E909004FCD0900FF2A
(izar) 000   : 19 length (25 bytes)
(izar) 001   : 44 dll-c (from meter SND_NR)
(izar) 002   : 304c dll-mfct (SAP)
(izar) 004   : 38962241 dll-id (41229638)
(izar) 008   : 00 dll-version
(izar) 009   : 07 dll-type (Water meter)
(izar) 010   : a2 tpl-ci-field (Mfct specific)
(izar) 011 C?: 511600139C63DD5B494294C43DCA51 mfct specific
(meters) render field prefix(Text)[0] without dventry
(meters)             "prefix":"C19SB"
(meters) render field serial_number(Text)[1] without dventry
(meters)             "serial_number":"043896"
(meters) render field total(Volume)[2] without dventry
(meters)             "total_m3":649.512
(meters) render field last_month_total(Volume)[3] without dventry
(meters)             "last_month_total_m3":642.383
(meters) render field last_month_measure_date(Text)[4] without dventry
(meters)             "last_month_measure_date":"2023-10-31"
(meters) render field remaining_battery_life(Time)[5] without dventry
(meters)             "remaining_battery_life_y":11
(meters) render field current_alarms(Text)[6] without dventry
(meters)             "current_alarms":"no_alarm"
(meters) render field previous_alarms(Text)[7] without dventry
(meters)             "previous_alarms":"no_alarm"
(meters) render field transmit_period(Time)[8] without dventry
(meters)             "transmit_period_s":8
(meters) render field manufacture_year(Text)[9] without dventry
(meters)             "manufacture_year":"2019"
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON""
(shell) env "METER_JSON={"media":"water","meter":"izar","name":"Licznik_wody","id":"41229638","prefix":"C19SB","serial_number":"043896","total_m3":649.512,"last_month_total_m3":642.383,"last_month_measure_date":"2023-10-31","remaining_battery_life_y":11,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"2019","timestamp":"2023-11-17T13:21:47Z","device":"im871a[00104084]","rssi_dbm":-66}"
(shell) env "METER_ID=41229638"
(shell) env "METER_NAME=Licznik_wody"
(shell) env "METER_MEDIA=water"
(shell) env "METER_TYPE=izar"
(shell) env "METER_TIMESTAMP=2023-11-17T13:21:47Z"
(shell) env "METER_TIMESTAMP_UTC=2023-11-17T13:21:47Z"
(shell) env "METER_TIMESTAMP_UT=1700227307"
(shell) env "METER_TIMESTAMP_LT=2023-11-17 14:21.47"
(shell) env "METER_DEVICE=im871a[00104084]"
(shell) env "METER_RSSI_DBM=-66"
(shell) env "METER_PREFIX=C19SB"
(shell) env "METER_SERIAL_NUMBER=043896"
(shell) env "METER_TOTAL_M3=649.512"
(shell) env "METER_LAST_MONTH_TOTAL_M3=642.383"
(shell) env "METER_LAST_MONTH_MEASURE_DATE=2023-10-31"
(shell) env "METER_REMAINING_BATTERY_LIFE_Year=11"
(shell) env "METER_CURRENT_ALARMS=no_alarm"
(shell) env "METER_PREVIOUS_ALARMS=no_alarm"
(shell) env "METER_TRANSMIT_PERIOD_Second=8"
(shell) env "METER_MANUFACTURE_YEAR=2019"
(shell) waiting for child 8233 to complete.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh: return code 1
(shell) /bin/sh exited with non-zero return code: 1

MichalSasinski, 2023-11-20:
Hello,
Little update. I can see the sensor now in HA. I was missing Mosquito broker. Now I have problems with passing values to broker:

(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON" -h localhost -t wmbusmeters/"$METER_ID" -m "$METER_JSON""
(shell) env "METER_JSON={"media":"water","meter":"izar","name":"Licznik_wody","id":"41229638","prefix":"C19SB","serial_number":"043896","total_m3":650.152,"last_month_total_m3":642.383,"last_month_measure_date":"2023-10-31","remaining_battery_life_y":11,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"2019","timestamp":"2023-11-20T08:22:07Z","device":"im871a[00104084]","rssi_dbm":-67}"
(shell) env "METER_ID=41229638"
(shell) env "METER_NAME=Licznik_wody"
(shell) env "METER_MEDIA=water"
(shell) env "METER_TYPE=izar"
(shell) env "METER_TIMESTAMP=2023-11-20T08:22:07Z"
(shell) env "METER_TIMESTAMP_UTC=2023-11-20T08:22:07Z"
(shell) env "METER_TIMESTAMP_UT=1700468527"
(shell) env "METER_TIMESTAMP_LT=2023-11-20 09:22.07"
(shell) env "METER_DEVICE=im871a[00104084]"
(shell) env "METER_RSSI_DBM=-67"
(shell) env "METER_PREFIX=C19SB"
(shell) env "METER_SERIAL_NUMBER=043896"
(shell) env "METER_TOTAL_M3=650.152"
(shell) env "METER_LAST_MONTH_TOTAL_M3=642.383"
(shell) env "METER_LAST_MONTH_MEASURE_DATE=2023-10-31"
(shell) env "METER_REMAINING_BATTERY_LIFE_Year=11"
(shell) env "METER_CURRENT_ALARMS=no_alarm"
(shell) env "METER_PREVIOUS_ALARMS=no_alarm"
(shell) env "METER_TRANSMIT_PERIOD_Second=8"
(shell) env "METER_MANUFACTURE_YEAR=2019"
(shell) waiting for child 97240 to complete.
/bin/sh: wmbusmeters/Licznik_wody: not found
(shell) /bin/sh: return code 127
(shell) /bin/sh exited with non-zero return code: 127

Thank you for your support,
Michal

BIBOLV, 2023-11-21:
Hi,
What is your current config `shell=` content?

MichalSasinski, 2023-11-21:
Hi,
It is:
shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON" -h localhost -t wmbusmeters/"$METER_ID" -m "$METER_JSON"

MichalSasinski, 2023-11-21:
Thank you for pointing this out. After testing so many options I ended up with actual
 "shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON" -h localhost -t wmbusmeters/"$METER_ID" -m "$METER_JSON""
as value. I have removed "shell=" and it works like a charm.
Thanks again

BIBOLV, 2023-11-21:
Actually it should work also with `/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"` - original config line for `shell`

closed 2023-11-21 by BIBOLV