wmbusmeters/wmbusmeters #1100: SENSUS Meter sends telegrams very infrequently

aifrog, 2023-11-19:
### Type of request

Question

### OS version

Raspian / Debian

### wmbusmeters version

12 (bookworm)

### Your message goes here

I installed wmbusmeters on a RPI4 using an im871USB-Stick. I am able to receive telgrams once in a while - every couple of hours. The meter is a SENSUS for water and I uses the iperl driver. I listen to c1,t1 telegrams using

device=/dev/ttyUSB0:im871a[00104335]:c1,t1

I am rather disappointed as I expectected the meter to send telegrams every x seconds/minutes. Is this expectation wrong or is the setup not correct? Is there any posibility to do polling and how?

I tried:

`wmbusmeters auto:c1,t1 Water auto 22345516 XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX`

which works VERY slow (hours), but I receive (and decode) telegrams successfully.

When trying

`wmbusmeters --pollinterval=60s auto:c2,t2 Water auto 22345516  XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX`

I get the error message:

`Warning! Desired link modes t2,c2 cannot be set for device /dev/ttyUSB0:im871a[00104335]`

Thx, H-C

Andi252, 2023-11-19:
Hello H-C,

what is the RSSI value in the c1/t1 mode you received from your meter? 
The c1/t1 is a unidirectional protocol, every meter broadcasts the current value in a fixed interval.

You can try to move the im871a dongle with a USB extension cord to a more open place.

In my limited experience, the im871a dongle is one of the most sensitive receivers on the market,
a lot better than a nanocul.

aifrog, 2023-11-19:
The RSSI value is -52dbm, its all in the same room (line of sight 4m). That sounds pretty much OK to me.
Whats a "more open place"? Closer of further away?

Thx, H-C

Andi252, 2023-11-19:
-52 dbm sounds great.

> Whats a "more open place"?

I thought, maybe you have your USB dongle with a closed box / cupboard and therefor a very low signal.

> What should I expect? Every 1min, 5min, 60min? And can I switch to a c2/t2 mode to poll?

I don't know, you should check the documentation for your meter. I found a description for a Sensus heat meter, which talks about 16 second in C1 mode and 900 seconds in T1 mode.

You should try a different USB port, I read somewhere there are problems with a raspberry with interference for WIFI 2.4 GHz when using the usb 3 port.

aifrog, 2023-11-19:
I get a telegram every hour now. This is true for USB2 and USB3 - no difference.
Does anybody have experience with the telegram frequency of a SENSUS water sensor? Or does anybody now how to poll?

weetmuts, 2023-11-20:
How often the meter transmits probably depends on how they collect the readings.

A drive-by or walk-by requires frequent transmits but if they have a permanent collecting station nearby
that relays the values  then they can send once per hour.

Some meters are configured to not transmit during night and weekends, because there will be no drive-by at that time anyway....

As for C2/T2 polling, only meters pre-configured for accepting a poll request will respond. So most likely is your meter not configured for this, so it will not listen to C2/T2 requests.

Also the support for polling in wmbusmeters right now only works for wired mbus. However all the infrastructure is in place for implementing C2/T2 polling using an im871a dongle. There has not yet been any user with that requirement and the necessary meter key, to develop this feature. :-)

aifrog, 2023-11-25:
I asked the local utility and got the following answer:

- The sensor is transmitting a telegram every hour via wmbus. It is not possible to change this. No polling.
- The sensor is transmitting much more frequent data via the SensusRF proprietary protocol (maybe FlexNet?)

Does anybody know about this? Is there a tool/gateway for receiving the data via the proprietay protocol?

I am not willing to send unknown data to external parties and if I cannot read the data I am inclined to setup a legal process according to the Data Protection Act. Any experience/links for that?

Best, H-C

swordeh, 2024-01-22:
@aifrog I spent some time looking into this. The 15 second message you're seeing is, I believe, what Sensus call their "Bubble Up Protocol (BUP)" which contains endpoint ID, total consumption, alarm flags. I tried to take a stab at capturing this and figuring out how to read the data, but it was a bit beyond my ability. Some extra info here: https://www.xylem.com/siteassets/industries--applications/resources/comm-networks/en---sensusrf-brochure.pdf

closed 2024-04-22 by github-actions[bot]