wmbusmeters/wmbusmeters #1102: Techem Meter Configuration

k1tfh, 2023-11-21:
### Type of request

Question

### OS version

wmbus HA-Addon

### wmbusmeters version

1.14.0

### Your message goes here

I have 2 watermetes and 1 heatmeter from techem. From all of them i receive an encrypted telegram. Today i got my aes-keys from techem for decryption. But sadly all my meters are getting irgnored with the hint of a wrong password. So i took a closer look and i dont know exactly how to configure my meters. This is what i receive:

Received telegram from: 13178487
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37) encrypted
                   ver: 0x95
      Concerning meter: 01464871
          manufacturer: (WEH) E Wehrle, Germany (0x5ca8)
                  type: Water meter (0x07) encrypted
                   ver: 0xfe
                device: rtlwmbus[00000001]
                  rssi: 135 dBm
                driver: mkradio4a weh_07

So i did the following meters-config:

name: Cold-Water
driver: mkradio4a weh_07
id: 01464871
key: -key provided as "key s" from techem-

It leads to this:

(meter) Cold_Water: meter detection did not match the selected driver ! correct driver is: mkradio4a weh_07
(meter) Not printing this warning again for id: 13178487 mfct: (TCH) Techem Service (0x5068) type: Radio converter (meter side) (0x37) ver: 0x9

Changing the driver:

name: Cold-Water
driver: mkradio4a
id: 01464871
key: -key provided as "key s" from techem-

(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 13178487 mfct: (TCH) Techem Service (0x5068) type: Radio converter (meter side) (0x37) ver: 0x95

Sill not working. 

name: Cold-Water
driver: mkradio4a
id: 13178487
key: -key provided as "s key" from techem-

(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 13178487 mfct: (TCH) Techem Service (0x5068) type: Radio converter (meter side) (0x37) ver: 0x95

name: Cold-Water
driver: mkradio4a
id: 13178487
key: -key provided as "a keys" from techem-

(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 13178487 mfct: (TCH) Techem Service (0x5068) type: Radio converter (meter side) (0x37) ver: 0x95
(meter) newly created meter (Cold_Water 01464871 mkradio4a) did not handle telegram!

So either my config is messed up or the keys are wrong. i can change the config not the key, am i doing it right, any suggestions?

Andi252, 2023-11-21:
Try to analyse the telegram with the website https://wmbusmeters.org/ its a web frontend for wmbusmeters.
You can add the telegram and your key and test with different decoders.
The key should be a string with 32 hex digits.

k1tfh, 2023-11-22:
Thank you! 

For everyone having trouble with techem, here are my meters and my meter-config:

```
 Received telegram from: 24256059
           manufacturer: (TCH) Techem Service (0x5068)
                   type: Radio converter (meter side) (0x37) encrypted
                    ver: 0x95
      Concerning meter: 24434599
           manufacturer: (WEH) E Wehrle, Germany (0x5ca8)
                   type: Warm Water (30°C-90°C) meter (0x06) encrypted
                   ver: 0xfe
                 device: rtlwmbus[00000001]
                  rssi: 137 dBm
                driver: mkradio4a

driver: topaseskr
id: 24434599
key: S key from techem
pollinterval: 60s
```

ignore this warning: (meter) Warmwasser: meter detection did not match the selected driver topaseskr! correct driver is: mkradio4a

```
Received telegram from: 13178487
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37) encrypted
                   ver: 0x95
      Concerning meter: 01464871
          manufacturer: (WEH) E Wehrle, Germany (0x5ca8)
                  type: Water meter (0x07) encrypted
                   ver: 0xfe
                device: rtlwmbus[00000001]
                  rssi: 140 dBm
                driver: mkradio4a weh_07

driver: weh_07
id: 01464871
key: A key from techem
pollinterval: 60s
```

```
Received telegram from: 13348741
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0x04) encrypted
                   ver: 0x17
                device: rtlwmbus[00000001]
                  rssi: 118 dBm
                driver: vario451mid

driver: vario451mid
id: 13348741
key: A key from techem
pollinterval: 60s
```

Hope this will save someone some time. Any improvements are welcome.

Andi252, 2023-11-23:
Hello, can you add you decrypted telegrams?

The decrypted telegrams are needed to create a new driver or update an existing one, because every driver and case should have some example telegrams.

closed 2024-02-23 by github-actions[bot]

suitable-hero, 2024-10-14:
Can you share the decryption key? Is that key unique to your devices only or can it be used to decrypt other techem meters? Sorry for the noob questions :)