wmbusmeters/wmbusmeters #1117: Integra Metering Topas Sonic

Flearz, 2023-12-04:
### Type of request

New device support

### Meter description

From the website https://ch.integra-metering.com/fr/product/topas-sonic/

Ultrasonic counter (static, no moving parts) for intelligent metering applications

Free-Flow technology: no reflectors in the pipe:
No adherence of dirt, high accuracy, low pressure drop

Wide range of connectivity options
- LoRaWAN (LoRa Alliance certified)
- MultiCom mode for flexible upgrades to LoRaWAN networks

Compatible with ParamApp application
- Easy configuration/diagnostics in the field with the user-friendly ParamApp application

Intelligent meter
- A wealth of information available, and a full range of event detectors to manage the network (leak, burst, overload, water temperature, reverse flow) 

Robust: IP68, up to 16-year life, tamper-proof by design

### Logged telegrams from meter in decrypted format

```shell
Auto driver  : not found!
Best driver  : hydrus 20/26
Using driver : hydrus 00/00
000   : 41 length (65 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b425 dll-mfct (IMT)
004   : 38982110 dll-id (10219838)
008   : 10 dll-version
009   : 07 dll-type (Water meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : e8 ell-acc
013   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
014   : 90 tpl-acc-field
015   : 10 tpl-sts-field (TEMPORARY_ERROR)(TEMPORARY_ERROR)
016   : 3025 tpl-cfg 2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
018   : 2f2f decrypt check bytes (OK)
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : 13 vif (Volume l)
022 C!: 02000000 ("total_m3":0.002)
026   : 84 dif (32 Bit Integer/Binary Instantaneous value)
027   : 10 dife (subunit=0 tariff=1 storagenr=0)
028   : 13 vif (Volume l)
029 C!: 00000000 ("total_tariff1_m3":0)
033   : 04 dif (32 Bit Integer/Binary Instantaneous value)
034   : 3B vif (Volume flow l/h)
035 C!: 00000000 ("flow_m3h":0)
039   : 02 dif (16 Bit Integer/Binary Instantaneous value)
040   : 5A vif (Flow temperature 10⁻¹ °C)
041 C!: E800 ("flow_temperature_c":23.2)
043   : 04 dif (32 Bit Integer/Binary Instantaneous value)
044   : 6D vif (Date and time type)
045 C!: 1630E42C ("meter_datetime":"2023-12-04 16:22")
049   : 7C dif (8 digit BCD Value during error state storagenr=1)
050   : 13 vif (Volume l)
051 C?: 00000000
055   : 02 dif (16 Bit Integer/Binary Instantaneous value)
056   : FD vif (Second extension FD of VIF-codes)
057   : 17 vife (Error flags (binary))
058 C?: 0008
060   : 02 dif (16 Bit Integer/Binary Instantaneous value)
061   : FD vif (Second extension FD of VIF-codes)
062   : 74 vife (Reserved)
063 C!: 0716 ("remaining_battery_life_y":15.439058)
065   : 2F skip
066   : 71 dif (8 Bit Integer/Binary Value during error state storagenr=1)

{
    "media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"10219838",
    "flow_m3h":0,
    "flow_temperature_c":23.2,
    "remaining_battery_life_y":15.439058,
    "total_m3":0.002,
    "total_tariff1_m3":0,
    "meter_datetime":"2023-12-04 16:22",
    "status":"TEMPORARY_ERROR",
    "timestamp":"2023-12-04T16:18:25Z"
}

Using: wmbusmeters: 1.14.0-48-gc4b9f0d
c4b9f0d10477cb6bb2472e1f2de5eb08ba42e7c4
```

gh0stface33, 2023-12-06:
Hi, i have the same meter but i can't retrieve decrypted key from my supplier.
Have you a decrypted key for your meter ?

If yes can you test to decrypt my telegram with your decrypted key ?

telegram=|_4E44B4252178101010077A1A00402571B721D881E204544353B85C1E05267927264C75D8ABAB21F6D7EC8081CD4494C59D1F24FD590FB5B253F768FA3DFC8404AB6910ABC01C17DCCCC38C24E06449|+12

Thanks

gh0stface33, 2023-12-22:
@Flearz 
Can you help me please ?

closed 2024-03-23 by github-actions[bot]